projectacrn/acrn-hypervisor #5406: Doc: list all PCI devices available when using `acrn-dm -s` (`emul` value)

gvancuts, 2020-10-21:
The ACRN Device Model can set-up PCI devices to be used by the User VM it is starting. The parameter to create such PCI device is `acrn-dm -s`. It takes, amongst other things, a string describing the PCI device being emulated and (optionally) some configuration values for it. Examples of such PCI devices are `virtio-net`, `virtio-blk`, etc.

We should document all possible values and the options that are available for each of those.

dbkinder, 2021-02-08:
@gvancuts Is the documentation here sufficient?  https://projectacrn.github.io/latest/user-guides/acrn-dm-parameters.html#device-model-parameters

gvancuts, 2021-02-08:
Not quite, what I would like to see is an exhaustive list of all possible values for `emul` and their options. Things that should go in such list would be: `virtio-blk`, `virtio-net`, `virtio-console`, `virtio-rnd`, `virtio-hyper_dmabuf`, `virtio-rpmb`, `passthru`, `xhci`, `wdt-i6300esb`,  `pci-gvt` and `ivshmem`. I may be missing some in there ;-)

dbkinder, 2021-02-10:
@NanlinXie Who in the development team can provide this information so we can update the documentation?

gvancuts, 2021-03-29:
Adding to the list of things to be documented:
* `passthru` and all its options, e.g. `gpu`, `d3hot_reset`, `keep_gsi`

NanlinXie, 2022-01-04:
@gvancuts ， could we close it since patch merged?

gvancuts, 2022-01-04:
Yes

closed 2022-01-04 by gvancuts