projectacrn/acrn-hypervisor #5451: vmei_proc_tx()  the function using data that coming from user buffer without checking the user buffer length

tomasbw, 2020-11-02:
**Describe the bug**
In  vmei_proc_tx()  the function using data that coming from user buffer without checking the user buffer length, that can cause to read out of bound. 

**Platform**
APL-I, Any apply 

**Codebase**
master 
Need backport to apl_sdc_stable

**Scenario**
Describe the scenario: Hybrid, Industry or logical_partition

**To Reproduce**
In case of misbehaving UOS. The issue discovered after code review. 

**Expected behavior**
Detect memory boundaries crossing 

**Additional context**
Attach the following logs if available:

- native Linux boot kernel message log
- ACRN serial port log (when issues happens)
- Service VM kernel message when issue happens
- acrn-dm log: all files in /var/log/acrn-tm
- console log when running script to launch guest VM

wenlingz, 2020-11-16:
please add JIRA info, thanks

fuzhongl, 2020-11-17:
[External_System_ID]ACRN-6829

closed 2022-05-04 by NanlinXie