projectacrn/acrn-hypervisor #5468: Compile SDC scenario for apl-up2-n3350

tomwei666, 2020-11-03:
**Describe the bug**
A clear and concise description of what the bug is.

I want to compile SDC scenario for my platfrom.
I compile like this:
make all BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/apl-up2-n3350.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/apl-up2-n3350/sdc_launch_1uos_laag.xml 

It output these error:
No TARGET_DIR parameter is specified, the original configuration source is overwritten!
make -C /home/tom/work/acrn-hypervisor/hypervisor BOARD=apl-up2-n3350 HV_OBJDIR=/home/tom/work/acrn-hypervisor/build/hypervisor BOARD_FILE=/home/tom/work/acrn-hypervisor/misc/vm_configs/xmls/board-xmls/apl-up2-n3350.xml SCENARIO_FILE=/home/tom/work/acrn-hypervisor/misc/vm_configs/xmls/config-xmls/apl-up2-n3350/sdc_launch_1uos_laag.xml TARGET_DIR= defconfig;
make[1]: Entering directory '/home/tom/work/acrn-hypervisor/hypervisor'
Traceback (most recent call last):
  File "../misc/acrn-config/board_config/board_cfg_gen.py", line 144, in <module>
    err_dic = main(ARGS)
  File "../misc/acrn-config/board_config/board_cfg_gen.py", line 116, in main
    err_dic = misc_cfg_h.generate_file(config)
  File "/home/tom/work/acrn-hypervisor/misc/acrn-config/board_config/misc_cfg_h.py", line 217, in generate_file
    sos_cmdlines = [i for i in sos_cmdlines[0].split() if i != '']
IndexError: list index out of range
/bin/sh: 9: [: /home/tom/work/acrn-hypervisor/misc/vm_configs/scenarios/sdc: unexpected operator
Writing /home/tom/work/acrn-hypervisor/build/hypervisor/.config with /home/tom/work/acrn-hypervisor/misc/vm_configs/scenarios/sdc uos_launcher=1/apl-up2-n3350/apl-up2-n3350.config
cp: target '/home/tom/work/acrn-hypervisor/build/hypervisor/.config' is not a directory
make[1]: Leaving directory '/home/tom/work/acrn-hypervisor/hypervisor'
make -C /home/tom/work/acrn-hypervisor/hypervisor BOARD=apl-up2-n3350 HV_OBJDIR=/home/tom/work/acrn-hypervisor/build/hypervisor BOARD_FILE=/home/tom/work/acrn-hypervisor/misc/vm_configs/xmls/board-xmls/apl-up2-n3350.xml SCENARIO_FILE=/home/tom/work/acrn-hypervisor/misc/vm_configs/xmls/config-xmls/apl-up2-n3350/sdc_launch_1uos_laag.xml TARGET_DIR= oldconfig;
make[1]: Entering directory '/home/tom/work/acrn-hypervisor/hypervisor'
Traceback (most recent call last):
  File "../misc/acrn-config/board_config/board_cfg_gen.py", line 144, in <module>
    err_dic = main(ARGS)
  File "../misc/acrn-config/board_config/board_cfg_gen.py", line 116, in main
    err_dic = misc_cfg_h.generate_file(config)
  File "/home/tom/work/acrn-hypervisor/misc/acrn-config/board_config/misc_cfg_h.py", line 217, in generate_file
    sos_cmdlines = [i for i in sos_cmdlines[0].split() if i != '']
IndexError: list index out of range
Configuration written to /home/tom/work/acrn-hypervisor/build/hypervisor/.config.
make[1]: Leaving directory '/home/tom/work/acrn-hypervisor/hypervisor'
make -C /home/tom/work/acrn-hypervisor/hypervisor BOARD=apl-up2-n3350 HV_OBJDIR=/home/tom/work/acrn-hypervisor/build/hypervisor BOARD_FILE=/home/tom/work/acrn-hypervisor/misc/vm_configs/xmls/board-xmls/apl-up2-n3350.xml SCENARIO_FILE=/home/tom/work/acrn-hypervisor/misc/vm_configs/xmls/config-xmls/apl-up2-n3350/sdc_launch_1uos_laag.xml TARGET_DIR=
make[1]: Entering directory '/home/tom/work/acrn-hypervisor/hypervisor'
Traceback (most recent call last):
  File "../misc/acrn-config/board_config/board_cfg_gen.py", line 144, in <module>
    err_dic = main(ARGS)
  File "../misc/acrn-config/board_config/board_cfg_gen.py", line 116, in main
    err_dic = misc_cfg_h.generate_file(config)
  File "/home/tom/work/acrn-hypervisor/misc/acrn-config/board_config/misc_cfg_h.py", line 217, in generate_file
    sos_cmdlines = [i for i in sos_cmdlines[0].split() if i != '']
IndexError: list index out of range
make[1]: *** No rule to make target 'update_config', needed by '/home/tom/work/acrn-hypervisor/build/hypervisor/.config'.  Stop.
make[1]: Leaving directory '/home/tom/work/acrn-hypervisor/hypervisor'
Makefile:145: recipe for target 'hypervisor' failed
make: *** [hypervisor] Error 2

**Platform**
Platform (Intel x86):  Apollo Lake
System memory: 4G
Storage capabilities:  32G EMMC

**Reason**
misc/vm_configs/xmls/config-xmls/apl-up2-n3350/sdc_launch_1uos_laag.xml  does not have such setting as below:

    <os_config>                                                                    
        <name desc="Specify the OS name of VM, currently it is not referenced by hypervisor code.">ACRN Service OS</name>
        <kern_type desc="Specify the kernel image type so that hypervisor could load it correctly. Currently support KERNEL_BZIMAGE and KERNEL_ZEPHYR.">KERNEL_BZIMAGE</kern_type>
        <kern_mod desc="The tag for kernel image which act as multiboot module, it must exactly match the module tag in GRUB multiboot cmdline.">Linux_bzImage</kern_mod>
        <ramdisk_mod desc="The tag for ramdisk image which act as multiboot module, it must exactly match the module tag in GRUB multiboot cmdline."></ramdisk_mod>
        <bootargs configurable="0" desc="Specify kernel boot arguments">SOS_VM_BOOTARGS</bootargs>
    </os_config> 
    <board_private>                                                                
        <rootfs desc="rootfs for Linux kernel">/dev/mmcblk0p3</rootfs>          
        <bootargs desc="Specify kernel boot arguments">                            
        rw rootwait console=tty0 consoleblank=0 no_timer_check quiet loglevel=3 
        i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x01010F i915.domain_plane_owners=0x011111110000 i915.enable_gvt=1
        hvlog=2M@0xe00000 memmap=0x600000$0xa00000 ramoops.mem_address=0xa00000 ramoops.mem_size=0x400000 ramoops.console_size=0x200000
        reboot_panic=p,w module_blacklist=dwc3_pci i915.enable_guc=0x02 cma=64M@0-
        </bootargs>                                                                
    </board_private> 

while invoke the function err_dic = misc_cfg_h.generate_file(config)--->generate_file(misc_cfg_h.py)----->parse_boot_info

in parse_boot_info，　sos_cmdlines = NULL.  While through this sentence: sos_cmdlines = [i for i in sos_cmdlines[0].split() if i != '']   access sos_cmdlines[0], there is error.

def parse_boot_info():                                                                                                                                                                                                                          
                                                                                
    err_dic = {}                                                                
                                                                                
    if 'SOS_VM' in common.VM_TYPES.values():                                    
        sos_cmdlines = list(common.get_leaf_tag_map(common.SCENARIO_INFO_FILE, "board_private", "bootargs").values())
        sos_rootfs = list(common.get_leaf_tag_map(common.SCENARIO_INFO_FILE, "board_private", "rootfs").values())
        (err_dic, vuart0_dic, vuart1_dic) = scenario_cfg_lib.get_sos_vuart_settings(launch_flag=False)
    else:                                                                       
        sos_cmdlines = list(common.get_leaf_tag_map(common.SCENARIO_INFO_FILE, "os_config", "bootargs").values())
                                                                                
        sos_rootfs = list(common.get_leaf_tag_map(common.SCENARIO_INFO_FILE, "os_config", "rootfs").values())
        (err_dic, vuart0_dic, vuart1_dic) = scenario_cfg_lib.get_sos_vuart_settings(launch_flag=False)
                                                                                
    return (err_dic, sos_cmdlines, sos_rootfs, vuart0_dic, vuart1_dic)

tomwei666, 2020-11-03:
My platform information:
Platform (Intel x86): Apollo Lake
System memory: 4G
Storage capabilities: 32G EMMC

UP2 - N3350

tomwei666, 2020-11-03:
My platform information:
Platform (Intel x86): Apollo Lake
System memory: 4G
Storage capabilities: 32G EMMC
Product/Kit Name: UP2 - N3350

I want to compile SDC scenario for platform.

But I found two setting xml in  apl-up2 and apl-up2-n3350 dir:
1)make all BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/apl-up2-n3350.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/apl-up2-n3350/sdc_launch_1uos_laag.xml
2)make all BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/apl-up2.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/apl-up2/sdc.xml

Which method to choose for my platform?

--

gvancuts, 2020-11-04:
Hi @tomwei666 , this is the second option that you really want. The `sdc_launch_1uos_laag.xml` is not a scenario file.

We have actually removed a lot of the `sdc.xml` files (which is why you were likely having a hard time locating it). I'd recommend you use the industry scenario instead as it's essentially the same but with more User VMs.

If you want to tweak that later, you can edit some of these files directly (hard way) or use the [ACRN configuration tool](https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html).

tomwei666, 2020-11-10:
OK, We talk on the another issue!

closed 2020-11-10 by tomwei666