projectacrn/acrn-hypervisor #5478: SBL firmware update error

tomwei666, 2020-11-06:
**Desccribe the bug**
There is an old sbl in my up2 board!
I compile the sbl and update throug the USB.

Compiling the sbl steps:
1. export SBL_KEY_DIR=/tmp/sbl_key_dir/
2. python3.6 BootloaderCorePkg/Tools/GenerateKeys.py -k $SBL_KEY_DIR
3. python3.6 BuildLoader.py build apl
4. python3.6 Platform/ApollolakeBoardPkg/Script/StitchLoader.py -b sbl.bios.bin -i UPA1AM52.bin -s Outputs/apl/Stitch_Components.zip -o up2_sbl.bin -p 0xAA00000E
ps: My board : up2-N3350, so I download the UPA1AM52.zip BIOS firmware.

Update the sbl firmware steps:
1. Copy up2_sbl.bin to the usb root dir, and rename up2_sbl.bin to FwuImage.bin
2. Enter SBL shell
3. input fwupdate command

The attach update-error.log: update sbl log.
The attach before-update.log: Before updating ,the sbl log.
The sbl in my board version is sbl_up2_ifwi.bin(2019-03-19 20:59) is from acrn daily server.

**Platform**
Product/Kit Name:   UP2-N3350
Memory:  4G
Storage: 32G

tomwei666, 2020-11-06:
[before-upgrade.txt](https://github.com/projectacrn/acrn-hypervisor/files/5498446/before-upgrade.txt)
[update-error.log](https://github.com/projectacrn/acrn-hypervisor/files/5498447/update-error.log)

tomwei666, 2020-11-06:
[before-upgrade.txt](https://github.com/projectacrn/acrn-hypervisor/files/5498713/before-upgrade.txt)
[update-error.log](https://github.com/projectacrn/acrn-hypervisor/files/5498714/update-error.log)

tomwei666, 2020-11-06:
This shoud report to the SBL issue, So I closed this bug!

terryzouhao, 2020-11-09:
@tomwei666 , actually from slimboot guide: https://slimbootloader.github.io/supported-hardware/up2.html#prerequisites
the 'up2_sbl.bin to FwuImage.bin'  you built from Slimboot source and stitched with APL BIOS (UPA1AM52.bin)  in step4, was full binary/bin and recommended to flash on UP2 board with SPI programmer. 
Step4: python3.6 Platform/ApollolakeBoardPkg/Script/StitchLoader.py -b **sbl.bios.bin** -i UPA1AM52.bin -s Outputs/apl/Stitch_Components.zip -o up2_sbl.bin -p 0xAA00000E

If you want to update slime boot with USB disc (already had workable sbl), you need generate CapsuleFirmware with below command:  python BootloaderCorePkg/Tools/GenCapsuleFirmware.py -k $SBL_KEY_DIR/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin -b bios.bin
Remember to generate IAS image for fastboot support also, Step 3: Generate IAS image containing fastboot.elf

You can find the detailed GUIDE of "Boot ACRN with SBL here ": https://slimbootloader.github.io/how-tos/boot-acrn.html

sbl_up2_ifwi.bin:  The binary file to be flashed by BIOS flash tool or SPI programmer
FwuImage.bin:    The binary file to be copied to USB flash drive to upgrade SBL from shell command

tomwei666, 2020-11-09:
./iasimage create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf

How to generate the file of TestSigningPrivateKey.pem!

tomwei666, 2020-11-09:
./iasimage create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf

How to generate the file of TestSigningPrivateKey.pem?

tomwei666, 2020-11-09:
Hi,
 If want to BIOS flash tool to update the SBL firmware, it need to use uefi shell. But my scenario is sbl in up2 already, so cannot use this method!
Brs,
Tom

terryzouhao, 2020-11-09:
> ./iasimage create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf
> 
> How to generate the file of TestSigningPrivateKey.pem?

Yes, There is one commit https://github.com/slimbootloader/slimbootloader/pull/757 that removed 'Keys' folder under Tools. but you can get such keys from older version (before Jun-20) in 'master' or switch to 'Testing' branch.

terryzouhao, 2020-11-09:
> Hi,
> If want to BIOS flash tool to update the SBL firmware, it need to use uefi shell. But my scenario is sbl in up2 already, so cannot use this method!
> Brs,
> Tom

If first time to update SBL, you can do it from UEFI BIOS shell interface.
If you want to update SBL firmware with newer versions on UP2:
Step 1: Copy '**FwuImage.bin**' onto USB flash drive formatted in FAT32
Step 2: Reset UP2 and press any key to enter shell
Run:
Shell>
Shell> fwupdate
HECI SecMode 0
Group    =00000020
Command  =00000007
IsRespone=00000001
Result   =00000000

BUT,  It is highly recommended to have an SPI programmer in case UP2 is bricked after firmware update.

tomwei666, 2020-11-10:
Hi,

I want to generate IAS image for fastboot support , I followed the link: https://slimbootloader.github.io/how-tos/boot-acrn.html

The steps I do:
1）mkdir -p Platform/ApollolakeBoardPkg/SpiIasBin
2）touch cmdline.txt
3）chmod 755 ./iasimage
4）sudo pip3 install cryptography==2.2.2 --target=/home/tom/work/slimbootloader
5） ./iasimage create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf
note: In SBL master branch ,there is no dir: BootloaderCorePkg/Tools/Keys/. I copy this dir from Testing branch. I compile the code in master branch.

python3.6 BuildLoader.py build apl
python3.6 Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i UPA1AM40.bin -s Outputs/apl/Stitch_Components.zip -o sbl_up2_ifwi.bin -p 0xAA00000E -b bios.bin
python3.6 BootloaderCorePkg/Tools/GenCapsuleFirmware.py -k $SBL_KEY_DIR/FirmwareUpdateTestKey_Priv_RSA2048.pem -o FwuImage.bin -p BIOS bios.bin
When I update FwuImage using USB, I got these error log in attach file of update-error-20201110.txt

[update-error-20201110.txt](https://github.com/projectacrn/acrn-hypervisor/files/5514517/update-error-20201110.txt)

terryzouhao, 2020-11-10:
@tomwei666 eventually we found a APL UP2 board, and tried your upgrade scenario.
1. yes, current SlimBoot 'Master' and 'Testing' had updated the source code, e.g., deleted 'Keys' folder in Master, and cannot just copy such folder from 'Testing' branch. we even tried build 'Testing' branch, same flash failure/error found. 

2. APL UP2 mainly supported ACRN1.0 acrh, and new features had switched to KBL/WHL/TGL. We didn't try SlimBoot 20' upgrades/versions. So you can try legacy 19' version of slimboot, actually we just verified below version and successfully flashed SBL. Please find detailed steps and let us know if you have any further questions:
**Connect serial adapter between UP2 and the host, enter Shell from host to 'Shell> fwupdate' with 'FwuImage.bin' in USB disck.**
**commit f4f16878abf1162fe22d22ad08c22bb1633a86e1 (HEAD)**
Author: Aiden Park <aiden.park@intel.com>
Date:   Mon Sep 9 14:58:04 2019 -0700

CMD history:
2027  cd $workspace/slimbootloader/
2114  git clean -dfx && git reset --hard && git checkout f4f16878abf1162fe22d22ad08c22bb1633a86e1
2117  cp -r ../iasimage.py ./
2118  cp -r ../fastboot.elf ./
2119  cp -r ../UPA1AM40.zip ./
2120  unzip UPA1AM40.zip
2121  chmod 755 ./iasimage.py
2122  mkdir -p Platform/ApollolakeBoardPkg/SpiIasBin
2123  touch cmdline.txt
2124  chmod 755 ./iasimage.py
2125  ./iasimage.py create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf
2126  python BuildLoader.py build apl
2127  python Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i UPA1AM40.bin -s Outputs/apl/Stitch_Components.zip -o sbl_up2_ifwi.bin -p 0xAA00000E -b bios.bin
2129  python BootloaderCorePkg/Tools/GenCapsuleFirmware.py -k ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -o FwuImage.bin -p BIOS bios.bin
2132  cp -r ./FwuImage.bin /media/ssg-hv/CB62-18DB/FwuImage.bin
2133  sync
2134  umount /dev/sdc1

tomwei666, 2020-11-10:
Shell> fwupdate
HECI SecMode 0
Group    =00000020
Command  =00000007
IsRespone=00000001
Result   =00000000
BtGuard: VB : 0, MB : 0 

============= Intel Slim Bootloader STAGE1A =============
SBID: SB_APLI 
ISVN: 001
IVER: 000.005.001.000.06157
SVER: EDC112328CF3E414
FDBG: BLD(D) FSP(R)
FSPV: ID($APLFSP$) REV(01040301)
Loader global data @ 0xFEF01D54
Run  STAGE1A @ 0xFEF80000
Load STAGE1B @ 0xFEF88000
No BtGuard verification !

============= Intel Slim Bootloader STAGE1B =============
CPU : APL-B1 [2C @ 1100MHz: premium SKU], uCode rev.32
Early GpioInit for EMMC
MMC global data init
Use SDMA instead of ADMA2
MMC Phase 1 init
SpiInstance = FEF034A0
Load EXT CFG Data @ 0xFEF03FC4:0x0C30 ... Success
HASH Verification Success! Component Type (4)
RSA Verification Success!
BOOT: BP0 
MODE: 0
BoardID: 0x0E
PlatformName: UP2
BootPolicy : 0x00000010
Memory Init
Looking for MRC saving data
Read MRC ParamData at 0xFF232000
Decompress ParamData
Read MRC VarData at 0xFF231000
Use slot 57
MRC data was found @ 0xFEF40400(Param) and 0xFEF40000(Var)
Load memory parameters from CfgData.
Load general parameter from CfgData.
Call FspMemoryInit ... Success
Loader global data @ 0x7ABBFD80
Enable Paging ...
Switch to memory stack @ 0x7ABFFF00
Stage1 stack: 0x2000 (0x1684 used)
Stage1 heap: 0x6000 (0x4B70 used)
Call FspTempRamExit ... Success
CSE: FW 3.1.50.2222, VB: 0, MB: 0
CSE: boot dev #2: SPI
  00000000: 43 46 47 44 10 00 00 00-34 00 00 00 80 00 00 00  *CFGD....4.......*
  00000010: 25 00 01 08 FF FF FF FF-05 00 00 00 00 00 00 02  *%...............*
  00000020: 46 77 75 49 6D 61 67 65-2E 62 69 6E 00 00 00 00  *FwuImage.bin....*
  00000030: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000040: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000050: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000060: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000070: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
MRC: [v0.56.89.20]
Memory FSP @ 0x7AC00000
Memory TOP @ 0x7A074000
HASH Verification Success! Component Type (1)
Loaded STAGE2 @ 0x7AB32000

============= Intel Slim Bootloader STAGE2 =============
Disable Paging ...
245 GPIO entries are populated from Base platform
[... 111 lines cut ...]
  00000090: 09 00 00 00 00 00 00 00-00 00 00 00 0A 00 00 00  *................*
  000000A0: 00 02 00 00 10 00 00 00-05 00 00 00 00 80 14 00  *................*
  000000B0: 00 10 14 00 01 00 00 00-00 50 00 00 00 A0 00 00  *.........P......*
  000000C0: 0C 00 00 00 00 00 00 00-00 00 00 00 0D 00 00 00  *................*
  000000D0: 00 00 00 00 00 00 00 00-02 00 00 00 00 F0 01 00  *................*
  000000E0: 00 10 06 00 11 00 00 00-10 02 00 00 08 01 00 00  *................*
  000000F0: 00 00 00 00 00 10 00 00-00 40 00 00 0E 00 00 00  *.........@......*
CapsuleImage: 0x787AF000, CapsuleSize: 0xEFE280
HASH Verification Success! Component Type (5)
RSA Verification Success!
The new BOOTLOADER image passed verification
BPDT header Signature or partition number verification failed.
SetBootPartitionInvalid failed! Status = 0x80000003
GetFirmwareUpdateInfo, Status = 0x80000003
Firmware update failed with Status = Unsupported
Firmware update Done! clear CSE flag to normal boot mode.
HECI SecMode 0
Group    =00000020
Command  =00000007
IsRespone=00000001
Result   =00000000
Rebooting ...
BtGuard: VB : 0, MB : 0 

yet another error!

terryzouhao, 2020-11-12:
@tomwei666 , we were also using the same N3350 APL-UP2 board (Intel® Celeron™ N3350 (up to 2.4 GHz) ). 
Just re-tried another commit #4803d94 (Sep 10, 2019 ), one day earlier than the previous one, all passed and flashed to N3350 again with enclosed log.  
Actually just guess your failure "BPDT header Signature or partition number verification failed" was still because of the different "Priv key":
./iasimage.py create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf
python BootloaderCorePkg/Tools/GenCapsuleFirmware.py -k ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem  -o FwuImage.bin -p BIOS bios.bin

Please try with # git checkout 4803d940b1e9e2eff9a9bac28e936b03d02fb246 commit, and refer to our successful log: [N3350_UP2_sbl_upgrade_sucess_log.txt] (https://github.com/projectacrn/acrn-hypervisor/files/5528388/N3350_UP2_sbl_upgrade_sucess_log.txt)

Detailed steps for your reference again:
cd slimbootloader/
git clean -dfx
git reset --hard
git checkout 4803d940b1e9e2eff9a9bac28e936b03d02fb246
cp -r ../UPA1AM40.zip .
cp -r ../iasimage.py .
cp -r ../fastboot.elf .
unzip UPA1AM40.zip
chmod +x ./iasimage.py
mkdir -p Platform/ApollolakeBoardPkg/SpiIasBin
touch cmdline.txt
./iasimage.py create -o Platform/ApollolakeBoardPkg/SpiIasBin/iasimage1.bin -d ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem -i 0x40000 cmdline.txt fastboot.elf
python BuildLoader.py build apl
python Platform/ApollolakeBoardPkg/Script/StitchLoader.py -i UPA1AM40.bin -s Outputs/apl/Stitch_Components.zip -o sbl_up2_ifwi.bin -p 0xAA00000E -b bios.bin
python BootloaderCorePkg/Tools/GenCapsuleFirmware.py -k ./BootloaderCorePkg/Tools/Keys/TestSigningPrivateKey.pem  -o FwuImage.bin -p BIOS bios.bin
cp -r FwuImage.bin /media/ssg-hv/CB62-18DB/
sync
umount /dev/sdc1
Then insert the Udisk into UP2 board and start upgrading SBL via “fwupdate” cmd.  You will see the sbl update log as the given attachment.

closed 2020-11-20 by tomwei666