projectacrn/acrn-hypervisor #5480: System-crash

tomwei666, 2020-11-06:
**Desccribe the bug**
I create sos_boot.img,sos_rootfs.img  and up2_laag.img using create-up2-images.sh.
Flashing these images using Platform flash Toollite.

Is my sbl version too old?
The sbl in my board version is sbl_up2_ifwi.bin(2019-03-19 20:59) is from acrn daily server.
[system-crash-error.txt](https://github.com/projectacrn/acrn-hypervisor/files/5498617/system-crash-error.txt)

Steps:
./create-up2-images.sh --images-type all --clearlinux-version 31030 --laag-json uos.json --acrn-code-path=/work/acrn-hypervisor

The attach system-crash-error.txt: Aftering flashing, reboot system.

**Platform**
Product/Kit Name:   UP2-N3350
Memory:  4G
Storage: 32G

gvancuts, 2020-11-06:
Hi @tomwei666 , it's more likely the version of ACRN that's too new. I don't believe we test this configuration anymore, but let's try to get you up and running. Can you try the following tests?
1. Create the image using `./create-up2-images.sh --images-type all --clearlinux-version 31030 --laag-json uos.json` (i.e. leaving out the `--acrn-code-path` argument)
2. Create the image using the same command you used before but **after** checking out the `2019w37.4-140000p` tag in your ACRN repo (`acrn-hypervisor`). Make sure to delete any existing `hypervisor/build` folder if it exists.

There are a couple of things that could be happening:
1. I don't know if the latest version of ACRN can run on UP2 with SBL
2. **But**, even if it can, we'd need to fix the `create-up2-images.sh` script when it uses the `--acrn-code-path` argument as it uses a build command (see [this section](https://github.com/projectacrn/acrn-hypervisor/blob/master/doc/tutorials/create-up2-images.sh#L54-L59)) that is no longer valid. I'm actually wondering what it's using when you run the script as that command should fail with the latest codebase and the `acrn.32.out` shouldn't be created. Could you possibly have one that was built in the past still lying around (under `hypervisor/build/`)?

Thanks!
Geoffroy

tomwei666, 2020-11-07:
Dear Geoffroy:

Sorry,  I do not descripe the steps clearly, it may confuse you.

The steps below:

1. git clone the new acrn code
2. compile like this: 
make all BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/apl-up2-n3350.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/apl-up2-n3350/logical_partition.xml
3. I copy acrn.32.out to the assigned dir such as: /A/B
4. I modify the file of create-up2-images.sh

 54     if [ ${ACRN_HV_CODE_PATH} ]                                                 
 55     then                                                                        
 56         make -C ${ACRN_HV_CODE_PATH} clean || return 1                          
 57         make -C ${ACRN_HV_CODE_PATH} hypervisor BOARD=apl-up2 FIRMWARE=sbl || return 1
 58         ACRN_SBL=${ACRN_HV_CODE_PATH}/build/hypervisor/acrn.32.out                                                                                                          
 59     fi   

========>

 54     if [ ${ACRN_HV_CODE_PATH} ]                                                 
 55     then                                                                        
 56         make -C ${ACRN_HV_CODE_PATH} clean || return 1                          
 57         make -C ${ACRN_HV_CODE_PATH} hypervisor BOARD=apl-up2 FIRMWARE=sbl || return 1
 58         ACRN_SBL=${ACRN_HV_CODE_PATH}/build/hypervisor/acrn.32.out                                                                                                          
 59     fi   
60     ACRN_SBL=/A/B/acrn.32.out 

5. Then create sos_boot.img and up2_laag.img.
./create-up2-images.sh --images-type all --clearlinux-version 33600 --laag-json uos.json --acrn-code-path=/work/acrn-hypervisor

Question:

I find acpi bin in the build dir:
tom@ubuntu:~/work/acrn-hypervisor/build/hypervisor/acpi$ ls
ACPI_VM0.bin  ACPI_VM1.bin  VM0  VM1

I think shoud use iasimage command to stitch the acpi bin to the images, I should modify the create-up2-images.sh.

In the create-up2-images.sh, there is such command,does not find acpi bin.

iasimage create -o iasImage -i 0x40300 -d tmp/bxt_dbg_priv_key.pem -p 4 tmp/hv_cmdline ${ACRN_SBL} tmp/linux.txt ${SOS_KERNEL} ||

Brs,
Tom

tomwei666, 2020-11-08:
Hi,
The build/hypervisor files show as below:

pateo@pateo-Latitude-5400:~/work/acrn/acrn-hypervisor/build/hypervisor$ ls
acpi  acpi_parser  acrn.32.out  acrn.bin  acrn.map  acrn.out  arch  boot  cfg.log  common  debug  dm  home  hv_prebuild_check.out  hw  include  lib  link_ram.ld  modules

Brs,
Tom

gvancuts, 2020-11-09:
Hi @tomwei666 , thanks for these additional details, now I understand this a little better :-)

Out of curiosity, is there any reason why you did not use the `--acrn-sbl-path` option of the script instead of modifying it? Did it not work as expected?

The scenario you've used is potentially problematic is it defines two VMs that are started by the hypervisor directly. What we call `laag` here is a Linux-based VM that is started by the `acrn-dm` (ACRN Device Manager) so it's a bit different. Is your requirement to use that scenario specifically? There are advantages and disadvantages to all configurations so it's important to analyse your requirements so as to use the best match for your application.

The last point is the ACRN version used, I don't know if we're still testing the latest version with SBL on UP2, I suspect we may not... @NanlinXie , do you know?

tomwei666, 2020-11-10:
Hi @gvancuts ,  My requirement scenario is SDC.  

But in the dir ,misc/vm_configs/xmls/config-xmls/apl-up2-n3350 does not have xml for SDC senario.

tom@ubuntu:~/work/acrn-hypervisor/misc/vm_configs/xmls/config-xmls/apl-up2-n3350$ ls
logical_partition.xml  sdc_launch_1uos_laag.xml

So, I compile logicat_partition instead!

Question:
Can you create a SDC XML for apl-up2-n33350 board?

As I know from acrn webchat group , the latest verion acrn does not be tested with SBL on UP2.

tomwei666, 2020-11-10:
Hi,
 
  I create sdc.xml for apl-up2-n3350 base on the sdc.xml of apl-up2 board using acrn config tools.

  It can successfully run acrn, but there is a little error log.

Brs,
Tom

gvancuts, 2020-11-16:
Hi @tomwei666 , can you share the error logs?

closed 2020-11-20 by tomwei666