projectacrn/acrn-hypervisor #5510: Can't create the boot file with a new compilation of acrn hypervisor

ionutnechita, 2020-11-13:
Hi ACRN team,

I tried to make a new compile, but I can't.
I notice some problems with an object (`sos_pci_devs') and ld issue.
Error is this:
**_/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/hypervisor/modules/boot_mod.a(cpu_primary.o): warning: relocation in read-only section `multiboot_header'
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/hypervisor/modules/hw_mod.a(vm_configurations.o):(.data.rel.vm_configs+0x80): undefined reference to `sos_pci_devs'
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:462: /tmp/acrn-hypervisor/build/hypervisor/acrn.out] Error 1
make[1]: Leaving directory '/tmp/acrn-hypervisor/hypervisor'
make: *** [Makefile:148: hypervisor] Error 2_**

Development machine:
_cat /etc/os-release
NAME="openSUSE Tumbleweed"
# VERSION="20201111"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20201111"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20201111"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo"

ldd --version
ldd (GNU libc) 2.32
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

gcc --version
gcc (SUSE Linux) 10.2.1 20201028 [revision a78cd759754c92cecbf235ac9b447dcdff6c6e2f]
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

iasl -v
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20200717
Copyright (c) 2000 - 2020 Intel Corporation_

commit 49a773ab54bbcf99462319442068dad973cc2513 (HEAD -> master, origin/master, origin/HEAD)
Author: Yonghua Huang <yonghua.huang@intel.com>
Date:   Sun Nov 8 10:46:59 2020 +0800

    hv: bugfix for hv-emulated device de-init

     - fix bug in 'hcall_destroy_vdev()', the availability of
       vpci device shall be checked on 'target_vm".

     - refine 'vpci_update_one_vbar()' to avoid potential NULL
       pointer access.

    Tracked-On: #5490
    Signed-off-by: Yonghua Huang <yonghua.huang@intel.com>

commit 68dc8d9f8f919b480f1b4319dea19ca4b8ecc499
Author: Peter Fang <peter.fang@intel.com>
Date:   Thu Nov 5 17:40:54 2020 -0800

    hv: pm: avoid duplicate shutdowns on RTVM

    It is possible for more than one vCPUs to trigger shutdown on an RTVM.
    We need to avoid entering VM_READY_TO_POWEROFF state again after the
    RTVM has been paused or shut down.

    Also, make sure an RTVM enters VM_READY_TO_POWEROFF state before it can
    be paused.

    v1 -> v2:
    - rename to poweroff_if_rt_vm for better clarity

    Tracked-On: #5411
    Signed-off-by: Peter Fang <peter.fang@intel.com>

make all BOARD=nuc7i7dnb SCENARIO=industry RELEASE=0 2>&1 | tee -a log_compile1.txt
_**hw_mod.a(vm_configurations.o):(.data.rel.vm_configs+0x80): undefined reference to `sos_pci_devs'**_

[log_compile1.txt](https://github.com/projectacrn/acrn-hypervisor/files/5538410/log_compile1.txt)

yonghuah, 2020-11-14:
Hi Nechita,

   Can you share your build/compile  command  for this?

commit 49a773a<https://github.com/projectacrn/acrn-hypervisor/commit/49a773ab54bbcf99462319442068dad973cc2513> (HEAD -> master, origin/master, origin/HEAD)
Author: Yonghua Huang yonghua.huang@intel.com<mailto:yonghua.huang@intel.com>

commit 68dc8d9<https://github.com/projectacrn/acrn-hypervisor/commit/68dc8d9f8f919b480f1b4319dea19ca4b8ecc499>
Author: Peter Fang peter.fang@intel.com<mailto:peter.fang@intel.com>

ionutnechita, 2020-11-14:
This command is used:
make all BOARD=nuc7i7dnb SCENARIO=industry RELEASE=0 2>&1 | tee -a log_compile1.txt

yonghuah, 2020-11-14:
Please use xml config files as input, example:
make clean && make BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/nuc7i7dnb.xml \
SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/nuc7i7dnb/industry.xml FIRMWARE=uefi

Seems that global variable is not defined by default and shall be generated by configtool,  will double check that,  you can walk around it by this way.

Thanks.
-Yonghua

ionutnechita, 2020-11-14:
Hi @yonghuah,

I recompiled again with the parameters sent by you, and the issue is reproduced.
(make clean && make BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/nuc7i7dnb.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/nuc7i7dnb/industry.xml FIRMWARE=uefi)

...
hypervisor/modules/librelease.a /tmp/acrn-hypervisor/build/hypervisor/modules/sys_init_mod.a -Wl,--end-group
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/hypervisor/modules/boot_mod.a(cpu_primary.o): warning: relocation in read-only section `multiboot_header'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/hypervisor/modules/hw_mod.a(vm_configurations.o):(.data.rel.vm_configs+0x80): undefined reference to `sos_pci_devs'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:462: /tmp/acrn-hypervisor/build/hypervisor/acrn.out] Error 1
make[1]: Leaving directory '/tmp/acrn-hypervisor/hypervisor'
make: *** [Makefile:148: hypervisor] Error 2

I looked through the code and saw a conflict between two declared structures ( in files pci_dev.h and vm_configurations.c ).
Issue would be here:
1. pci_dev.h ( ./hypervisor/include/arch/x86/pci_dev.h ) 
    - extern struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];
2. vm_configurations.c ( ./misc/vm_configs/scenarios/hybrid_rt/vm_configurations.c )
    - extern struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];
3. vm_configurations.c ( ./misc/vm_configs/scenarios/hybrid/vm_configurations.c )
    - extern struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM]
4. vm_configurations.c ( ./misc/vm_configs/scenarios/industry/vm_configurations.c )
    - extern struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];

A regression could have crept in between versions.
Or the problem is new and has not been seen so far.

To get over this issue, I made the change in the vm_configurations.c files.

**_extern_** struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM]; 
--> struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];

gvancuts, 2020-11-19:
I ran some quick experiments on my environment: I can reproduce the problem when using `make clean && make all BOARD=nuc7i7dnb SCENARIO=industry RELEASE=0`.

I then tried the other command (minus the `FIRMWARE=uefi` since this is obsolete), and that worked: `make clean && make BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/nuc7i7dnb.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/nuc7i7dnb/industry.xml`

I noticed though that some files are modified by this latest operation:
```
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   misc/vm_configs/scenarios/industry/nuc7i7dnb/pci_dev.c
	modified:   misc/vm_configs/scenarios/industry/nuc7i7dnb/vbar_base.h
	modified:   misc/vm_configs/scenarios/industry/vm_configurations.c
```

If I re-run the first command (i.e. `make clean && make all BOARD=nuc7i7dnb SCENARIO=industry RELEASE=0`) on that modified tree, the build is successful.

Here is the diff:
```
$ git diff
diff --git a/misc/vm_configs/scenarios/industry/nuc7i7dnb/pci_dev.c b/misc/vm_configs/scenarios/industry/nuc7i7dnb/pci_dev.c
index 79033ec6b..90b1a6138 100644
--- a/misc/vm_configs/scenarios/industry/nuc7i7dnb/pci_dev.c
+++ b/misc/vm_configs/scenarios/industry/nuc7i7dnb/pci_dev.c
@@ -10,3 +10,5 @@
 #include <vbar_base.h>
 #include <mmu.h>
 #include <page.h>
+
+struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];
diff --git a/misc/vm_configs/scenarios/industry/nuc7i7dnb/vbar_base.h b/misc/vm_configs/scenarios/industry/nuc7i7dnb/vbar_base.h
index 9077629d8..8045a3194 100644
--- a/misc/vm_configs/scenarios/industry/nuc7i7dnb/vbar_base.h
+++ b/misc/vm_configs/scenarios/industry/nuc7i7dnb/vbar_base.h
@@ -7,4 +7,38 @@
 #ifndef VBAR_BASE_H_
 #define VBAR_BASE_H_
 
+#define VGA_COMPATIBLE_CONTROLLER_0_VBAR              .vbar_base[0] = 0xde000000UL, \
+                                                      .vbar_base[2] = 0xc0000000UL
+
+#define SYSTEM_PERIPHERAL_0_VBAR                      .vbar_base[0] = 0xdf252000UL
+
+#define USB_CONTROLLER_0_VBAR                         .vbar_base[0] = 0xdf230000UL
+
+#define SIGNAL_PROCESSING_CONTROLLER_0_VBAR           .vbar_base[0] = 0xdf251000UL
+
+#define SIGNAL_PROCESSING_CONTROLLER_1_VBAR           .vbar_base[0] = 0xdf250000UL
+
+#define SIGNAL_PROCESSING_CONTROLLER_2_VBAR           .vbar_base[0] = 0xdf24f000UL
+
+#define COMMUNICATION_CONTROLLER_0_VBAR               .vbar_base[0] = 0xdf24e000UL
+
+#define SERIAL_CONTROLLER_0_VBAR                      .vbar_base[1] = 0xdf24d000UL
+
+#define SATA_CONTROLLER_0_VBAR                        .vbar_base[0] = 0xdf248000UL, \
+                                                      .vbar_base[1] = 0xdf24c000UL, \
+                                                      .vbar_base[5] = 0xdf24b000UL
+
+#define MEMORY_CONTROLLER_0_VBAR                      .vbar_base[0] = 0xdf244000UL
+
+#define AUDIO_DEVICE_0_VBAR                           .vbar_base[0] = 0xdf240000UL, \
+                                                      .vbar_base[4] = 0xdf220000UL
+
+#define SMBUS_0_VBAR                                  .vbar_base[0] = 0xdf24a000UL
+
+#define ETHERNET_CONTROLLER_0_VBAR                    .vbar_base[0] = 0xdf200000UL
+
+#define NETWORK_CONTROLLER_0_VBAR                     .vbar_base[0] = 0xdf100000UL
+
+#define NON_VOLATILE_MEMORY_CONTROLLER_0_VBAR         .vbar_base[0] = 0xdf000000UL
+
 #endif /* VBAR_BASE_H_ */
diff --git a/misc/vm_configs/scenarios/industry/vm_configurations.c b/misc/vm_configs/scenarios/industry/vm_configurations.c
index af8483261..9edd0fc08 100644
--- a/misc/vm_configs/scenarios/industry/vm_configurations.c
+++ b/misc/vm_configs/scenarios/industry/vm_configurations.c
@@ -155,6 +155,7 @@ struct acrn_vm_config vm_configs[CONFIG_MAX_VM_NUM] = {
                .vuart[0] = {
                        .type = VUART_LEGACY_PIO,
                        .addr.port_base = INVALID_COM_BASE,
+                       .irq = COM1_IRQ,
                },
                .vuart[1] = {
                        .type = VUART_LEGACY_PIO,
(END)

```

ionutnechita, 2020-11-19:
Ok, understood.
This is modification for vUART, no?
Thanks for check.

ionutnechita, 2020-11-24:
Hi @gvancuts,

This issue it's fixed.
Thanks.

This issue is fixed:
hypervisor/modules/librelease.a /tmp/acrn-hypervisor/build/hypervisor/modules/sys_init_mod.a -Wl,--end-group
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/hypervisor/modules/boot_mod.a(cpu_primary.o): warning: relocation in read-only section multiboot_header' /usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/hypervisor/modules/hw_mod.a(vm_configurations.o):(.data.rel.vm_configs+0x80): undefined reference to sos_pci_devs'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:462: /tmp/acrn-hypervisor/build/hypervisor/acrn.out] Error 1
make[1]: Leaving directory '/tmp/acrn-hypervisor/hypervisor'
make: *** [Makefile:148: hypervisor] Error 2

gvancuts, 2020-11-25:
> This issue is fixed:

Thanks... closing!

closed 2020-11-25 by gvancuts