projectacrn/acrn-hypervisor #5511: [Documentation] Build ACRN with OpenSUSE Tumbleweed

ionutnechita, 2020-11-14:
Hi ACRN team,

OpenSUSE Tumbleweed - The makers' choice for sysadmins, developers and desktop users.
https://www.opensuse.org/#Tumbleweed
Any user who wishes to have the newest packages that include, but are not limited to, the Linux kernel, SAMBA, git, desktops, office applications and many other packages, will want Tumbleweed. Tumbleweed appeals to Power Users, Software Developers and openSUSE Contributors. If you require the latest software stacks and Integrated Development Environment or need a stable platform closest to bleeding edge Linux, Tumbleweed is the best choice for you.

Is one of the most powerful operating systems, we should make a page for it too.
New page should contain steps to create a new ACRN image with OpenSUSE Tumbleweed.

I could also help with the documentation.

Thanks to the whole team.

gvancuts, 2020-11-16:
Hi @ionutnechita ,

Thanks for your request! The team is concentrating on Ubuntu and the Yocto Project at the moment. I'm not, personally, convinced we need to add Tumbleweed to the mix. More importantly, I don't believe the current ACRN team has the resources to do that... so it's also a great area for contribution! ;-)

Back to the reason I saw I'm not convinced we need to add this to the mix is based on the fact that the target segments for ACRN are IoT, Edge, embedded devices, and especially those where hard real-time and functional safety environments are required. I haven't seen many using Tumbleweed in that context, but happy to be proven wrong! What is your use-case by the way? Perhaps this is where I'm not seeing the immediate link between Tumbleweed and ACRN.

Unless you plan to run Tumbleweed as the Service VM OS, there is little reason to spend too much time making sure we can work with the tooling included in the distro. If Tumbleweed is to be used as a developer workstation, then we still need to ensure that we're compiling ACRN for the actual target environment. You can often re-use binaries across distros but I feel the "cleanest" way would still be to build ACRN with the tools, libraries, environment from the target OS. This is where using a container may be a useful approach.

Having said that, I saw your issue on the mailing list, once we're passed that. If you'd like to contribute some documentation on how to use Tumbleweed, we'd be happy to take it!

ionutnechita, 2021-01-23:
Hi @gvancuts,

I started working on the documentation to be able to build the whole environment in OpenSUSE Tumbleweed.

At the moment I have the building part in the docker.
https://github.com/ionutnechita/acrn-dockerfiles/blob/master-ionutnechita/Dockerfile.tumbleweed

If you have time to create a page in the official documentation for OpenSUSE Tumbleweed with ACRN build.

Thanks.

gvancuts, 2021-02-04:
Thanks @ionutnechita ! I took a quick look at your `Dockerfile`, it looks good... I will take some of the latest patches you made and merge them in my own ;-) (note: you don't need `kconfiglib` with the latest ACRN codebase. You may want to keep it if you want to compile existing releases but otherwise you can remove this)

We could add a section in this doc [here](https://projectacrn.github.io/latest/getting-started/building-from-source.html#install-build-tools-and-dependencies) to list the tools/build dependencies in Tumbleweed. Is there anything more than that to get Tumbleweed up and running as a Service VM OS? If not, I would prefer **not** to create an additional user guide for it, I'd rather add the section as described above.

amyreye, 2022-02-17:
@gvancuts What is the status?

ionutnechita, 2022-02-18:
Hi.
I will try to test and create some of the steps with the latest version of ACRN.
I'm going to need time to find the hardware and create the ACRN configuration.
I would like this ticket to remain open.