projectacrn/acrn-hypervisor #5512: [Documentation] Create and boot ACRN SOS with rootfs based on OpenSUSE Tumbleweed

ionutnechita, 2020-11-14:
Hi ACRN team,

OpenSUSE Tumbleweed - The makers' choice for sysadmins, developers and desktop users.
https://www.opensuse.org/#Tumbleweed
Any user who wishes to have the newest packages that include, but are not limited to, the Linux kernel, SAMBA, git, desktops, office applications and many other packages, will want Tumbleweed. Tumbleweed appeals to Power Users, Software Developers and openSUSE Contributors. If you require the latest software stacks and Integrated Development Environment or need a stable platform closest to bleeding edge Linux, Tumbleweed is the best choice for you.

Is one of the most powerful operating systems, we should make a page for it too.
New page should contain steps to create a new SOS with rootfs based on OpenSUSE Tumbleweed based.

This would bring long-term benefits to the community.

Thanks to the whole team.

gvancuts, 2020-11-16:
Hi @ionutnechita , would you be willing to create this tutorial? We'd be happy to review and take this!

The skeleton of such guide would be very similar to the existing ones, it's about compiling `acrn-hypervisor`, the `acrn-kernel` and installing all the build artefacts in the right place. Does Tumbleweed use Grub? We need multiboot(2) support to boot ACRN.

I am assuming that Tumbleweed is an RPM-based distribution, isn't it? If true, it would be even better to be able to create an RPM package for all the ACRN components and make that available to the Tumbleweed users. If you have expertise in that space and some time, it would be great to see this! ;-)

amyreye, 2022-02-17:
@gvancuts What is the status?

gvancuts, 2022-02-17:
> @gvancuts What is the status?

We never got around to writing it. I think we should drop this requirement unless @ionutnechita is willing to help?

ionutnechita, 2022-02-18:
Hi.
I will try to test and create some of the steps with the latest version of ACRN.
I'm going to need time to find the hardware and create the ACRn configuration.
I would like this ticket to remain open.