projectacrn/acrn-hypervisor #5517: SOS boots with only one CPU core even though two cpu cores are assigned to it in hybrid scenario on TGL-U

vmurthy-intel, 2020-11-16:
**Description of issue**
Even though I have assigned two CPU cores to SOS, after it boots, I can SOS is running with only on CPU core. I have assigned CPU cores 0 and 1 to SOS but I can see after boot that SOS is running with only core 0.

ACRN:\>vcpu_list

VM ID    PCPU ID    VCPU ID    VCPU ROLE    VCPU STATE    THREAD STATE
=====    =======    =======    =========    ==========    ==========
  0         2          0       PRIMARY      Running          RUNNING
  0         3          1       SECONDARY    Running          RUNNING
**1         0          0       PRIMARY      Running          RUNNING**
ACRN:\>

The SOS dmesg shows the following error:

[   35.472057] vhm_dev_open: opening device node
[   35.488865] Failed to destroy VM -1!
[   35.488943] vhm: freed vm
**[   35.502754] smpboot: CPU 1 is now offline**
[   36.731929] vhm: try to offline cpu 1 with lapicid 2

**My observation**:
We have libvirt enabled. The above issue is observed only when libvirt is enabled. When I disable the libvirt and reboot, I can see SOS running with two cpu cores (core 0 and 1) without any issue. Looks like libvirt is forcing to make CPU 1 as offline.

**Platform**: TGL-U RVP board  
**Codebase**: ACRN Release2.2_RC5  
**Scenario**: Hybrid scenario  
  
**Note:**  
1. Here the Pre-launch RT linux is booted from NVMe. (i.e. NVMe is made pass thru to pre-launch RT linux)  
2. Here, libvirt, VM2VM communication, gvt-d for windows are enabled.  
3. AN i210 intel giganet card is made as passthru to pre-launch VM

**To reproduce:**  
1. Use the attached board, scenario and launch XML files to build the ACRN hypervisor  
2. build SOS with libviet enabled.  
3. flash it onto tgl-u rvp board  
4. reboot the target  
5. Do vcpu_list from acrn console and you will see SOS running with only 1 cpu core.  
6. open dmesg logs from SOS and you can see the message that "cpu1 is offline."  

**Expected behavior**
SOS should run with two cpu cores (CPU 0 and CPU1)

**Attachments**  
board, scenario and launch xml files

[tgl-rvp.zip](https://github.com/projectacrn/acrn-hypervisor/files/5549199/tgl-rvp.zip)

wenlingz, 2020-11-18:
@peterfang please give the comments, thanks

peterfang, 2020-11-19:
This is expected behavior.

The current design is: we offline all pCPUs that can be used by a post-launched VM.

We use that to keep accounting of the current load of pCPUs (pCPUs other than PCPU0 can start at load 0). This helps us with CPU placement on a VM.

gvancuts, 2020-11-19:
> 
> The current design is: we offline all pCPUs that can be used by a post-launched VM.

But based on their configuration, that pCPU is not to be used by a post-launched VM, they want it assigned to the Service VM, so that logic is broken.

That logic is really something we ought to fix in all our scripts. Why offlining all pCPUs if the VMs do not yet require them? This is very sub-optimal.

peterfang, 2020-11-19:
We offline all pCPUs (except for pCPU0) that can potentially be used by a post-launched VM, regardless of whether that VM is created the moment libvirtd is launched.

In order to perform pCPU load balancing (one of the requirements of the ACRN driver), the simplest solution is to vacate all the pCPUs that can be potentially used by a post-launched VM and start from 0 load. This is the easiest way to keep track and perform CPU placement.

gvancuts, 2020-11-19:
> We offline all pCPUs (except for pCPU0) that can potentially be used by a post-launched VM, regardless of whether that VM is created the moment libvirtd is launched.
>
> In order to perform pCPU load [...]

What pCPU load balancing are you referring to?

So how does that work in the case where libvirt is not involved?

I understand that the current mechanism is the easiest but it's really sub-optimal, I'd like to think we can do better to get more out of the HW resources available ;-)

peterfang, 2020-11-19:
I'm referring to the "--cpu_affinity" parameter in dm.

When a user uses an xml file to launch a VM in libvirt, CPU placement is usually set to automatic. It is libvirtd's job to find a VM and place its vCPUs on pCPUs. We want to give VMs as much CPU resource as possible.

All of this is performed by libvirtd. So if the libvirt service is disabled, only a launch script will offline the CPUs.

I agree it's sub-optimal at the moment. We can consider reworking this if there is a strong need to improve this.

vmurthy-intel, 2020-11-26:
@peterfang one quick question: So you mean, libvirtd, if it is enabled, will assign minimum pCPUs to SOS (I believe it will always assign only one pCPU, i.e. pCPU 0) and offline other pCPUs for SOS even though u have assigned 2 pCPUs in scenario XML for SOS. Correct ?? Then I believe in this case, libvirt is kind of overriding the pcpu assignment done in the scenario xml.  
  
Also I observed that "cpu offline command" is generated in the launch script only in industry scenario and not in hybrid scenario.
i.e. if you use my attached XML files (hybrid scenario) to generate the launch scripts, it will not have "cpu offline command" generated in it. I have attached the generated launch scripts also with this message.  
  
Please correct me if any of my above understanding is incorrect.  
  
Thanks and Regards,
Vinay
[launch_uos_id1.txt](https://github.com/projectacrn/acrn-hypervisor/files/5602631/launch_uos_id1.txt)
[launch_uos_id2.txt](https://github.com/projectacrn/acrn-hypervisor/files/5602632/launch_uos_id2.txt)

peterfang, 2020-11-27:
> @peterfang one quick question: So you mean, libvirtd, if it is enabled, will assign minimum pCPUs to SOS (I believe it will always assign only one pCPU, i.e. pCPU 0) and offline other pCPUs for SOS [...]

I can only respond to the libvirtd part of your question and the answer is yes. For simplicity's sake, we try to vacate all pCPUs that can be potentially used by a post-launched VM because we want to reduce pCPU sharing between the Service VM and the post-launched VMs.

fuzhongl, 2022-02-25:
Closed since it works as design.

closed 2022-02-25 by fuzhongl