projectacrn/acrn-hypervisor #5522: [Feature] Does the ACRN Hypervisor support OpenvSwitch ( OVS )?

ionutnechita, 2020-11-17:
Hi ACRN team,

OpenvSwitch ( OVS )
https://www.openvswitch.org/
Open vSwitch, sometimes abbreviated as OVS, is an open-source implementation of a distributed virtual multilayer switch. The main purpose of Open vSwitch is to provide a switching stack for hardware virtualization environments, while supporting multiple protocols and standards used in computer networks.

Open vSwitch is a software implementation of a virtual multilayer network switch, designed to enable effective network automation through programmatic extensions, while supporting standard management interfaces and protocols such as NetFlow, sFlow, SPAN, RSPAN, CLI, LACP and 802.1ag. In addition, Open vSwitch is designed to support transparent distribution across multiple physical servers by enabling creation of cross-server switches in a way that abstracts out the underlying server architecture, similar to the VMware vNetwork distributed vswitch or Cisco Nexus 1000V.

I asked this question to find out if there is support for this type of network bridges.
I would like to do such a scenario in the future with the ACRN hypervisor.
And I wanted to know your opinion.

Thanks to the whole team.

gvancuts, 2020-11-17:
Hi @ionutnechita , we used to have a tutorial about that - see https://projectacrn.github.io/2.0/tutorials/open_vswitch.html

We pulled it mainly because it was based on Clear Linux and we switched to Ubuntu as our reference Service VM OS. Having skimmed through the steps quickly now, it _should_ still be valid and the difference would mostly be in how you install OVS itself (which I can't think would be a big issue in Ubuntu or even Tumbleweed if that's what you're using).

It would be great if you would be willing to refresh (and revive) this tutorial if you try this! I'm more than happy to help with the document formatting! Let me know!

ionutnechita, 2020-11-17:
Hi @gvancuts,

I saw the document.
I will reproduce the steps from ClearLinux for OVS for first time.
I'm interested in making a page for OVS with updated steps.
But for me the final setup will be with OpenSUSE Tumbleweed.

Thanks for your support.
I'll be back with feedback.

gvancuts, 2020-11-18:
Thanks @ionutnechita !! I'm looking forward to your update :-)

closed 2022-07-05 by fuzhongl