projectacrn/acrn-hypervisor #5524: [ACRN] CPU parameters with advanced configuration

ionutnechita, 2020-11-17:
Hi ACRN team,

I looked through your documentation now, and from what I noticed, there is currently no advanced configuration for pCPU and vCPU.
https://projectacrn.github.io/latest/tutorials/cpu_sharing.html#scheduling-framework
From what I saw there is only one CPU control parameter. "--cpu_affinity". From what I've seen and read, it's pretty static.
I think it should be improved in the future.

For this ticket there would be three questions:
1. Are there any configuration parameter for the number of virtual cores allocated to a virtual machine at startup?
I do not notice in the documentation a parameter for the number of virtual cores. 
Example: 
parameter: --vcpu 
acrn-dm --vcpu 3
This number would help make virtual machines easier to set up. Set number virtual cpu for virtual machine.

2. CPU affinity is the static one, right? But how can I allocate for vCPU 0, another physical core.
Example: 
configuration vm: vCPU 0 with pCPU7

3. How can I make a more complex CPU affinity setup?
Example:
configuration vm:
VM1: 
vCPU 0 with pCPU7
vCPU 1 with pCPU5

**_--cpu_affinity 0:7, 1:5_**

Thanks to the whole team.

gvancuts, 2020-11-19:
> Hi ACRN team,
>

Hi @ionutnechita 
 
> I looked through your documentation now, and from what I noticed, there is currently no advanced configuration for pCPU and vCPU. [...]

This is statically defined for each VM in your scenario. You can further refine this using the `--cpu_affinity` parameter, by specifying which of the pCPUs you want to use. So if you want to have only three vCPUs in your VM you can list those in the command-line. I actually suspect you already understood that part... but I'm not clear what is the improvement you would like to see? Is that you don't want to have to state which pCPUs to use, you want to just be able to give a number?

>     2. CPU affinity is the static one, right? But how can I allocate for vCPU 0, another physical core. ( VM1 UOS )
>        Example:
>        configuration vm: vCPU 0 with pCPU7
> 
>     3. How can I make a more complex CPU affinity setup?
>        Example:
>        configuration vm:
>        VM1:
>        vCPU 0 with pCPU7
>        vCPU 1 with pCPU5

I think this 3rd example is a generalization of your second scenario above. As far as I know, it is not possible to do that. I'm not immediately seeing why you would want such fine-grained control, do you have a use-case in mind?

> 
> 
> **_--cpu_affinity 0:7, 1:5_**
> 
> Thanks to the whole team.

ionutnechita, 2020-11-25:
For this "I think this 3rd example is a generalization of your second scenario above. As far as I know, it is not possible to do that. I'm not immediately seeing why you would want such fine-grained control, do you have a use-case in mind?"

Yes, I would like to create an api for virtual machines in the future, or use libvirt support.

An advanced affinity change would bring more flexibility in creating virtual machines.
Follow these features: Virtual CPU Hotplug, Live Sharing/Live migration, Dynamic resources.

ionutnechita, 2020-11-25:
For this "Is that you don't want to have to state which pCPUs to use, you want to just be able to give a number?"

Yes, I would like to give a number, for a simple configuration.
And for a complex configuration, to give a number for vCPU, but also the affinity parameters with pCPU and vCPU.