projectacrn/acrn-hypervisor #5542: Can't create the boot file with a new compilation of acrn hypervisor #2

ionutnechita, 2020-11-24:
Hi ACRN team,

I tried to make a new compile, but I can't.
I notice some problems with an object (`sos_pci_devs') and ld issue.

Error is this:

**_Found nuc7i7dnb configuration for SCENARIO industry under /tmp/acrn-hypervisor/misc/vm_configs/scenarios/industry/nuc7i7dnb
cc main.c static_checks.c vm_cfg_checks.c ../../hypervisor/arch/x86/configs/vm_config.c /tmp/acrn-hypervisor/misc/vm_configs/scenarios/industry/vm_configurations.c /tmp/acrn-hypervisor/misc/vm_configs/scenarios/industry/nuc7i7dnb/pt_intx.c /tmp/acrn-hypervisor/misc/vm_configs/scenarios/industry/nuc7i7dnb/pci_dev.c  -I /tmp/acrn-hypervisor/hypervisor/include  -I /tmp/acrn-hypervisor/hypervisor/include/lib  -I /tmp/acrn-hypervisor/hypervisor/include/lib/crypto  -I /tmp/acrn-hypervisor/hypervisor/include/common  -I /tmp/acrn-hypervisor/hypervisor/include/arch/x86  -I /tmp/acrn-hypervisor/hypervisor/include/arch/x86/boot  -I /tmp/acrn-hypervisor/hypervisor/include/arch/x86/guest  -I /tmp/acrn-hypervisor/hypervisor/include/arch/x86/lib  -I /tmp/acrn-hypervisor/hypervisor/include/debug  -I /tmp/acrn-hypervisor/hypervisor/include/public  -I /tmp/acrn-hypervisor/hypervisor/include/dm  -I /tmp/acrn-hypervisor/hypervisor/include/hw  -I /tmp/acrn-hypervisor/hypervisor/boot/include  -I /tmp/acrn-hypervisor/hypervisor/boot/include/guest  -I /tmp/acrn-hypervisor/build/hypervisor/include  -I /tmp/acrn-hypervisor/misc/vm_configs/boards/nuc7i7dnb  -I /tmp/acrn-hypervisor/misc/vm_configs/scenarios/industry/nuc7i7dnb  -I /tmp/acrn-hypervisor/misc/vm_configs/scenarios/industry -include /tmp/acrn-hypervisor/build/hypervisor/include/config.h -I . -fno-stack-protector -fno-builtin -W -Wall -o /tmp/acrn-hypervisor/build/hypervisor/hv_prebuild_check.out
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/ccFQPdJY.o:(.bss+0x0): multiple definition of `sos_pci_devs'; /tmp/ccpIooiZ.o:(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:43: default] Error 1
make[2]: Leaving directory '/tmp/acrn-hypervisor/misc/hv_prebuild'
make[1]: *** [Makefile:397: pre_build] Error 2
make[1]: Leaving directory '/tmp/acrn-hypervisor/hypervisor'
make: *** [Makefile:148: hypervisor] Error 2_**

Development machine:
cat /etc/os-release
NAME="openSUSE Tumbleweed"
VERSION="20201121"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20201121"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20201121"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo"

ldd --version
ldd (GNU libc) 2.32
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

gcc --version
gcc (SUSE Linux) 10.2.1 20201028 [revision a78cd759754c92cecbf235ac9b447dcdff6c6e2f]
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

iasl -v
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20200717
Copyright (c) 2000 - 2020 Intel Corporation

commit 2767eb0d292ff6cb8c360aeb2bf760e5ac35cecf (HEAD -> master, origin/master, origin/HEAD)
Author: Yang,Yu-chu <yu-chu.yang@intel.com>
Date:   Fri Nov 6 15:26:23 2020 -0800

    acrn-config:refine the ivshmem for pre-launched vm

    1. Discard the method to find unused vbar bases from system ram, find
    unused mmio windows from 2G to 4G range.
    2. Refine the ivshmem devices declaration.

    Tracked-On: #5530
    Signed-off-by: Yang,Yu-chu <yu-chu.yang@intel.com>

commit b6a18fa0fb43a8f39a757029529a89c917e354d2
Author: Yang Yu-chu <yu-chu.yang@intel.com>
Date:   Tue Nov 17 13:23:48 2020 -0800

    acrn-config: update default pci_dev.c

    update all default pci_dev.c under
    misc/vm_configs/scenarios/<scenario>/<platform> for non-xml compilation

    Tracked-On: #5425
    Signed-off-by: Yang Yu-chu <yu-chu.yang@intel.com>

make all BOARD=nuc7i7dnb SCENARIO=industry RELEASE=0 2>&1 | tee -a log_compile1.txt
**_/tmp/ccmFEbRS.o:(.bss+0x0): multiple definition of `sos_pci_devs'; /tmp/cc0yvimU.o:(.bss+0x0): first defined here_**

[log_compile1.txt](https://github.com/projectacrn/acrn-hypervisor/files/5590948/log_compile1.txt)

gvancuts, 2020-11-25:
Hi @ionutnechita , I don't see this on my dev environment (still running Clear Linux). But I just created a Tumbleweed Docker container to try this out and I can reproduce the same error using it.

Not sure if it's the right approach but this patch got me passed this error:
```
diff --git a/misc/hv_prebuild/vm_cfg_checks.c b/misc/hv_prebuild/vm_cfg_checks.c
index 4c73fccaf..00b4392ed 100644
--- a/misc/hv_prebuild/vm_cfg_checks.c
+++ b/misc/hv_prebuild/vm_cfg_checks.c
@@ -21,7 +21,7 @@ static uint8_t rtvm_uuids[][16] = {
 static uint8_t safety_vm_uuid1[16] = SAFETY_VM_UUID1;
 
 /* sanity check for below structs is not needed, so use a empty struct instead */
-struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];
+/* struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM]; */
 const struct pci_vdev_ops vhostbridge_ops;
 const struct pci_vdev_ops vpci_ivshmem_ops;
 const struct pci_vdev_ops vmcs9900_ops;
```
Unfortunately, there is another error after that one... see:
```
make -C acrnprobe
make[3]: Entering directory '/workspace/misc/tools/acrn-crashlog/acrnprobe'
cc -c -g -O0 -std=gnu11 -D_GNU_SOURCE -m64 -Wall -ffunction-sections -Werror -O2 -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -fno-strict-aliasing -fpie -Wall -Wextra -pedantic -I/workspace/misc/tools/acrn-crashlog/include -I/workspace/misc/tools/acrn-crashlog/include/public -fstack-protector-strong -I /workspace/misc/tools/acrn-crashlog/common/include -I /workspace/misc/tools/acrn-crashlog/acrnprobe/include -I /usr/include/libxml2 -I /workspace/build/misc/tools/acrn-crashlog/include/acrnprobe -fdata-sections load_conf.c -o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/load_conf.o
cc -o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/bin/acrnprobe /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o /workspace/build/misc/tools/acrn-crashlog/common/obj/log_sys.o /workspace/build/misc/tools/acrn-crashlog/common/obj/cmdutils.o /workspace/build/misc/tools/acrn-crashlog/common/obj/fsutils.o /workspace/build/misc/tools/acrn-crashlog/common/obj/strutils.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/load_conf.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/channels.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/event_queue.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/event_handler.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/crash_reclassify.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/startupreason.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/property.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/probeutils.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/history.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/android_events.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/loop.o /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/vmrecord.o -Wl,-z,noexecstack -Wl,-z,relro,-z,now -pie -lpthread -lxml2 -lcrypto -lrt -lblkid -lext2fs -lcom_err -lsystemd -Wl,--gc-sections
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/load_conf.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: multiple definition of `conf'; /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/channels.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: multiple definition of `conf'; /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/event_handler.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: multiple definition of `conf'; /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/crash_reclassify.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: multiple definition of `conf'; /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: multiple definition of `conf'; /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:153: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/property.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/property.h:33: multiple definition of `gbuildversion'; /workspace/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o:/workspace/misc/tools/acrn-crashlog/acrnprobe/include/prope [... cut ...] /load_conf.h:153: first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:42: /workspace/build/misc/tools/acrn-crashlog/acrnprobe/bin/acrnprobe] Error 1
make[3]: Leaving directory '/workspace/misc/tools/acrn-crashlog/acrnprobe'
make[2]: *** [Makefile:88: all] Error 2
make[2]: Leaving directory '/workspace/misc/tools/acrn-crashlog'
make[1]: *** [Makefile:13: acrn-crashlog] Error 2
make[1]: Leaving directory '/workspace/misc'
make: *** [Makefile:181: tools] Error 2
```

gvancuts, 2020-11-25:
FWIW, Ubuntu 20.04 does not have the problem either.

gvancuts, 2020-11-25:
@NanlinXie who can look at this? You can reproduce this problem very easily by using my [OpenSUSE Tumbleweed Dockerfile](https://github.com/gvancuts/acrn-dockerfiles/blob/master/Dockerfile.tumbleweed) (for quick instructions, read the [README.md](https://github.com/gvancuts/acrn-dockerfiles/blob/master/README.md))

@ionutnechita , if you want to use Ubuntu 20.04 in the meantime, you can do so via my [Ubuntu 20.04 Dockerfile](https://github.com/gvancuts/acrn-dockerfiles/blob/master/Dockerfile.ubuntu20.04) ;-)

ionutnechita, 2020-11-25:
Ok, thanks for reproduce this.

ionutnechita, 2020-11-25:
For the Dockerfile with OpenSUSE, I will review it and come back with a feedback.
Thanks.

ionutnechita, 2020-11-25:
I also tried to make a change in the file: misc/hv_prebuild/vm_cfg_checks.c and I notice now that the issue no longer appears.
This would be the difference:

diff --git a/misc/hv_prebuild/vm_cfg_checks.c b/misc/hv_prebuild/vm_cfg_checks.c
index 4c73fcca..33454613 100644
--- a/misc/hv_prebuild/vm_cfg_checks.c
+++ b/misc/hv_prebuild/vm_cfg_checks.c
@@ -21,7 +21,7 @@ static uint8_t rtvm_uuids[][16] = {
 static uint8_t safety_vm_uuid1[16] = SAFETY_VM_UUID1;

 /* sanity check for below structs is not needed, so use a empty struct instead */
-struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];
+extern struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];
 const struct pci_vdev_ops vhostbridge_ops;
 const struct pci_vdev_ops vpci_ivshmem_ops;
 const struct pci_vdev_ops vmcs9900_ops;

struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];
 --> **_extern_** struct acrn_vm_pci_dev_config sos_pci_devs[CONFIG_MAX_PCI_DEV_NUM];

At the moment there are no more errors from: sos_pci_devs.
Thanks.

ionutnechita, 2020-11-25:
I created another ticket for the acrn-crashlog issue. ACRN#5549