projectacrn/acrn-hypervisor #5549: Can't create the acrn tools with a new compilation of acrn hypervisor

ionutnechita, 2020-11-25:
Hi ACRN team,

I tried to make a new compile, but I can't.
I notice some problems with acrn-crashlog tools and ld issue.

Error is this:
**_/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/load_conf.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/channels.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/event_handler.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/crash_reclassify.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/property.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:33: multiple definition of `gbuildversion'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:33: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/property.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:32: multiple definition of `guuid'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:32: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/property.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/probeutils.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:33: multiple definition of `gbuildversion'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:33: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/probeutils.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:32: multiple definition of `guuid'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/sender.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:32: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/probeutils.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/history.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/android_events.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: multiple definition of `conf'; /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/obj/main.o:/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:42: /tmp/acrn-hypervisor/build/misc/tools/acrn-crashlog/acrnprobe/bin/acrnprobe] Error 1
make[3]: Leaving directory '/tmp/acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe'
make[2]: *** [Makefile:88: all] Error 2_**

Development machine:
cat /etc/os-release
NAME="openSUSE Tumbleweed"
VERSION="20201121"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20201121"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20201121"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo"

ldd --version
ldd (GNU libc) 2.32
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

gcc --version
gcc (SUSE Linux) 10.2.1 20201028 [revision a78cd759754c92cecbf235ac9b447dcdff6c6e2f]
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

iasl -v
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20200717
Copyright (c) 2000 - 2020 Intel Corporation

commit 2767eb0d292ff6cb8c360aeb2bf760e5ac35cecf (HEAD -> master, origin/master, origin/HEAD)
Author: Yang,Yu-chu <yu-chu.yang@intel.com>
Date:   Fri Nov 6 15:26:23 2020 -0800

    acrn-config:refine the ivshmem for pre-launched vm

    1. Discard the method to find unused vbar bases from system ram, find
    unused mmio windows from 2G to 4G range.
    2. Refine the ivshmem devices declaration.

    Tracked-On: #5530
    Signed-off-by: Yang,Yu-chu <yu-chu.yang@intel.com>

commit b6a18fa0fb43a8f39a757029529a89c917e354d2
Author: Yang Yu-chu <yu-chu.yang@intel.com>
Date:   Tue Nov 17 13:23:48 2020 -0800

    acrn-config: update default pci_dev.c

    update all default pci_dev.c under
    misc/vm_configs/scenarios/<scenario>/<platform> for non-xml compilation

    Tracked-On: #5425
    Signed-off-by: Yang Yu-chu <yu-chu.yang@intel.com>

make all BOARD=nuc7i7dnb SCENARIO=industry RELEASE=0 2>&1 | tee -a log_compile1.txt
**_multiple definition of `guuid';
multiple definition of `conf';
multiple definition of `gbuildversion';
acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/property.h:32: first defined here
acrn-hypervisor/misc/tools/acrn-crashlog/acrnprobe/include/load_conf.h:155: first defined here_**

[log_compile1.txt](https://github.com/projectacrn/acrn-hypervisor/files/5597337/log_compile1.txt)

gvancuts, 2020-11-26:
It seems like the problems you've encountered are related to gcc version 10.x. I can reproduce these on Fedora 33 too (which also uses gcc 10.x). The new default to `-fno-common` seems to be the culprit here (see also https://gcc.gnu.org/gcc-10/porting_to.html for more details).

The following patch should solve your issues:
```
diff --git a/misc/hv_prebuild/Makefile b/misc/hv_prebuild/Makefile
index 0e5ecdee..d1b2b4fb 100644
--- a/misc/hv_prebuild/Makefile
+++ b/misc/hv_prebuild/Makefile
@@ -31,7 +31,7 @@ PRE_BUILD_SRCS += $(BOARD_CFG_DIR)/pt_intx.c
 ifneq (,$(wildcard $(BOARD_CFG_DIR)/pci_dev.c))
 PRE_BUILD_SRCS += $(BOARD_CFG_DIR)/pci_dev.c
 endif
-PRE_BUILD_CFLAGS += -fno-stack-protector -fno-builtin -W -Wall
+PRE_BUILD_CFLAGS += -fno-stack-protector -fno-builtin -W -Wall -fcommon
 PRE_BUILD_INCLUDE := $(patsubst %, -I %, $(INCLUDE_PATH)) -include $(HV_CONFIG_H) -I .
 
 .PHONY: default
diff --git a/misc/tools/acrn-crashlog/acrnprobe/Makefile b/misc/tools/acrn-crashlog/acrnprobe/Makefile
index 7bc25674..86f6f49e 100644
--- a/misc/tools/acrn-crashlog/acrnprobe/Makefile
+++ b/misc/tools/acrn-crashlog/acrnprobe/Makefile
@@ -9,6 +9,7 @@ INCLUDE         += -I $(CURDIR)/include -I $(SYSROOT)/usr/include/libxml2
 INCLUDE                += -I $(BUILDDIR)/include/acrnprobe
 CFLAGS                 += $(INCLUDE)
 CFLAGS                 += -fdata-sections
+CFLAGS                 += -fcommon
 
 LDFLAGS        += $(LIBS) -Wl,--gc-sections
```

Can you test it and confirm back to us whether it helps?

@NanlinXie , not sure if this is the right way of fixing the problem. If so, I can prepare a PR. If, on the other hand, this should really be fixed in the code and how we declare and define variables, then I'd need the engineering team to jump in to fix it. Thanks!

gvancuts, 2020-11-26:
As an FYI, compiling the ACPICA tools (`iasl`) will also require a newer version than the one in our documentation for those distros that come with gcc 10. The version from March 2020 (https://acpica.org/node/178) seems to be good but I'd recommend we try with the latest (e.g.: https://acpica.org/sites/acpica/files/acpica-unix2-20200717.tar.gz)

ionutnechita, 2020-11-26:
Hi @gvancuts,

I have now tried to compile.
And the change you propose works.
(+CFLAGS                 += -fcommon)
(PRE_BUILD_CFLAGS += -fno-stack-protector -fno-builtin -W -Wall -fcommon)

I saw the ticket: Get ready or gcc-10 #5553, ACRN#5553.
Someone should apply these changes for working with GCC 10, GLIBC 2.32 and IASL/ACPICA 20200717.

Thank you for checking and finding a fix.

fuzhongl, 2020-11-27:
[External_System_ID]ACRN-6856

closed 2020-12-22 by fuzhongl