projectacrn/acrn-hypervisor #5553: Get ready or gcc-10

gvancuts, 2020-11-26:
`gcc-10` will gradually be included in Linux distributions. OpenSUSE Tumbelweed and Fedora 33 for example already ship it.

There are a couple of issues with it:
1. Cannot compile ACRN: see also #5510 and #5549 for more details about the errors.
2. Cannot compile the ACPICA tools: these compilation problems were fixed in the March 2020 version (see https://acpica.org/sites/acpica/files/changes_49.txt). We should update our tests and documentation to use the latest, e.g. https://acpica.org/sites/acpica/files/acpica-unix2-20200717.tar.gz

Some background info and workaround for the first problem is to apply the patch listed in https://github.com/projectacrn/acrn-hypervisor/issues/5549#issuecomment-734216271

fuzhongl, 2020-11-27:
[External_System_ID]ACRN-6856

gvancuts, 2020-11-27:
@JasonChenCJ 

Here is how to reproduce the compilation error with gcc-10:
```
cd acrn-hypervisor
sudo docker run -v $PWD:/workspace gvancuts/tumbleweed-acrn-builder bash -c "make clean && make all BOARD=nuc7i7dnb SCENARIO=industry RELEASE=0"
```

gvancuts, 2020-12-01:
Hi @fuzhongl , I would like to keep this open until we have also fixed the ACPICA version issue.

gvancuts, 2021-03-15:
We still need to address this piece before we can close this bug report:
> Cannot compile the ACPICA tools: these compilation problems were fixed in the March 2020 version (see https://acpica.org/sites/acpica/files/changes_49.txt). We should update our tests and documentation to use the latest, e.g. https://acpica.org/sites/acpica/files/acpica-unix2-20200717.tar.gz

gvancuts, 2021-05-10:
All problems initially reported have been addressed, closing.

closed 2021-05-10 by gvancuts