projectacrn/acrn-hypervisor #5558: can't boot  RTVM in industry scenario

rome-xi, 2020-11-28:
Ubuntu version: 20.04
GCC version: 9.3
ACRN-hypervisor branch: release_2.2 (acrn-2020w40.1-180000p)
ACRN-Kernel (Service VM kernel): release_2.2 (5.4.43-PKT-200203T060100Z)
RT kernel for Ubuntu User OS: 4.19/preempt-rt (4.19.72-rt25)
HW: Maxtang Intel WHL-U i7-8565U 
toturial:https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html
I refer to the website tutorial,sos is booted,but the rtvm can't.
I guess the problem is my launch script.

#!/bin/bash
# Copyright (C) 2019 Intel Corporation.
# SPDX-License-Identifier: BSD-3-Clause
# This is an example of launch script for KBL NUC7i7DNH, may need to revise for other platform.

# pci devices for passthru
declare -A passthru_vpid
declare -A passthru_bdf

passthru_vpid=(
["eth"]="8086 156f"
["sata"]="8086 9dd3"
["nvme"]="8086 f1a6"
)
passthru_bdf=(
["eth"]="0000:00:1f.6"
["sata"]="0000:00:17.0"
["nvme"]="0000:02:00.0"
)

function launch_hard_rt_vm()
{
#for memsize setting
mem_size=1024M

modprobe pci_stub
# Ethernet pass-through
#echo ${passthru_vpid["eth"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["eth"]} > /sys/bus/pci/devices/${passthru_bdf["eth"]}/driver/unbind
#echo ${passthru_bdf["eth"]} > /sys/bus/pci/drivers/pci-stub/bind

# SATA pass-through
echo ${passthru_vpid["sata"]} > /sys/bus/pci/drivers/pci-stub/new_id
echo ${passthru_bdf["sata"]} > /sys/bus/pci/devices/${passthru_bdf["sata"]}/driver/unbind
echo ${passthru_bdf["sata"]} > /sys/bus/pci/drivers/pci-stub/bind

# NVME pass-through
#echo ${passthru_vpid["nvme"]} > /sys/bus/pci/drivers/pci-stub/new_id
#echo ${passthru_bdf["nvme"]} > /sys/bus/pci/devices/${passthru_bdf["nvme"]}/driver/unbind
#echo ${passthru_bdf["nvme"]} > /sys/bus/pci/drivers/pci-stub/bind

# for pm setting
pm_channel="--pm_notify_channel uart "
pm_by_vuart="--pm_by_vuart tty,/dev/ttyS1"

/usr/bin/acrn-dm -A -m $mem_size -s 0:0,hostbridge \
   --lapic_pt \
   --rtvm \
   --virtio_poll 1000000 \
   -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
   -s 2,passthru,00/17/0 \
   -s 3,virtio-console,@stdio:stdio_port \
   -s 8,virtio-net,tap0 \
   $pm_channel $pm_by_vuart \
/usr/bin/acrn-dm -A -m $mem_size -s 0:0,hostbridge \
   --lapic_pt \
   --rtvm \
   --virtio_poll 1000000 \
   -U 495ae2e5-2603-4d64-af76-d4bc5a8ec0e5 \
   -s 2,passthru,00/17/0 \
   -s 3,virtio-console,@stdio:stdio_port \
   -s 8,virtio-net,tap0 \
   $pm_channel $pm_by_vuart \
   --ovmf /usr/share/acrn/bios/OVMF.fd \
   -B"root=/dev/sda2 rw rootwait nohpet console=hvc0 console=ttyS0 \
   no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 \
   clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 \
   intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 \
   isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll \
   irqaffinity=0" \
   hard_rtvm

}

# -s 2,passthru,0/17/0 \ #please using "lspci -nn" check the bdf info

# Depends on which partation RT_LaaG is installed in;maybe need to change
# /dev/nvme0n1p3 to /dev/sda3 on NUC and uncomment SATA pass-through
# Add following RT_LaaG kernel cmdline into loader/entries/xxx.conf of EFI partation
#root=/dev/nvme0n1p3 rw rootwait nohpet console=hvc0 console=ttyS0 \
#no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 \
#clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 \
#intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 \
#isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll \
#irqaffinity=0

# offline SOS CPUs except BSP before launch UOS
for i in `ls -d /sys/devices/system/cpu/cpu[1-99]`; do
        online=`cat $i/online`
        idx=`echo $i | tr -cd "[1-99]"`
        echo cpu$idx online=$online
        if [ "$online" = "1" ]; then
                echo 0 > $i/online
                echo $idx > /sys/class/vhm/acrn_vhm/offline_cpu
        fi
done
How should I modify it so that I can launch rtvm?
launch_hard_rt_vm

fuzhongl, 2020-11-29:
@rome-xi Please remove **-s 8,virtio-net,tap0** in your launch script.
Thanks!

rome-xi, 2020-11-29:
> @rome-xi Please remove **-s 8,virtio-net,tap0** in your launch script.
> Thanks!
The result has not changed.

rome-xi, 2020-11-29:
linux /boot/bzImage root=PARTUUID="fe3100fa-5742-421f-8381-c63a4977e9c4" rw rootwait nohpet console=hvc0 console=ttyS0 no_timer_check ignore_loglevel log_buf_len=16M consoleblank=0 clocksource=tsc tsc=reliable x2apic_phys processor.max_cstate=0 intel_idle.max_cstate=0 intel_pstate=disable mce=ignore_ce audit=0 isolcpus=nohz,domain,1 nohz_full=1 rcu_nocbs=1 nosoftlockup idle=poll irqaffinity=0

fuzhongl, 2020-11-29:
zcat /proc/config.gz

check if following is in:

CONFIG_SERIAL_8250_EXTENDED=y
CONFIG_SERIAL_8250_DETECT_IRQ=y

fuzhongl, 2020-12-04:
@rome-xi Please apply following patch in 4.19 rt kernel: Then, you can login without uart cable.

```
From f27141bd61bf4dad80300a18d10c4a170caedac7 Mon Sep 17 00:00:00 2001
From: "wei.tan@intel.com" <wei.tan@intel.com>
Date: Thu, 6 Jun 2019 02:18:23 +0000
Subject: [PATCH] virtio: add virtio polling support for ACRN

add virtio polling mode for ACRN.

---
 drivers/block/virtio_blk.c    |   3 +
 drivers/char/virtio_console.c |   3 +
 drivers/net/virtio_net.c      |   3 +
 drivers/virtio/Kconfig        |  10 ++++
 drivers/virtio/virtio.c       | 101 ++++++++++++++++++++++++++++++++++
 drivers/virtio/virtio_input.c |   3 +
 drivers/virtio/virtio_ring.c  |  35 +++++++++++-
 include/linux/virtio.h        |  21 +++++++
 include/linux/virtio_config.h |  17 ------
 x86-64_defconfig              |   4 ++
 10 files changed, 181 insertions(+), 19 deletions(-)

diff --git a/drivers/block/virtio_blk.c b/drivers/block/virtio_blk.c
index f88d39ca53fb..3c1fcbe9c958 100644
--- a/drivers/block/virtio_blk.c
+++ b/drivers/block/virtio_blk.c
@@ -984,6 +984,9 @@ static struct virtio_driver virtio_blk = {
 	.driver.name			= KBUILD_MODNAME,
 	.driver.owner			= THIS_MODULE,
 	.id_table			= id_table,
+#ifdef CONFIG_VIRTIO_PMD
+	.polling_mode        = true,
+#endif
 	.probe				= virtblk_probe,
 	.remove				= virtblk_remove,
 	.config_changed			= virtblk_config_changed,
diff --git a/drivers/char/virtio_console.c b/drivers/char/virtio_console.c
index c55f6aeb4227..d0970d334df0 100644
--- a/drivers/char/virtio_console.c
+++ b/drivers/char/virtio_console.c
@@ -2239,6 +2239,9 @@ static struct virtio_driver virtio_console = {
 	.driver.name =	KBUILD_MODNAME,
 	.driver.owner =	THIS_MODULE,
 	.id_table =	id_table,
+#ifdef CONFIG_VIRTIO_PMD
+	.polling_mode        = true,
+#endif
 	.probe =	virtcons_probe,
 	.remove =	virtcons_remove,
 	.config_changed = config_intr,
diff --git a/drivers/net/virtio_net.c b/drivers/net/virtio_net.c
index c88ee376a2eb..b7ab99b46be3 100644
--- a/drivers/net/virtio_net.c
+++ b/drivers/net/virtio_net.c
@@ -3226,6 +3226,9 @@ static struct virtio_driver virtio_net_driver = {
 	.driver.owner =	THIS_MODULE,
 	.id_table =	id_table,
 	.validate =	virtnet_validate,
+#ifdef CONFIG_VIRTIO_PMD
+	.polling_mode        = true,
+#endif
 	.probe =	virtnet_probe,
 	.remove =	virtnet_remove,
 	.config_changed = virtnet_config_changed,
diff --git a/drivers/virtio/Kconfig b/drivers/virtio/Kconfig
index 51d43fd4bd2b..3e8f13a08a16 100644
--- a/drivers/virtio/Kconfig
+++ b/drivers/virtio/Kconfig
@@ -100,4 +100,14 @@ config ACRN_VIRTIO_DEVICES
[... 370 lines cut ...]
index c1f8175925fc..4de1162c13d2 100644
--- a/x86-64_defconfig
+++ b/x86-64_defconfig
@@ -127,6 +127,8 @@ CONFIG_RCU_FANOUT_LEAF=16
 # CONFIG_RCU_BOOST is not set
 CONFIG_RCU_NOCB_CPU=y
 # CONFIG_IKCONFIG is not set
+CONFIG_IKCONFIG=y
+CONFIG_IKCONFIG_PROC=y
 CONFIG_LOG_BUF_SHIFT=17
 CONFIG_LOG_CPU_MAX_BUF_SHIFT=12
 CONFIG_PRINTK_SAFE_LOG_BUF_SHIFT=13
@@ -5707,6 +5709,8 @@ CONFIG_VIRTIO_BALLOON=m
 CONFIG_VIRTIO_INPUT=m
 CONFIG_VIRTIO_MMIO=m
 # CONFIG_VIRTIO_MMIO_CMDLINE_DEVICES is not set
+CONFIG_ACRN_VIRTIO_DEVICES=y
+CONFIG_VIRTIO_PMD=y
 # CONFIG_ACRN_VIRTIO_DEVICES is not set
 
 #
-- 
2.17.1

```

closed 2020-12-22 by fuzhongl

zhangshuai75, 2021-04-19:
> @rome-xi Please apply following patch in 4.19 rt kernel: Then, you can login without uart cable. We will switch rt kernel to 5.4 later; so 4.19 kernel isn't maintained any more.

Hi @fuzhongl 
Can you update the patch file?
When copy the above patch content to a file, it will cause the use of “spaces” to replace the “Tab”, which will lead to patching failure

fuzhongl, 2021-04-19:
Hi ZhangShai
The patch is generated by git.
The format maybe changed when copied to github.
This patch is NOT MUST if your board has serial port or you use 5.4 rt kernel.

BR.
Fuzhong

iaGuoZhi, 2021-04-25:
Hi @fuzhongl 
Can ACRN hypervisor v2.0 support rt kernel 5.4? I encountered the same problem when boot rt kernel 4.19 on ACRN hypervisor v2.0.

iaGuoZhi, 2021-04-25:
Is ACRN document out of date? Documents whether v2.0 or latest version both tell me to build rt kernel with version 4.19. which will cause above problem.

fuzhongl, 2021-04-27:
@iaGuoZhi It is fine for v2.0.
Which board are you using? with serial port or not? If without serial please adding virtio-console in launch script?

iaGuoZhi, 2021-04-27:
Hi @fuzhongl , I'm working on ACRN v2.0 on MAXTANG WHL Board.(WHL-IPC-I5). However I came across such error when launch RTVM with kernel version 4.19. 

https://github.com/projectacrn/acrn-hypervisor/issues/5970#issuecomment-826776579

Would you mind having a look at this issue? Thanks a lot.