projectacrn/acrn-hypervisor #5561: Log output in `build/hypervisor/cfg.log` is polluted with '-e' characters

gvancuts, 2020-11-30:
The log output found in `build/hypervsior/cfg.log` is polluted with `-e` sequences - see:
```
-e 
ESC[47;30mACRN Configuration Summary:ESC[0m 
BOARD = nuc7i7dnb        SCENARIO = industry
-e BUILD type = -e DEBUG
-e Hypervisor configuration is based on:
-e      defconfig file:                 /home/acrn/work/acrn-hypervisor/misc/vm_configs/scenarios/industry/nuc7i7dnb/nuc7i7dnb.config;
-e      Others are set by default in:   /home/acrn/work/acrn-hypervisor/hypervisor/arch/x86/Kconfig;
-e VM configuration is based on:
-e      Source code at:                 /home/acrn/work/acrn-hypervisor/misc/vm_configs
```

wenlingz, 2020-12-01:
@gvancuts please help add JIRA ID, thanks

fuzhongl, 2020-12-01:
[External_System_ID]ACRN-6868

closed 2020-12-22 by fuzhongl