projectacrn/acrn-hypervisor #5570: [Python 3.9] AttributeError: 'xml.etree.ElementTree.Element' object has no attribute 'getchildren'

saininav, 2020-12-02:
Python 3.9 does not support xml.etree.ElementTree.Element.getchildren. It throws build error.

Error log:
|   File "acrn-workspace/poky/build-gatesgarth/master-acrn-sos/work/x86_64-linux/acrn-hypervisor-native/2.3-r0/git/misc/acrn-config/scenario_config/../acpi_gen/asl_gen.py", line 426, in main
|     for mmio_dev_node in mmio_dev_nodes.getchildren():
| AttributeError: 'xml.etree.ElementTree.Element' object has no attribute 'getchildren'
| make: *** No rule to make target 'update_config', needed by 'acrn-workspace/poky/build-gatesgarth/master-acrn-sos/work/x86_64-linux/acrn-hypervisor-native/2.3-r0/build/hypervisor/.config'.  Stop.
| make: Leaving directory 'acrn-workspace/poky/build-gatesgarth/master-acrn-sos/work/x86_64-linux/acrn-hypervisor-native/2.3-r0/git/hypervisor'
| WARNING: exit code 1 from a shell command.

Ref: https://docs.python.org/3.8/library/xml.etree.elementtree.html#xml.etree.ElementTree.Element.getchildren

Board: nuc7i7dnb
Scenario: hybrid
Python version: 3.9

szhen11, 2020-12-07:
[External_System_ID]ACRN-6881

wenlingz, 2020-12-15:
@saininav patch has been merged, please help verify and close this issue, thanks

saininav, 2020-12-16:
Issue fixed.  Thanks.

closed 2020-12-16 by saininav