projectacrn/acrn-hypervisor #5572: [WHL][Hybrid_rt] build hybrid_rt hv from source, boot RTVM fail

Leo2Chang, 2020-12-02:
**Describe the bug**
[WHL][Hybrid_rt] build hybrid_rt hv from source, boot RTVM fail

**Platform**
whl-ipc-i7; whl-ipc-i5

**Codebase**
repo: https://github.com/projectacrn/acrn-hypervisor
branch: release_2.3
commit d57a213242741e23c4ffe1832c57551e05a1406b (HEAD -> release_2.3, tag: acrn-2020w48.5-180000p, origin/release_2.3)

**Scenario**
Hybrid_rt

**To Reproduce**
1. git clone https://github.com/projectacrn/acrn-hypervisor
2. git checkout release_2.3
3. make all BOARD=whl-ipc-i7 SCENARIO=hybrid_rt RELEASE=0
4. scp acrn.bin ACPI_VM0.bin onto whl board
5. poweron board to boot HV

**Expected behavior**
can boot HV, pre-launched RTVM and SOS successfully

**Actual result**
can boot HV and SOS successfully
can not boot pre-launched RTVM, log as:
[rel2.3_hybridRT_buildSource_bootRTVMfail.txt](https://github.com/projectacrn/acrn-hypervisor/files/5628302/rel2.3_hybridRT_buildSource_bootRTVMfail.txt)

**Reproduce rate**
5/5

**Debugging info**
build hybrid_rt hv with xml file:
"make BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/whl-ipc-i7.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/whl-ipc-i7/hybrid_rt.xml"
the HV can boot HV, pre-lanuched RTVM and SOS;

can reproduce this issue on whl-ipc-i5;

fuzhongl, 2020-12-03:
[External_System_ID]ACRN-6871

Leo2Chang, 2021-02-25:
can not reproduce this issue on whl-ipc-i7 and whl-ipc-i5
with both cmd:
`make all BOARD=whl-ipc-i7 SCENARIO=hybrid_rt RELEASE=0
make BOARD_FILE=$PWD/misc/config_tools/data/whl-ipc-i7/whl-ipc-i7.xml SCENARIO_FILE=$PWD/misc/config_tools/data/whl-ipc-i7/hybrid_rt.xml`

HV commit:
bc22a62b4d3d3d5e8c3b4b884fdcfca746dd4d3f (HEAD -> release_2.4, origin/release_2.4)

closed 2021-02-25 by Leo2Chang