projectacrn/acrn-hypervisor #5573: [EHL][logical_partition] build EHL logical_partition hv from source Fail

Leo2Chang, 2020-12-02:
**Describe the bug**
[EHL][logical_partition] build EHL logical_partition hv from source Fail

**Platform**
board: ehl-crb-b
scenario: logical_partition
build cmd: make all BOARD=ehl-crb-b SCENARIO=logical_partition RELEASE=0

**Codebase**
repo: https://github.com/projectacrn/acrn-hypervisor
branch: release_2.3
commit d57a213242741e23c4ffe1832c57551e05a1406b (HEAD -> release_2.3, tag: acrn-2020w48.5-180000p, origin/release_2.3)

**build enviroment**
ssg-kvm@HP:acrn-hypervisor$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
......
gcc version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04)
ssg-kvm@HP:acrn-hypervisor$ ld -v
GNU ld (GNU Binutils for Ubuntu) 2.30
ssg-kvm@HP:acrn-hypervisor$ iasl -v
Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20190703
Copyright (c) 2000 - 2019 Intel Corporation
ssg-kvm@HP:acrn-hypervisor$ uname -r
4.15.0-121-generic
ssg-kvm@HP:acrn-hypervisor$

**To Reproduce**
1. git clone https://github.com/projectacrn/acrn-hypervisor
2. git checkout release_2.3
3. make all BOARD=ehl-crb-b SCENARIO=logical_partition RELEASE=0

**Expected behavior**
can build logical_partition scenario HV successfully for ehl-crb-b

**Actual result**
build HV fail

**error**
/home/ssg-kvm/share/2020nativeUbtuBuildenv/acrn-hypervisor/build/hypervisor/hv_prebuild_check.out
/tmp/ccGSlpXy.o:(.data.rel+0x80): undefined reference to `vm0_pci_devs'
/tmp/ccGSlpXy.o:(.data.rel+0xb48): undefined reference to `vm1_pci_devs'
collect2: error: ld returned 1 exit status
Makefile:39: recipe for target 'default' failed
make[2]: *** [default] Error 1

**Debugging info**
in same env can build EHL industry/hybrid/hybrid_rt hv from source PASS;
in same env can build EHL logical_partition Pass with xml:
make BOARD_FILE=$PWD/misc/vm_configs/xmls/board-xmls/ehl-crb-b.xml SCENARIO_FILE=$PWD/misc/vm_configs/xmls/config-xmls/ehl-crb-b/logical_partition.xml

fuzhongl, 2020-12-03:
[External_System_ID]ACRN-6872

Leo2Chang, 2021-02-25:
can not reproduce this issue on **release_2.4** with cmdline:
`make all BOARD=ehl-crb-b SCENARIO=logical_partition RELEASE=0`
`make BOARD_FILE=$PWD/misc/config_tools/data/ehl-crb-b/ehl-crb-b.xml SCENARIO_FILE=$PWD/misc/config_tools/data/ehl-crb-b/logical_partition.xml`

HV commit bc22a62b4d3d3d5e8c3b4b884fdcfca746dd4d3f (HEAD -> release_2.4, origin/release_2.4)

close this issue as fixed.
thank you~

closed 2021-02-25 by Leo2Chang