projectacrn/acrn-hypervisor #5588: ACRN Configuration Summary Message is not well formed on the screen

ionutnechita, 2020-12-05:
Hi ACRN team,

I tried to make a new compilation, but the result summary is a bit changed as well.
Error is this:
**_make[1]: Leaving directory '/acrn-hypervisor/devicemodel'
\n\033[47;30mACRN Configuration Summary:\033[0m \nBOARD = nuc7i7dnb\t SCENARIO = industry
BUILD type = \c
RELEASE
Hypervisor configuration is based on:
\tdefconfig file:\t\t\t/acrn-hypervisor/misc/vm_configs/scenarios/industry/nuc7i7dnb/nuc7i7dnb.config;
\tOthers are set by default in:\t/acrn-hypervisor/hypervisor/arch/x86/Kconfig;
VM configuration is based on:
\tSource code at:\t\t\t/acrn-hypervisor/misc/vm_configs_**

cat build/hypervisor/cfg.log
\n\033[47;30mACRN Configuration Summary:\033[0m \nBOARD = nuc7i7dnb\t SCENARIO = industry
BUILD type = \c
RELEASE
Hypervisor configuration is based on:
\tdefconfig file:\t\t\t/acrn-hypervisor/misc/vm_configs/scenarios/industry/nuc7i7dnb/nuc7i7dnb.config;
\tOthers are set by default in:\t/acrn-hypervisor/hypervisor/arch/x86/Kconfig;
VM configuration is based on:
\tSource code at:\t\t\t/acrn-hypervisor/misc/vm_configs

From what I see the echo command that makes the on-screen display and the file, no longer works properly.

[log_compile1.txt](https://github.com/projectacrn/acrn-hypervisor/files/5647695/log_compile1.txt)

ionutnechita, 2020-12-05:
With git revert from this commit:
git revert e2590d88aa3e1c72dc6473eadc83312412a604b3

    Revert "hv: fix formatting for cfg.log output"

    This reverts commit e2590d88aa3e1c72dc6473eadc83312412a604b3.

Summary is fine.

fuzhongl, 2020-12-07:
@gvancuts Could you please help to check it?
Thanks!

gvancuts, 2020-12-07:
Hmmm... OK, the behaviour is different based on different systems. Tumbleweed and Clear Linux need this `-e` option to format things properly, but Ubuntu does not (and actually prints out this `-e` inside the file which also messes up (differently) the format.

gvancuts, 2020-12-07:
OK, I think I found what's causing this. On Clear Linux and Tumbleweed, `/bin/sh` is a symlink to `/bin/bash`. `/bin/sh` is the shell used by default by `make` unless otherwise stated. The built-in `echo` command in bash understands the `-e` argument.

On Ubuntu, `/bin/sh` points at `/bin/dash` and the built-in `echo` command in dash does not understand the `-e` option, so it simply prints it out on the screen.

`-e` is not a POSIX compliant argument, so it is considered not very portable to use it.

Having said that, the simplest option I came up with in this case is to explicitly set the shell to be used by `make` to `/bin/bash`

gvancuts, 2020-12-07:
@ionutnechita , can you do a sanity check on your side with this fix? I tried it on my Docker image based on Tumbleweed and it solved the problem... so hopefully this works for your native env too!

ionutnechita, 2020-12-07:
Hi @gvancuts,

I tested now with this modification is fine now.
Using OpenSUSE Tumbleweed, native OS.
It's fine to merge this modification.

Thanks.

fuzhongl, 2020-12-08:
[External_System_ID] ACRN-6882

gvancuts, 2020-12-15:
patch has been merged, closing this issue.

closed 2020-12-15 by gvancuts