projectacrn/acrn-hypervisor #5594: documentation issues on "Getting Started Guide for ACRN Industry Scenario with Ubuntu Service VM"

SLinke-intel, 2020-12-09:
The "Getting Started Guide for ACRN Industry Scenario with Ubuntu Service VM" has some misleading content:
 https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html
- In “Build ACRN HV on Ubuntu” -> “5. Build ACRN”
https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#build-the-acrn-hypervisor-on-ubuntu
missing: ”sudo mkdir /boot/acrn/”
- **False syntax in the Grub menuentry**?
https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#update-grub-for-the-ubuntu-service-vm 
I expect I **should be**
“search --no-floppy --fs-uuid **–set=root** 9bd58889-add7-410c-bdb7-1fbc2af9b0e1”
- Unlike (as far as I remember) the 1.6 version - now /dev/nvme0p1 is used for HV/SOS and /dev/sda is used for the RT OS.
Now there is this hint to use blkid to read the UUIDs:
“sudo blkid /dev/sda*”
I assume this should be
“sudo blkid /dev/nvme0n1*”

-	In the original /etc/grub.d/40_custom file there is  a line “exec tail -n +3 $0” that should remain there. Is that correct? (This is not mentioned In the GSG.) 

**Platform**
Maxtang/WHL 

**Codebase**
2.3

**Scenario**
Industry

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html
2. Follow the steps

gvancuts, 2020-12-09:
> * In “Build ACRN HV on Ubuntu” -> “5. Build ACRN”
>   https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#build-the-acrn-hypervisor-on-ubuntu
>   missing: ”sudo mkdir /boot/acrn/”

Good catch!

> * **False syntax in the Grub menuentry**?
>   https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#update-grub-for-the-ubuntu-service-vm
>   I expect I **should be**
>   “search --no-floppy --fs-uuid **–set=root** 9bd58889-add7-410c-bdb7-1fbc2af9b0e1”

You can do that but my understanding is that this is not necessary as `root` is the value by default if none is provided (see https://www.gnu.org/software/grub/manual/grub/html_node/search.html)

> * Unlike (as far as I remember) the 1.6 version - now /dev/nvme0p1 is used for HV/SOS and /dev/sda is used for the RT OS.
>   Now there is this hint to use blkid to read the UUIDs:
>   “sudo blkid /dev/sda*”
>   I assume this should be
>   “sudo blkid /dev/nvme0n1*”

I tend to agree with you. The hint really is about using `blkid` to figure out the right UUID but using a different device node makes it a bit confusing.

> * In the original /etc/grub.d/40_custom file there is  a line “exec tail -n +3 $0” that should remain there. Is that correct? (This is not mentioned In the GSG.)

This is correct. That line is at the top. There is a comment in that file that explains the line **should** stay intact and that any custom entry should be put below the comment explaining this. It's a bit redundant to mention it in the GSG again but that certainly wouldn't cause any harm.

gvancuts, 2020-12-09:
@SLinke-intel , take a look at #5595 and let me know if this addresses your comments.

SLinke-intel, 2020-12-09:
Hi Geoffrey,

Thanks for your response. I agree that your changes cover the remaining points.
As a feedback for you: I think that the correctly exchanging the UUID PARTUUID in the grub menu entry is a critical part, that can easily become a pitfall. (I failed myself at the beginning and noticed the same mistake done by a colleague). The documentation is in fact correct, but maybe the wording could be reviewed so it becomes more clear even for the unexpected reader.

Best Regards, Stephan

https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#build-the-acrn-hypervisor-on-ubuntu

https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#update-grub-for-the-ubuntu-service-vm

Intel Deutschland GmbH
Registered Address: Am Campeon 10-12, 85579 Neubiberg, Germany
Tel: +49 89 99 8853-0, www.intel.de
Managing Directors: Christin Eisenschmid, Gary Kershaw
Chairperson of the Supervisory Board: Nicole Lau
Registered Office: Munich
Commercial Register: Amtsgericht Muenchen HRB 186928

closed 2020-12-10 by SLinke-intel

gvancuts, 2020-12-15:
> As a feedback for you: I think that the correctly exchanging the UUID PARTUUID in the grub menu entry is a critical part, that can easily become a pitfall. (I failed myself at the beginning and [...]

Hi @SLinke-intel , I'm not completely following you on this comment. Can you explain what mistake you made initially (and where we could make the documentation clearer)? Thanks!

SLinke-intel, 2020-12-22:
Hi Geoffrey,

in the GSG adapting the new entry in /etc/grub.d/40_custom UUSD/PARTUUID lines

  search --no-floppy --fs-uuid --set 9bd58889-add7-410c-bdb7-1fbc2af9b0e1
AND
  multiboot2 /boot/acrn/acrn.bin  root=PARTUUID="e515916d-aac4-4439-aaa0-33231a9f4d83"

is described as 
"Update this to use the UUID (--set) and PARTUUID (root= parameter) (...) of the root partition"

Sometimes this leads to user replacing both uuid with UUID or PARTUUID, since the concept itself is not clear to the user.
Fatally this leads to the OS not booting.

I suggest using these two lines for setting the proper values.

	sudo sed -i 's/9bd58889-add7-410c-bdb7-1fbc2af9b0e1/'`sudo blkid -o value -s UUID /dev/nvme0n1p2`'/g' 
	sudo sed -i 's/e515916d-aac4-4439-aaa0-33231a9f4d83/'`sudo blkid -o value -s PARTUUID /dev/nvme0n1p2`'/g' 

For easier changing of the target device /dev/nvme0n1p2 could be replaced by a preset variable...

SLinke-intel, 2020-12-22:
Sorry my for late response to your question...
Have a nice holiday season.

Stephan Linke

gvancuts, 2020-12-23:
> Sorry my for late response to your question...

No worries at all, and thanks for the clarification!

> Have a nice holiday season.

You too! :-)