projectacrn/acrn-hypervisor #5596: Issues configuring Intel NUC7i7DNHE and running Ubutntu 18.04 as SOS and windows LTS as UOS

sunilmpatel, 2020-12-09:
**Description**
Issues configuring Intel NUC7i7DNHE, goal is to get following configuration working using Intel NUC:
- SOS (UBUNTU 18.04) 
- UOS Windows 10 LTS
- GVT-g/GVT-d (GPU pass thru) to windows guest

Issue #1 Once I boot into SOS, I get folllowing messages periodically on see attached dMesg.txt file.
[  377.126628] [drm:gen8_de_irq_handler] *ERROR* Fault errors on pipe A: 0x00000080

Issue #2: Goal is to get windows running as UOS once I get SOS running.
I following the getting started guide to create windows image
https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#using-windows-as-uos
After I prepare the script to create an image I execute install_win.sh and I do not see UEFIshell, I also tried calling install_win.sh using weston Insted am gettting GVT initialization failed errors "GVT: init failed". See attached gvtlogs.txt.
**Platform**
Intel NUC7i7DNHE,
Attached lspci.txt, lshw.txt, lscpu.txt and my 40_custom file parameters that I boot with. I have also attached GVTlogs indicating GVT related failures.

**Codebase**
2.3 ACRN release

**Scenario**
Describe the scenario: Industry

**To Reproduce**
To Setup Intel NUC7i7DNHE we are following instructions listed here:
https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html
Since we do not have access to Maxtang/WHL board, we are using one of the recommended hardware platforms NUC7i7DNHE.

[40_custom.txt](https://github.com/projectacrn/acrn-hypervisor/files/5669011/40_custom.txt)
[dMesg.txt](https://github.com/projectacrn/acrn-hypervisor/files/5669012/dMesg.txt)
[gvtlogs.txt](https://github.com/projectacrn/acrn-hypervisor/files/5669013/gvtlogs.txt)
[lscpu.txt](https://github.com/projectacrn/acrn-hypervisor/files/5669014/lscpu.txt)
[lshw.txt](https://github.com/projectacrn/acrn-hypervisor/files/5669015/lshw.txt)
[lspci.txt](https://github.com/projectacrn/acrn-hypervisor/files/5669016/lspci.txt)

fuzhongl, 2020-12-10:
@sunilmpatel 
About issue 1#:
This is known issue after v2.0. Could you please use v2.0 until this issue is fixed?
or using text mode for Ubuntu sos?

About issue 2#
Please refer [Enable GVT-d in ACRN](https://projectacrn.github.io/latest/tutorials/gpu-passthru.html) to enable GOP feature of GVT-d; which is need for WaaG image creation.

sunilmpatel, 2020-12-10:
About issue 1#:
After I rebuild v2.0 using stock board and scenerio files **nuc7i7dnb.xml**. Upon reboot into my SOS my Screen gets stuck at "Loading ACRN". I have tried following kernel options so far and nothing seems to be helping..
i915.enable_gvt=1 i915.gvt_workload_priority=1 i915.enable_initial_modeset=1 i915.nuclear_pageflip=1

Attached are dmesg log
[dmesg_2_0.txt](https://github.com/projectacrn/acrn-hypervisor/files/5675877/dmesg_2_0.txt)

About Issue 2#
Do you would want me to use v2.3 or stay with V2.0 and go thru the steps to **Enable GVT-d in ACRN** to enable GOP feature of GVT-d.

fuzhongl, 2020-12-15:
@sunilmpatel  
Issue 1#:
Please refer  [Using the Serial Port on KBL Intel NUC](https://projectacrn.github.io/latest/tutorials/using_serial_port.html) and share console log?
If there is no this serial cable on your side, please using **text mode** for ubuntu sos with ACRN v2.3.

Issue 2#:
We prefer v2.3.

Thanks!

fuzhongl, 2022-03-22:
Closed since no feedback long time.

closed 2022-03-22 by fuzhongl