projectacrn/acrn-hypervisor #5599: Issues configuring Maxtang WL10  (i58265U) and running Ubutntu 18.04 as SOS and windows LTS as UOS with GVT-g

slinke67, 2020-12-13:
**Describe the bug**
When following the "Getting Started Guide for ACRN Industry Scenario with Ubuntu Service VM" and the "Launch Windows as the Guest VM on ACRN" tutorial of ACRN 2.3  WaaG fails to enable GVT-g.

**Platform**
Maxtang WL10  (i58265U)
32GB
HDMI + DP connected (Ubuntu on both displays)

**Codebase**
Ubuntu 18.04 desktop + ACRN 2.3

**Scenario**
Industry scenario following GSG, HV/SOS on NVME

**To Reproduce**
Follow the GSG for industry scenario and the WaaG tutorial.
https://projectacrn.github.io/latest/getting-started/rt_industry_ubuntu.html#getting-started-guide-for-acrn-industry-scenario-with-ubuntu-service-vm
https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#launch-windows-as-the-guest-vm-on-acrn
(SSD remained disabledbefore launching WaaG install script)
Launch the install_win script
-> script stops

**Expected behavior**
prompt for booting installer CD and later the Windows10 installer should show up on one display

**Additional context**
 ->  Need instructions how to add attachments
available:
 

- SOS dmesg
- hypervisor log
- install_win.sh script
- log of WaaG install launch
- output of "acrnctl list"
- script used to setup from original UBuntu desctop installation following GSG and WaaG tutorial

slinke67, 2020-12-14:
adding the missing log files.
[dmesg.log](https://github.com/projectacrn/acrn-hypervisor/files/5687010/dmesg.log)
[hv.log](https://github.com/projectacrn/acrn-hypervisor/files/5687015/hv.log)
[install_win.sh.txt](https://github.com/projectacrn/acrn-hypervisor/files/5687017/install_win.sh.txt)
[waag.log](https://github.com/projectacrn/acrn-hypervisor/files/5687009/waag.log)
[acrncrl.list.txt](https://github.com/projectacrn/acrn-hypervisor/files/5687014/acrncrl.list.txt)
[acrn_2.3_Maxtang_WHL_setup.txt](https://github.com/projectacrn/acrn-hypervisor/files/5687020/acrn_2.3_Maxtang_WHL_setup.txt)

slinke67, 2020-12-14:
FYI: also tried to install Ubuntu 18.04 server variant to run WaaG with GVT-d but I read **somewhere** this is currently not supported for WaaG installation.

I read (not related to WaaG installation tutorial) that "If you use GVT-g, you can set up Weston in the Service VM, and set up Weston as the desktop environment in the Service VM". Does that mean I should switch from regular Ubuntu desktop (X11 based?) to Weston before starting WaaG installation?
https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html#boot-windows-on-acrn-with-a-default-configuration

I am missing the i915 kernel parameters of the SOS kernel that used to be used to controll how each of the virtual OS is allowed to access the graphics hardware.g. before ACRN 2.0 we used these (GVT-d, Weston):
i915.enable_gvt=1 i915.enable_guc=0 i915.nuclear_pageflip=1 i915.avail_planes_per_pipe=0x010000 i915.domain_plane_owners=0x000022221111

fuzhongl, 2020-12-21:
@slinke67 Sorry for response so late!
After v2.0, the acrngt switches to IDV solution; but there is known issue for GVT-g.
Please refer following [PR](https://github.com/projectacrn/acrn-hypervisor/pull/5617/) before it is merged.
Thanks!

fuzhongl, 2021-01-05:
@slinke67 Please have a try with: https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html
Thanks!

slinke67, 2021-03-11:
Thanks for the support. Unfortunately the focus has changed and I may not follow up.

closed 2021-03-11 by slinke67