projectacrn/acrn-hypervisor #5625: [CFL][industry] Fewer warnings and errors when starting SOS

ionutnechita, 2020-12-24:
Hi ACRN Team,

I started the SOS, and I notice some errors in the ACRN console, but also in dmesg.
Some of them seem quite important, but others are just low-impact warnings.
I will attach the logs to talk about them.

One of the issues:
...
[Thu Dec 24 15:01:25 2020] platform idma64.0: failed to claim resource 0: [mem 0x00000800-0x00000fff]
[Thu Dec 24 15:01:25 2020] intel-lpss 0000:00:15.0: Failed to add idma64, fallback to PIO
[Thu Dec 24 15:01:25 2020] ACPI Warning: SystemMemory range 0x0000000000000000-0x00000000000001FF conflicts with OpRegion 0x0000000000000000-0x0000000000000037 (\_SB.UBTC.USBC) (20190816/utaddress-204)
[Thu Dec 24 15:01:25 2020] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[Thu Dec 24 15:01:25 2020] intel-lpss: probe of 0000:00:15.0 failed with error -16
...

[dmesg.txt](https://github.com/projectacrn/acrn-hypervisor/files/5739839/dmesg.txt)
[boot_console_acrn-cfl.txt](https://github.com/projectacrn/acrn-hypervisor/files/5739845/boot_console_acrn-cfl.txt)

Thanks.

fuzhongl, 2021-01-05:
@YangLiang3 Could you please help to check this issue?
Thanks!

gvancuts, 2021-01-19:
Hi @ionutnechita , can you be more explicit about the errors you want to raise in this bug report?

ionutnechita, 2021-01-23:
Hi @gvancuts,

Yes. 
For HV this errors/warnings:
1. [37967034us][cpu=0][vm0:vcpu0][sev=2][seq=66]:vpci_write_cfg 0:1c.1 not found! off: 0xdb, val: 0xbf
[37977886us][cpu=0][vm0:vcpu0][sev=2][seq=67]:vpci_write_cfg 0:1c.3 not found! off: 0xdb, val: 0xbf

For kernel this errors/warnings:
1. [Thu Dec 24 15:01:25 2020] platform idma64.0: failed to claim resource 0: [mem 0x00000800-0x00000fff]
[Thu Dec 24 15:01:25 2020] intel-lpss 0000:00:15.0: Failed to add idma64, fallback to PIO
[Thu Dec 24 15:01:25 2020] ACPI Warning: SystemMemory range 0x0000000000000000-0x00000000000001FF conflicts with OpRegion 0x0000000000000000-0x0000000000000037 (\_SB.UBTC.USBC) (20190816/utaddress-204)
[Thu Dec 24 15:01:25 2020] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
[Thu Dec 24 15:01:25 2020] intel-lpss: probe of 0000:00:15.0 failed with error -16
...
2. RIP: 0010:get_mem_cgroup_from_mm+0x0/0xe0
handlers:
[<0000000037ad97c9>] i801_isr
Disabling IRQ #16
...

fuzhongl, 2022-03-22:
@ionutnechita 
Do you see  this   kernel errors/warnings for native?

fuzhongl, 2022-07-05:
Close it since no update log time.

closed 2022-07-05 by fuzhongl