projectacrn/acrn-hypervisor #5626: [CFL][industry] Host Call Trace once detected

ionutnechita, 2020-12-24:
Hi ACRN Team,

I restarted the SOS a few times, and I saw this.
I tried to reproduce again, but I didn't notice it again.

HV version 2.4-unstable-2020-12-24 12:15:55-703e0e2d DBG (daily tag:acrn-2020w51.5-180000p) industry@cfl-k700-i7
SOS: 
Multiboot2 Bootloader: GRUB2 2.04
OpenSUSE Tumbleweed

HV:
commit 703e0e2d52c39da339cd4085caf1ba5596abc32d (HEAD -> master, origin/master, origin/HEAD)
Author: fuzhongl <fuzhong.liu@intel.com>
Date:   Tue Dec 22 14:11:50 2020 +0800

    Doc: update Debian and Ubuntu as the User VM

    Pipe info isn't needed for IDV solution; so remove this part in
    the document.

    Signed-off-by: fuzhongl <fuzhong.liu@intel.com>

commit 529a2ac4976f8d6f9bc05f4fd40a7e3a76335e68
Author: Shuang Zheng <shuang.zheng@intel.com>
Date:   Wed Dec 16 01:21:32 2020 +0800

    acrn-config: config the entry number of vcpu_clos same as that of pcpu_id

    configure the entry number of vcpu_clos same as the entry number
    of pcpu_id to avoid sanity check error.

    Tracked-On: #5600
    Signed-off-by: Shuang Zheng <shuang.zheng@intel.com>
    Acked-by: Victor Sun <victor.sun@intel.com>

Kernel: 

commit 0dad2fa8edcbe288a1737d485959c5800caa47a7 (HEAD -> master, tag: acrn-2020w51.5-180000p, tag: acrn-2020w50.5-180000p, tag: acrn-2020w50.1-180000p, tag: acrn-2020w49.3-180000p, origin/master, origin/HEAD)
Author: Yonghua Huang <yonghua.huang@intel.com>
Date:   Tue Nov 24 17:25:13 2020 +0800

    vhm: validate input for acrn_ioreq_complete_request()

     Input 'vcpu' of function 'acrn_ioreq_complete_request()'
     is not validated and may overflow.

    Tracked-On: projectacrn/acrn-hypervisor#5548
    Signed-off-by: Yonghua Huang <yonghua.huang@intel.com>

commit ff4e0786d6b50c7d6cf3a8aaa15e2287adb19aff
Author: Yonghua Huang <yonghua.huang@intel.com>
Date:   Fri Nov 20 09:19:17 2020 +0800

    vhm: fix NULL pointer access in update_assigned_vf_state()

      pci_find_bus() may return null if invalid bdf
      value is injected, this patch checks its return
      value.

    Tracked-On: projectacrn/acrn-hypervisor#5535
    Signed-off-by: Yonghua Huang <yonghua.huang@intel.com>

Error:
...
36489193us][cpu=0][(null)][sev=2][seq=48]:Start VM id: 0 name: ACRN SOS VM
ACRN:\>
ACRN:\>[38284233us][cpu=0][vm0:vcpu0][sev=1][seq=51]:ept_misconfig_vmexit_handler, Guest linear address: 0x000000000b400000
[38296668us][cpu=0][vm0:vcpu0][sev=1][seq=52]:ept_misconfig_vmexit_handler, Guest physical address: 0x000000000b400000
[38309388us][cpu=0][vm0:vcpu0][sev=2][seq=53]:Assertion failed in file arch/x86/guest/ept.c,line 103 : fatal error
[38321591us][cpu=0][vm0:vcpu0][sev=2][seq=54]:
 delta = (actual_load_address - CONFIG_HV_RAM_START) = 0x0

[38333364us][cpu=0][vm0:vcpu0][sev=2][seq=55]:
Host Stack: CPU_ID = 0

[boot_console_acrn_with_call_trace.txt](https://github.com/projectacrn/acrn-hypervisor/files/5739886/boot_console_acrn_with_call_trace.txt)

fuzhongl, 2020-12-26:
@ionutnechita This issue is related with the passthrough device to Pre_launched VM.
Do you use the default hybrid_rt.xml when build acrn.bin?
If not, please share the scenario.xml you used to build acrn.bin.
Thanks!

ionutnechita, 2020-12-29:
Hi @fuzhongl,

I generated the ACRN Hypervisor binary like this:
make all BOARD=cfl-k700-i7 SCENARIO=industry RELEASE=0

ACRN Configuration Summary:
BOARD = cfl-k700-i7      SCENARIO = industry
BUILD type = DEBUG
Hypervisor configuration is based on:
        defconfig file:                 /tmp/acrn-hypervisor/misc/vm_configs/scenarios/industry/cfl-k700-i7/cfl-k700-i7.config;
        Others are set by default in:   /tmp/acrn-hypervisor/hypervisor/arch/x86/Kconfig;
VM configuration is based on:
        Source code at:                 /tmp/acrn-hypervisor/misc/vm_configs

fuzhongl, 2020-12-30:
@ionutnechita 
Please try to build with xml:
`make all BOARD_FILE=misc/vm_configs/xmls/board-xmls/cfl-k700-i7.xml SCENARIO_FILE=misc/vm_configs/xmls/config-xmls/cfl-k700-i7/industry.xml RELEASE=0`

ionutnechita, 2021-01-04:
Hi @fuzhongl,

I observe this issue with your command for build.

ionutnechita, 2021-01-04:
This is my cmdline for starting SOS:
Linux version 5.4.73-PKT-200203T060100Z (root@demo-acrn-opensuse-lab1) (gcc version 10.2.1 20201202 [revision e563687cf9d3d1278f45aaebd03e0f66531076c9] (SUSE Linux)) #1 SMP PREEMPT 

Command line: root=/dev/sda3 console=ttyS0 idle=halt rw rootwait console=tty0 consoleblank=0 no_timer_check quiet loglevel=3 i915.nuclear_pageflip=1 hvlog=2M@0xe00000 memmap=0x200000$0xe00000 maxcpus=8 root=PARTUUID=86378762-2bb1-4689-88ae-fe412e7a8357 hugepagesz=1G hugepages=28

fuzhongl, 2021-01-05:
@ionutnechita 
This issue can't be reproduced on our side, please boot into native and share `dmesg`  `cat /proc/iomem` log.
Thanks!

ionutnechita, 2021-01-06:
Hi @fuzhongl,

I try now with NUC board and Industrial scenario, and this issue appear on this platform ( KBL ).
NUC Platform: 
Intel(R) Client Systems NUC7i7BNH/NUC7i7BNB, BIOS BNKBL357.86A.0082.2020.0505.1614 05/05/2020

ACRN DM:
DM version is: 2.4-unstable-08bafdff-dirty (daily tag:acrn-2020w51.5-180000p), build by root@2021-01-06 14:27:49

[acrn-dm-version.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776193/acrn-dm-version.txt)
[dmesg-5.4.73-PKT-200203T060100Z.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776194/dmesg-5.4.73-PKT-200203T060100Z.txt)
[iomem.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776195/iomem.txt)
[ioports.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776196/ioports.txt)
[journalctl-5.4.73-PKT-200203T060100Z.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776197/journalctl-5.4.73-PKT-200203T060100Z.txt)
[lsblk.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776198/lsblk.txt)
[lscpu.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776199/lscpu.txt)
[mem-free.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776200/mem-free.txt)

XML Configuration:
[nuc7i7dnb.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776209/nuc7i7dnb.xml.txt)
[industry.xml.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776214/industry.xml.txt)

fuzhongl, 2021-01-07:
@ionutnechita Could you please try acrn v2.3?
Thanks!

ionutnechita, 2021-01-07:
Hi @fuzhongl,

Can you tell me why you closed this ticket?

ionutnechita, 2021-01-07:
I will try to build on 2.3 today.

ionutnechita, 2021-01-07:
This issue/problem is present in 2.3.
Call Trace and SOS can not booting.
After 32 SOS reboots.

gvancuts, 2021-01-07:
Hi @ionutnechita , I think I had missed an important detail in your bug description and that it's not 100% reproductible. You mentionned this happened after 32 Service VM reboots in your last comment. Is this the typical rate of occurrence?

@fuzhongl , you mentionned above you are not able to reproduce this. Have you tried with multiple reboots as well? (Or did you miss this subtlety just like I did :-)

fuzhongl, 2021-01-08:
@ionutnechita sorry for close it ticket by mistake.
CFL is very limited on our side, What's the the typical rate of occurrence for KBL NUC? 
@gvancuts Yes, I also missed it like you did: not aware this stability related issue.
Will try to reproduce on our side.

ionutnechita, 2021-01-08:
Hi @fuzhongl,

I try now with ACRN HV 2.3.
> What's the the typical rate of occurrence for KBL NUC?

On KBL NUC, with multiple restart, it happens to be quite often.
In the range of 100, after 6-7 reboot is stuck or after 32-33 is stuck.

fuzhongl, 2021-01-15:
@ionutnechita This issue is reproduced on our side. 
Please add `nokalsr` in sos kernel command line for workaround.
We will narrow down and fix it later.
Thanks!

fuzhongl, 2021-01-19:
[External_System_ID]ACRN-6933

ionutnechita, 2021-02-03:
Hi @fuzhongl, @gvancuts

I tested it today, and I notice this issue again. ( using kaslr parameter )

uname -a
Linux demo-acrn-opensuse-labl1 5.4.90-PKT-200203T060100Z-g42a5e48c3e11 #4 SMP PREEMPT Wed Feb 3 13:05:20 EET 2021 x86_64 x86_64 x86_64 GNU/Linux
acrn-dm --version
DM version is: 2.4-unstable-94a980c9-dirty (daily tag:acrn-2021w05.5-180000p), build by root@2021-02-02 20:42:05

cat /proc/cmdline
root=/dev/sda2 console=ttyS0 idle=halt rw rootwait console=tty0 kaslr consoleblank=0 no_timer_check quiet loglevel=3 i915.nuclear_pageflip=1 hvlog=2M@0xe00000 memmap=0x200000$0xe00000 maxcpus=4 root=PARTUUID=d444e5a8-ef96-48da-a81b-d8c450b39b35 hugepagesz=1G hugepages=28

Files:
[acrn.zip](https://github.com/projectacrn/acrn-hypervisor/files/5920233/acrn.zip)

[nuc7i7dnb.xml-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920335/nuc7i7dnb.xml-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[lspci-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920336/lspci-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[lscpu-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920337/lscpu-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[kallsyms-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920338/kallsyms-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[journalctl-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920339/journalctl-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[ioports-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920340/ioports-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[iomem-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920341/iomem-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[industry.xml-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920342/industry.xml-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[hwinfo-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920343/hwinfo-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[dmesg-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920344/dmesg-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)
[config-kernel-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt](https://github.com/projectacrn/acrn-hypervisor/files/5920345/config-kernel-5.4.90-PKT-200203T060100Z-g42a5e48c3e11.txt)

jsun26intel, 2021-02-08:
@ionutnechita can you still see this issue with nokaslr parameter?

ionutnechita, 2021-02-08:
Hi @gvancuts, @jsun26intel 

I tested now with nokaslr and not observe this behaviour.
Thanks for this parameter,

In this moment my issue is resolved.

gvancuts, 2021-02-08:
Thanks for the confirmation @ionutnechita , closing this ticket.

closed 2021-02-08 by gvancuts

fuzhongl, 2021-03-29:
Add `nokalsr` in sos kernel command line for workaround.
The formal fix is under review.