projectacrn/acrn-hypervisor #5637: [ACRN-DM] Segmentation fault starting VM that does not exist

ionutnechita, 2021-01-06:
Hi ACRN Team,

I tried to start a virtual machine without any configuration or script.
From what I noticed, the acrn-dm tool does not check first if there is any previous configuration or stored on disk.
The acrn-dm process stopped but with an internal program error. ( Segmentation fault (core dumped) )
I think acrn-dm should do a check before starting a virtual machine.

Scenario: All scenario, but using industrial ACRN  HV in this moment.
DM version is: 2.4-unstable-08bafdff-dirty (daily tag:acrn-2020w51.5-180000p), build by root@2021-01-06 14:27:49

Error:

acrn-dm:
acrn-dm test-vm-not-exist
vm_create: test-vm-not-exist
VHM api version 1.0
vm_setup_memory: size=0x10000000
...
vm_init_vdevs
No correct pm notify channel given
Listening 0...
polling 0...
Segmentation fault (core dumped)

dmesg:
[  165.394254] acrn-dm[1159]: segfault at 8 ip 000055b0724e9941 sp 00007ffd29b2f5f0 error 4 in acrn-dm[55b0724c1000+51000]

[segfault-output-acrn-dm.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776936/segfault-output-acrn-dm.txt)
[dmesg-segfault-acrn-dm.txt](https://github.com/projectacrn/acrn-hypervisor/files/5776937/dmesg-segfault-acrn-dm.txt)

gvancuts, 2021-01-25:
@NanlinXie , can we assign someone to this?

A quick run in gdb tells me it's happening at this line: https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/hw/pci/core.c#L1448 (`bus0_memlimit = bi->memlimit32;`)

NanlinXie, 2021-01-26:
@shuox , could you help to check this issue? Thanks!

ionutnechita, 2021-02-03:
Any update here?