projectacrn/acrn-hypervisor #5653: top level ''make clean'' reports errors

dbkinder, 2021-01-20:
On the main branch, at the top level, ``make clean`` reports some errors:
```
~/projectacrn/acrn-hypervisor$ make clean
make -C /home/dbkinder/projectacrn/acrn-hypervisor/misc OUT_DIR=/home/dbkinder/projectacrn/acrn-hypervisor/build/misc/tools clean
make[1]: Entering directory '/home/dbkinder/projectacrn/acrn-hypervisor/misc'
make -C /home/dbkinder/projectacrn/acrn-hypervisor/misc/tools/acrn-crashlog OUT_DIR=/home/dbkinder/projectacrn/acrn-hypervisor/build/misc/tools clean
make[2]: Entering directory '/home/dbkinder/projectacrn/acrn-hypervisor/misc/tools/acrn-crashlog'
Package libsystemd was not found in the pkg-config search path.
Perhaps you should add the directory containing `libsystemd.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libsystemd' found
Package libtelemetry was not found in the pkg-config search path.
Perhaps you should add the directory containing `libtelemetry.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libtelemetry' found
. . .

(no other errors reported)

```

gvancuts, 2021-01-21:
> Package libsystemd was not found in the pkg-config search path.
> Perhaps you should add the directory containing `libsystemd.pc'
> to the PKG_CONFIG_PATH environment variable
> No package 'libsystemd' found

You're missing the `libsystemd-dev` dependency on your system. It should have been installed as per https://projectacrn.github.io/latest/getting-started/building-from-source.html#install-build-tools-and-dependencies, or do you use a different environment?

> Package libtelemetry was not found in the pkg-config search path.
> Perhaps you should add the directory containing `libtelemetry.pc'
> to the PKG_CONFIG_PATH environment variable
> No package 'libtelemetry' found

`libtelemetry` really is a Clear Linux thing if I'm not mistaking. We could/should remove this actually...

gvancuts, 2021-01-21:
> `libtelemetry` really is a Clear Linux thing if I'm not mistaking. We could/should remove this actually...

Just submitted a PR (see #5655) if we care to remove this (which I think we should).

ionutnechita, 2021-01-23:
Hi @gvancuts,

This "No package 'libtelemetry' found" is optionally.
Without libtelemetry, compiling is working.
It would be better to remain optional, and not to remove it altogether.

In OpenSUSE Tumbleweed, library for telemetry exist.
For install: zypper in -y telemetrics-client telemetrics-client-devel

Thanks.

fuzhongl, 2021-01-25:
[External_System_ID]ACRN-6939

gvancuts, 2021-01-25:
> This "No package 'libtelemetry' found" is optionally.
> Without libtelemetry, compiling is working.

Yes, making this optional is something we did back in the days specifically to avoid compilation errors on systems that did not ship this (e.g. Ubuntu, Fedora, etc.).

> It would be better to remain optional, and not to remove it altogether.

OK, we can certainly keep it but I'm wondering if anyone uses this, do you, or do you know of people using it?

> 
> In OpenSUSE Tumbleweed, library for telemetry exist.
> For install: zypper in -y telemetrics-client telemetrics-client-devel

This is good to know. I wasn't aware of the fact it was packaged up for OpenSUSE Tumbleweed. Despite this, I'm a little cautious as the code seems to not be much maintained anymore.

gvancuts, 2021-01-28:
Patch was merged, closing this.

closed 2021-01-28 by gvancuts