projectacrn/acrn-hypervisor #5660: Improve life_mngr Makefile

gvancuts, 2021-01-26:
`life_mngr` Makefile should be improved to reflect better what needs to be installed and how. The entire picture is not clear to me so what I would like to review and agree on in this feature request is what should be built, how and what should be installed. Here are a few initial thoughts (with some assumptions) and some questions:
* Copying the `life_mngr.service` file to the build directory shouldn't be done in the `all` target
* I understand that service file is **only** to be used in a Linux-based User VM, is that correct?
* The `life_mngr` binary needs to be installed in both Service **and** User VM, is that correct?
* How is the `life_mngr` run in the Service VM?
* I think we should have two different `install` targets, one for a User VM and one for the Service VM
* The `install-uservm` target (exact name to be defined) would also install the `*.service` file and enable the service
* The classical `install` target would only copy the binary in the right place (is that all that's needed)?
* Possibly create a third `install` target for Windows that wouldn't really run on Windows but would create a zip file with the `*.exe` and `COPYING.MinGW-w64-runtime.txt` file in it. That zip could be transferred manually to the WaaG and installed there.
* Only the `install` target that does what is relevant in the Service VM would be called by default. Other `install` targets would be invoked manually.
Thoughts?

gvancuts, 2021-03-15:
[External_System_ID] ACRN-7018

gvancuts, 2021-05-07:
Additional thoughts:
* Since `life_mngr` starts in both the Service VM and User VM, we need separate install `make` targets for each and also a separate systemd service file for each
* We probably need to be able to coordinate the socket used by `acrn-dm` and `life_mngr` somehow... but I really don't know how to do that.

dbkinder, 2022-01-26:
@gvancuts @NanlinXie What's the status on this and target milestone?

closed 2022-03-04 by fuzhongl