projectacrn/acrn-hypervisor #5671: [ACRN] compile not working with master branch

ionutnechita, 2021-01-28:
Hi ACRN Team,

I try to compile new ACRN, but observe one issue.

Error:
mapi.o /workspace/acrn-hypervisor/build/devicemodel/core/mptbl.o /workspace/acrn-hypervisor/build/devicemodel/core/main.o /workspace/acrn-hypervisor/build/devicemodel/core/hugetlb.o /workspace/acrn-hypervisor/build/devicemodel/core/vrpmb.o /workspace/acrn-hypervisor/build/devicemodel/core/timer.o /workspace/acrn-hypervisor/build/devicemodel/arch/x86/pm.o /workspace/acrn-hypervisor/build/devicemodel/arch/x86/power_button.o /workspace/acrn-hypervisor/build/devicemodel/log/log.o /workspace/acrn-hypervisor/build/devicemodel/log/kmsg_logger.o /workspace/acrn-hypervisor/build/devicemodel/log/disk_logger.o -lrt -lpthread -lcrypto -lpciaccess -luuid -lusb-1.0 -lacrn-mngr
/usr/lib64/gcc/x86_64-suse-linux/10/../../../../x86_64-suse-linux/bin/ld: /workspace/acrn-hypervisor/build/devicemodel/hw/pci/virtio/virtio_gpio.o:/workspace/acrn-hypervisor/devicemodel/include/acpi.h:61: multiple definition of `__packed'; /workspace/acrn-hypervisor/build/devicemodel/hw/platform/atkbdc.o:/workspace/acrn-hypervisor/devicemodel/include/acpi.h:61: first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:191: /workspace/acrn-hypervisor/build/devicemodel/acrn-dm] Error 1
make[1]: Leaving directory '/workspace/acrn-hypervisor/devicemodel'
make: *** [Makefile:108: devicemodel] Error 2

OS:
 cat /etc/os-release
NAME="openSUSE Tumbleweed"
VERSION="20210127"
ID="opensuse-tumbleweed"
ID_LIKE="opensuse suse"
VERSION_ID="20210127"
PRETTY_NAME="openSUSE Tumbleweed"
ANSI_COLOR="0;32"
CPE_NAME="cpe:/o:opensuse:tumbleweed:20210127"
BUG_REPORT_URL="https://bugs.opensuse.org"
HOME_URL="https://www.opensuse.org/"
DOCUMENTATION_URL="https://en.opensuse.org/Portal:Tumbleweed"
LOGO="distributor-logo"

gcc --version
gcc (SUSE Linux) 10.2.1 20201202 [revision e563687cf9d3d1278f45aaebd03e0f66531076c9]
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

ldd --version
ldd (GNU libc) 2.32
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

iasl -v

Intel ACPI Component Architecture
ASL+ Optimizing Compiler/Disassembler version 20210105
Copyright (c) 2000 - 2021 Intel Corporation

commit d55ab8733152450c98dae3cf348f49d7d5863a0f (HEAD -> master, origin/master, origin/HEAD)
Author: Geoffroy Van Cutsem <geoffroy.vancutsem@intel.com>
Date:   Thu Jan 28 11:54:59 2021 +0100

    Makefile: make internal comment in doc Makefile silent

    There is a comment in the doc/Makefile that is beinhg spit out
    when calling 'make clean'. This is harmless but can be confusing
    to users so let's make it silent.

    Tracked-On: #5669
    Signed-off-by: Geoffroy Van Cutsem <geoffroy.vancutsem@intel.com>

commit c9ca23d268e111725da68ebabd70925737bc2b01
Author: Yonghua Huang <yonghua.huang@intel.com>
Date:   Wed Jan 27 10:19:23 2021 +0800

    hv: refine RTCM initialization code

     - RTCM is initialized in hypervisor only
       if RTCM binaries are detected.
     - Remove address space of RTCM binary from
       Software SRAM region.
     - Refine parse_rtct() function, validity of
       ACPI RTCT table shall be checked by caller.

    Tracked-On: #5649
    Signed-off-by: Yonghua Huang <yonghua.huang@intel.com>
    Reviewed-by: Fei Li <fei1.li@intel.com>
    Acked-by: Eddie Dong <eddie.dong@intel.com>

command:
make all  BOARD_FILE=./misc/config_tools/data/cfl-k700-i7/cfl-k700-i7.xml SCENARIO_FILE=./misc/config_tools/data/cfl-k700-i7/industry.xml RELEASE=0 &> log.txt

[log.txt](https://github.com/projectacrn/acrn-hypervisor/files/5889409/log.txt)

gvancuts, 2021-01-29:
I can reproduce this, this is an issue triggered by gcc 10.x's new default to `-fno-common`.

There are 2 ways we can fix this:
```
$ git diff
diff --git a/devicemodel/include/acpi.h b/devicemodel/include/acpi.h
index e01744b4..20347585 100644
--- a/devicemodel/include/acpi.h
+++ b/devicemodel/include/acpi.h
@@ -58,7 +58,7 @@ struct acpi_table_hdr {
        char                    asl_compiler_id[4];
        /* ASL compiler version */
        uint32_t                asl_compiler_revision;
-} __packed;
+} __attribute__((__packed__));
 
 /* All dynamic table entry no. */
 #define NHLT_ENTRY_NO          8

```

Or

```
$ git diff
diff --git a/devicemodel/Makefile b/devicemodel/Makefile
index d4157ba6..e3fa286d 100644
--- a/devicemodel/Makefile
+++ b/devicemodel/Makefile
@@ -24,6 +24,7 @@ CFLAGS += -Wformat -Wformat-security -fno-strict-aliasing
 CFLAGS += -fno-delete-null-pointer-checks -fwrapv
 CFLAGS += -fpie
 CFLAGS += -Wno-stringop-truncation -Wno-address-of-packed-member
+CFLAGS += -fcommon
 
 CFLAGS += -I$(BASEDIR)/include
 CFLAGS += -I$(BASEDIR)/include/public
```

I don't understand the implications of either of these to determine which is preferable... but if someone can help me answer that question, I'm happy to raise a PR.

In the meantime, can you quickly confirm that these solve the issue on your side too? @ionutnechita

ionutnechita, 2021-01-29:
Hi @gvancuts,

First and second variant working for me.

Thanks for reproduce this issue.

gvancuts, 2021-01-29:
We have many `__attribute__((packed))` instances in the code already. The few `__packed` ones left are "fixed" by a `#define __packed __attribute__((packed))` line in the code.

The only ones which do not seem to be touched are the ones in `devicemodel/include/rtct.h`... and I don't know why they are not causing any issue. But perhaps we should also change those to `__attribute__((packed))`?

fuzhongl, 2021-02-01:
[External_System_ID] ACRN-6947

gvancuts, 2021-02-01:
Patch was merged, closing.

closed 2021-02-01 by gvancuts

ionutnechita, 2021-02-01:
I will try tomorrow to test the master branch again.

Thanks.