projectacrn/acrn-hypervisor #5688: Can I use nuc7i7dnb config files for NUC7i5DNB boad?

SIMING29, 2021-02-02:
There are no board and scenario config files for NUC7i5DNB in source code. So, can I use the nuc7i7dnb config files for my board? Or should I modify the config files according to the difference of processors?

Thanks.

gvancuts, 2021-02-02:
There is a fair chance it will work as-is. Take a look though at things like number of cores and memory size. If you are developing serious on this, I would recommend that you create a specific config for your board at some point, but that may not be required to get this up and running for the first time.

SIMING29, 2021-02-02:
ok,I will try that first, thanks.

closed 2021-02-02 by SIMING29

SIMING29, 2021-02-02:
Hi @gvancuts 
If I want to creat my board config file, can I only get it by manually modifying the file?  Is there a tool to make board config file for specific board?

gvancuts, 2021-02-02:
There is a tool for that, its use is described here: https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html

Please note that it is undergoing pretty invasive surgery at the moment (for the better). What is in our documentation reflects what the latest release includes (i.e. v2.3). We will update the docs once the new version has landed (various patches already have).

gvancuts, 2021-02-02:
If you don't need to modify the scenario (i.e. number and type of VMs), you can jump directly to this section that explains how to collect the board info: https://projectacrn.github.io/latest/tutorials/acrn_configuration_tool.html#board-and-vm-configuration-workflow