projectacrn/acrn-hypervisor #5690: [SerialConsole ACRN]  Added support in ACRN HV for usbserial/pl2303

ionutnechita, 2021-02-02:
Hi ACRN Team,

Is it possible to use a serial converter to be able to see the ACRN HV console?

[ 1942.614512] usb 1-1: USB disconnect, device number 2
[ 1943.314580] usb 1-2: USB disconnect, device number 3
[ 1943.314588] usb 1-2.1: USB disconnect, device number 5
[ 1946.176054] usb 1-2: new full-speed USB device number 6 using xhci_hcd
[ 1946.325238] usb 1-2: New USB device found, idVendor=067b, idProduct=2303, bcdDevice= 3.00
[ 1946.325246] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1946.325250] usb 1-2: Product: USB-Serial Controller
[ 1946.325254] usb 1-2: Manufacturer: Prolific Technology Inc.
[ 1946.353523] usbcore: registered new interface driver usbserial_generic
[ 1946.353537] usbserial: USB Serial support registered for generic
[ 1946.359593] usbcore: registered new interface driver pl2303
[ 1946.359630] usbserial: USB Serial support registered for pl2303
[ 1946.359672] pl2303 1-2:1.0: pl2303 converter detected
[ 1946.361110] usb 1-2: pl2303 converter now attached to ttyUSB0

Thanks.

gvancuts, 2021-02-03:
Hi @ionutnechita , the last time I checked that (well, over a year ago), it was not possible. This is something we had explored as it would really help when people have platforms that do not expose a native serial port. As far as I know though, it is still not possible (and I don't even know if this can easily be added).

ionutnechita, 2021-02-03:
Hi @gvancuts,

I understand, so it would be very complicated to implement this function.
Good thing you tested, now I know an answer.
I think that's enough for me.

Thank you for helping me.

Ticket can be closed.

gvancuts, 2021-02-04:
Thanks @ionutnechita , closing this.

closed 2021-02-04 by gvancuts