projectacrn/acrn-hypervisor #5700: IVSHMEM with Zephyr not working

hidrol, 2021-02-03:
We are trying to use the IVSHMEM Device with following setting:

**Hardware:**
EHL A0 Board

**Software:**
Logical Partition Mode with Yocto

**VM0:**
Linux RT

**VM1:**
Zephyr

The xml scenario file regarding ivshmem looks like this:

```
<IVSHMEM desc="IVSHMEM configuration">
	    <IVSHMEM_ENABLED desc="Enable Share Memory between VMs by IVSHMEM.">y</IVSHMEM_ENABLED>
            <IVSHMEM_REGION desc="the name, size, colon separated IDs of communication VMs of share memory region, separated by comma like this: hv:/shm_region_0, 2, 0:2">hv:/shm_prl1_prl2, 2, 0:1</IVSHMEM_REGION>
            <IVSHMEM_REGION desc="the name, size, colon separated IDs of communication VMs of share memory region, separated by comma like this: hv:/shm_region_0, 2, 0:2">hv:/shm_prl2_prl1, 2, 0:1</IVSHMEM_REGION>
	    </IVSHMEM>
```

**Zephyr prj.conf file:**

```
CONFIG_PCIE=y
CONFIG_PCIE_MSI=y
CONFIG_PCIE_MSI_X=y
CONFIG_PCIE_MSI_MULTI_VECTOR=y
CONFIG_VIRTUALIZATION=y
CONFIG_IVSHMEM=y
```

**Zephyr Kconf in drivers/virtualization:**

```
# Virtualization drivers configuration options

# Copyright (c) 2015-2020 Intel Corporation
# SPDX-License-Identifier: Apache-2.0

menuconfig VIRTUALIZATION
	bool "Virtualization guests drivers"
	help
	  This contains various drivers meant to support and expose features
	  when Zephyr is running as a guest in a virtualized or emulated
	  environment.

if VIRTUALIZATION

config IVSHMEM
	bool "Inter-VM shared memory device (ivshmem)"
	depends on PCIE
	help
	  This will enable support of qemu's ivshmem device, which is also
	  present in ACRN hypervizor, and lets VM sharing memory with each
	  other.

if IVSHMEM

config IVSHMEM_DEV_NAME
       string
       default "IVSHMEM"

module = IVSHMEM
module-str = ivshmem
source "subsys/logging/Kconfig.template.log_config"

config IVSHMEM_DOORBELL
	bool "Support interrupt based ivshmem (doorbell version)"
	depends on PCIE_MSI_X && PCIE_MSI_MULTI_VECTOR
	help
	  This will enable support of ivshmem-doorbell, i.e. the interrupt
	  based ivshmem.

config IVSHMEM_MSI_X_VECTORS
	int "How many notification vectors should be pre-allocated?"
	default 2
	depends on IVSHMEM_DOORBELL
	help
	  MSI-X vector holders must be pre-allocated. One can pre-allocate
	  more or less than necessary. Depends on how many VMs will connect
	  with each other. These are know to be the notification vectors in
	  ivshmem.

config IVSHMEM_INT_PRIORITY
	int "Interrupt priority"
	default 2
	depends on IVSHMEM_DOORBELL
	help
	  Interrupt priority used for the MSI-X generated interrupts.

config IVSHMEM_SHELL
	bool "Enable IVshmem shell module"
	depends on SHELL
	help
	  This is mostly a module to help getting info the ivshmem and/or
	  quickly testing your ivshmem client application (from another VM).

endif # IVSHMEM

endif # VIRTUALIZATION
```

**Zephyr user application c code:**

```
#include <zephyr.h>
#include <drivers/virtualization/ivshmem.h>
#include <sys/printk.h>
#include <string.h>
#include <device.h>

int main()
{
	printk("IVSHMEM_TEST\n");
	struct device *ivshmem;
	printk("1\n");
	uintptr_t mem;
	printk("2\n");
	size_t size;
	printk("3\n");
	char message[] = "Hello from Zephyr!";	
	printk("4\n");
	ivshmem = device_get_binding(CONFIG_IVSHMEM_DEV_NAME);
	printk("5\n");
	size = ivshmem_get_mem(ivshmem, &mem);
	printk("6\n");
	if (size < 0){ 
		printk("Could not get Shared Memory Adress %s\n", CONFIG_BOARD);
		return -1;
	}
	else{
		strcpy((char*)mem, message);
		printk("Memory written, Message: %s \n", message);
	}

	return 0;
}
```

When we run this application on qemu it works without problems, on acrn instead it gets stuck in 
`size = ivshmem_get_mem(ivshmem, &mem);`

We are not quite sure how we must edit the xml or Kconfig files in ACRN or Zephyr.

yonghuah, 2021-02-03:
@hidrol  can you list all the PCI devices detected in Zephyr?

thanks.
-Yonghua

hidrol, 2021-02-03:
We are quite new in working zephyr. How do we check for PCI Devices in Zephyr?

gvancuts, 2021-02-03:
Have you tested the `ivshmem` device using Zephyr IVshmem module (from the shell)?

wilky93, 2021-02-03:
Hi, I am working with hidrol on the same project.

What do you mean exactly? We are not using shell in Zephyr. We boot ACRN, with "vm_console 0" we get into Zephyr VM. There the application is executed immediately. And it gets stucked as described above. 

If we run Zephyr in qemu, then the application does not get stucked.

Application and config-files are described above.
We also do not use virt_ivshmem_shell.c

hidrol, 2021-02-03:
Also, we use the plain ivshmem version. The ACRN hypervisor version (Ver 2.2) we use does not support ivshmem with interrupt functionality. Could this maybe be the problem?

hidrol, 2021-02-03:
@gvancuts @yonghuah We have 2 IVSHMEM devices defined in the xml file. How does zephyr know which one to chose?

yonghuah, 2021-02-04:
@hidrol Doorbell (interrupt) of ivshmem device is optional, ivshmem device should work without it.

hidrol, 2021-02-04:
@yonghuah I just wanted to make sure that there are no incompatibilities between our ACRN and Zephyr Versions, because at the time our ACRN revision launched IVSHMEM wasn't yet implemented in Zehphyr.

hidrol, 2021-02-04:
With the following code we checked if the device is being found. Turns out Zephyr can not find the ivshmem device.
```
if (!ivshmem) {
                printk("IVshmem device cannot be found");
                return -1;
}
```

gvancuts, 2021-02-04:
> @gvancuts @yonghuah We have 2 IVSHMEM devices defined in the xml file. How does zephyr know which one to chose?

That's a good question... I do not know. Could you perhaps try with only one such device in case this is what confuses Zephyr?

Do you see these `ivshmem` devices in the Yocto VM (ID 1)?

hidrol, 2021-02-05:
Fortunately a patch from zephyr could fix the problem.

gvancuts, 2021-02-05:
> Fortunately a patch from zephyr fixed the problem.

Very good to hear! Care to indicate which one it is that solved it?

And please close this ticket as soon as you're happy with the resolution and if there is no further action required.

hidrol, 2021-02-05:
Sure, this is the patch: https://github.com/zephyrproject-rtos/zephyr/pull/32020
@gvancuts  @yonghuah thanks for the quick replies and your support.

closed 2021-02-05 by hidrol

yonghuah, 2021-02-07:
@hidrol , 

Copy my comment from below:
https://github.com/zephyrproject-rtos/zephyr/pull/32020

ACRN emulates MMIO/Register BAR for ivshmem device even for ivshmem-plain mode, but recent we catch a mismatch that ACRN set MMIO BAR size to 4096Bytes, but per the IVSHMEM device spec, this MMIO BAR size should be 256bytes, fixing of this issue is on the way.

With above mismatch bar size, Some guests (such as Windows) may refuse to regard it as a valid ivshmem MMIO/Reg BAR.

So could you double check why Zephr did not detect MMIO BAR for ivshmem device? please contact ACRN team if co-work is required on this.

thanks.
-Yonghua