projectacrn/acrn-hypervisor #5702: Documentation about GVT (Graphics Sharing) is outdated

gvancuts, 2021-02-04:
Our GVT-{g/d} documentation is largely outdated. There are at least these documents that should be looked at for an update:
- https://projectacrn.github.io/latest/user-guides/kernel-parameters.html#intel-gvt-g-acrngt-parameters
- https://projectacrn.github.io/latest/developer-guides/hld/hld-APL_GVT-g.html

gvancuts, 2021-02-23:
@NanlinXie , I have updated the kernel parameters doc.

Can you assign someone to look at https://projectacrn.github.io/latest/developer-guides/hld/hld-APL_GVT-g.html? I have also heard that we use something called `idv` nowadays. We should add details about this too.

amyreye, 2022-02-17:
@gvancuts It looks like the page above was removed. What's the status of this issue?

closed 2022-06-28 by gvancuts