projectacrn/acrn-hypervisor #5704: [OVMF] Boot with Zephyr not working

ionutnechita, 2021-02-04:
Hi ACRN Team,

I create environment with Zephyr, but vm not started.
Flag VM is Init, not Started or Created.

Boot for ./launch_zephyr.sh

cpu1 online=1
cpu2 online=1
cpu3 online=1
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: zephyr_vm1
VHM api version 1.0
vm_setup_memory: size=0x8000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/zephyr_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/zephyr_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:0/65 page size:0x200000
level 1 free/need pages:28/0 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 65 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
        level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
total_size 0x140200000

mmap ptr 0x0x7f4f5cbb3000 -> baseaddr 0x0x7f4f5cc00000
mmap 0x8000000@0x7f4f5cc00000
touch 64 pages with pagesz 0x200000
mmap 0x200000@0x7f505ca00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x8000000, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
vm_init_vdevs
No correct pm notify channel given
pci init hostbridge
pci init lpc
pci init virtio-blk
pci init virtio-console
***********************************************
                                               virt-console backend redirected to /dev/pts/1
                                                                                            ***********************************************
                                                                                                                                           tpm: init_vtpm2:Invalid socket path!
                                                                                                                                                                               polling 34...
                                                                                                                                                                                            Listening 34...
                                                                                                                                                                                                           /tmp/dm.Xgqn0ez     24:     Device (PCI0)
       Warning  3073 -        Multiple types ^  (Device object requires either a _HID or _ADR, but not both)

                                                                                                            /tmp/dm.Xgqn0ez    785:     Processor (CPU0, 0x00, 0x00000000, 0x00) {}
                                                                                                                                                                                   Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                                                   /tmp/dm.Xgqn0ez    786:     Processor (CPU1, 0x01, 0x00000000, 0x00) {}
                                                                                                                          Warning  3168 -                          ^ Legacy Processor() keyword detected. Use Device() keyword instead.

                                                                                                                                                                                                                                       acrn_sw_load
      SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 0x200000 copied to addr 0x7f505ca00000
                                                                                                      SW_LOAD: build e820 7 entries to addr: 0x7f4f5ccef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x0000000000100000, size 0x0000000007f00000,  type 0x1
SW_LOAD: entry[2]: addr 0x000000003b800000, size 0x0000000004004000,  type 0x2
SW_LOAD: entry[3]: addr 0x000000007f800000, size 0x0000000000800000,  type 0x2
SW_LOAD: entry[4]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[5]: addr 0x0000000140000000, size 0x0000000000000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: ovmf_entry 0xfffffff0
                              add_cpu
> and wait more and stuck in this moment.

I think the problem would be with the virtual bios. /usr/share/acrn/bios/OVMF.fd
Can you help me with this?

OS:
OpenSUSE Tumbleweed
acrn-dm --version
DM version is: 2.4-unstable-94a980c9-dirty (daily tag:acrn-2021w05.5-180000p), build by root@2021-02-04 11:08:46

commit 94a980c923cb235ccdb7bf62c13ea86ff90aca05 (HEAD -> master, origin/master, origin/HEAD)
Author: Li Fei1 <fei1.li@intel.com>
Date:   Mon Feb 1 11:29:14 2021 +0800

    hv: hypercall: prevent sos can touch hv/pre-launched VM resource

    Current implementation, SOS may allocate the memory region belonging to
    hypervisor/pre-launched VM to a post-launched VM. Because it only verifies
    the start address rather than the entire memory region.

    This patch verifies the validity of the entire memory region before
    allocating to a post-launched VM so that the specified memory can only
    be allocated to a post-launched VM if the entire memory region is mapped
    in SOS’s EPT.

    Tracked-On: #5555
    Signed-off-by: Li Fei1 <fei1.li@intel.com>
    Reviewed-by: Yonghua Huang  <yonghua.huang@intel.com>

commit 0b6840d1be927023d808b798fa6ae1ff8803ec68
Author: Xie, nanlin <nanlin.xie@intel.com>
Date:   Tue Feb 2 23:07:21 2021 +0800

    acrn-config: Update generated configuration source code

    1.Reorg generated configuration source code structure
    2.Upstream generated configuration source code based on generic board infomation
    3.Update license date from 2020 to 2021

    Tracked-On: #5644
    Signed-off-by: Xie, nanlin <nanlin.xie@intel.com>

...
[   62.856294] IRQ 125: no longer affine to CPU1
[   62.857574] smpboot: CPU 1 is now offline
[   63.861164] vhm: try to offline cpu 1 with lapicid 2
[   63.881516] IRQ 123: no longer affine to CPU2
[   63.883231] smpboot: CPU 2 is now offline
[   64.885449] vhm: try to offline cpu 2 with lapicid 1
[   64.903123] IRQ 128: no longer affine to CPU3
[   64.904414] smpboot: CPU 3 is now offline
[   65.907416] vhm: try to offline cpu 3 with lapicid 3
[   65.931156] vhm_dev_open: opening device node
[   65.931660] vhm-ioreq: init request buffer @ 00000000122dae95!
[   65.931663] vhm-ioreq: created ioreq client 1 for ioeventfd-1
[   65.931702] ACRN vhm ioeventfd init done!
[   65.931706] ACRN vhm irqfd init done!
[   65.931708] vhm: VM 1 created
[   66.000111] vhm-ioreq: created ioreq client 2 for acrndm

ionutnechita, 2021-02-04:
Is possible to view grub console/prompt with acrnctl?

gvancuts, 2021-02-04:
Just noticed this on the mailing list (from @jsun26intel)
* https://lists.projectacrn.org/g/acrn-dev/topic/patch_1_2_dm_declare/80374574?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,80374574
* https://lists.projectacrn.org/g/acrn-dev/topic/patch_2_2_dm_remove_adr/80374575?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,80374575
You have these error messages on your system, not sure if this is what's stopping the boot though...

ionutnechita, 2021-02-04:
Hi @gvancuts,

I tried to setup from this wiki page:
https://projectacrn.github.io/latest/tutorials/using_zephyr_as_uos.html
But I didn't see anything appear on the console. ( acrnctl start launch_zephyr )

Using the HV ACRN console: 
vm console 2
I saw that it remained in EFI Shell.
I have to give commands in the UEFI shell to start the Zephyr application.

FS0:
cd efi
cd boot
SHELL:> grub_x86_64.efi
press enter in GRUB for Zephyr Kernel
and the application started.
Hello World! acrn

From what i said, the wiki is not complete to start an Zephyr application from the first.
Stays in SHELL and does not enter in GRUB.

Can the issue be fixed in the wiki?

gvancuts, 2021-02-04:
It used to work just fine and start Zephyr automatically. Perhaps there is a regression in newer OVMF, I'll try to reproduce this on my side.

In the meantime, I noticed that you use `grub_x86_64.efi` (in the `zephyr.img`, under `efi/boot`), can you rename that file into `bootx64.efi` (as in the tutorial). One possibility for what you're observing is that the `OVMF.fd` firmware is automatically picking up `grubx64.efi`, but **not** `grub_x86_64.efi`).

gvancuts, 2021-02-04:
> Is possible to view grub console/prompt with acrnctl?

I guess you're asking about the Grub from within the VM, aren't you?

I must admit I never use `acrnctl` myself so I don't know... :-) Which actually makes me wonder, do you use `acrnctl` extensively?

ionutnechita, 2021-02-04:
Yes, I use the acrnctl command more.

For the name file issue, I'll try again tomorrow.
grub_x86_64.efi -> grubx64.efi to rename in zephyr.img

If you have time, try to reproduce this problem too.

Thanks.

gvancuts, 2021-02-04:
I'll try this tomorrow (I need to set-up another system from scratch for that... too late for that today ;-)

gvancuts, 2021-02-05:
Hi @ionutnechita , I got around to testing this today. I can reproduce your problem, and it is solved by using `grubx64.efi` instead of `grub_x86_64.efi`. Can you try this on your side and close this ticket if that solves your problem too?

gvancuts, 2021-02-05:
> Yes, I use the acrnctl command more.

Just a heads-up that this tool may be deprecated in the future. The idea is that ACRN could/would be managed through an API and higher-level libraries and utilities such as `libvirt`. It's still around today and actively used by Kata Containers when using ACRN (and it only uses the `blkrescan` function if I'm not mistaking) but that's also an area where we'd like to change the Kata Containers implementation so that it uses a proper API for this.

ionutnechita, 2021-02-05:
Is a very good idea with API and libvirt.
I would be interesting for such an implementation with API in the future on ACRN.

But until then, I will test Windows, Linux, Zephyr, Linux-RT with acrnctl.

Thanks.

gvancuts, 2021-02-08:
> Is a very good idea with API and libvirt.
> I would be interesting for such an implementation with API in the future on ACRN.

I agree, this feels like the right way to go to me too ;-)

> But until then, I will test Windows, Linux, Zephyr, Linux-RT with acrnctl.

Yes, I just wanted to make sure you were not building up a huge dependency on `acrnctl` on your side and that we would be pulling the rug from under your feet ;-)

fuzhongl, 2021-03-16:
As mentioned in: https://projectacrn.github.io/latest/tutorials/using_zephyr_as_uos.html
/efi/boot/bootx64.efi should work.

closed 2021-03-16 by fuzhongl