projectacrn/acrn-hypervisor #5705: [WindowsGuest] Less memory in the virtual machine than the initialization

ionutnechita, 2021-02-04:
Hi ACRN Team,

I managed to start a virtual machine with Windows 10 on NUC.
But I noticed that there is less memory in Windows than the initialization script.

In script / launch_win.sh memory is: 
#for memsize setting
mem_size=8192M

But in Windows is:
(systeminfo | Select-String 'Total Physical Memory:').ToString().Split(':')[1].Trim()
7,095 MB

Is it normal or isn't all the memory allocated to the virtual machine?

All the best.

[image]
[image]

fuzhongl, 2021-02-22:
narrow down to following patch:

```
commit 46afb946dad8e0298e3f813b5fcd1f44bed779c2
Author: Qian Wang <qian1.wang@intel.com>
Date:   Fri Oct 30 11:37:09 2020 +0800

    DM: modify vE820 to adapt to pSRAM

    dm: modify vE820 to adapt to pSRAM for post-launched RTVM

    When pSRAM is enabled for post-launched RTVM, we add a segment in vE820 for
    pSRAM, and therefore the lowmem RAM will be split into part1 and part2.
    Also, code of post vE820 initialization is refined

    Tracked-On: #5330
    Signed-off-by: Qian Wang <qian1.wang@intel.com>

```

fuzhongl, 2021-04-14:
@ionutnechita This issue is fixed in v2.4.
Please double check it.
Thanks!

ionutnechita, 2021-04-15:
Hi @fuzhongl,

This issue is fixed.
It's fine now.

Total Physical Memory: 7.94 GB

I tested with ACRN 2.4 and Scenario: Industrial / Industry

It's fine for close this ticket.

closed 2021-04-15 by fuzhongl