projectacrn/acrn-hypervisor #5710: [EHL][hybrid_rt_fusa] Fail to generate hybrid_rt_fusa scenario

szhen11, 2021-02-05:
Step:
run command to generate hybrid_rt_fusa scenario:

 make clean && make hypervisor BOARD=ehl-crb-b SCENARIO=hybrid_rt_fusa
And getting err:

/home/rui/cwpqa-acrn-robot/acrn-hypervisor/build/hypervisor/include/config.h generated
Warning: ttyS are fully used. ttyS0 is used for hv_console, ttyS1 is used for vuart1!
Board configurations for ehl-crb-b is generated successfully.
Error: Scenario xml file validation failed:
Error: vm:id=0,p2sb: Cannot enable p2sb and tpm passthru at the same time
Start pre-build static check ...
make -C ../misc/hv_prebuild BOARD=ehl-crb-b SCENARIO=hybrid_rt_fusa TARGET_DIR=/home/rui/cwpqa-acrn-robot/acrn-hypervisor/build/hypervisor/configs
make[2]: Entering directory '/home/rui/cwpqa-acrn-robot/acrn-hypervisor/misc/hv_prebuild'
make[2]: *** No rule to make target '/home/rui/cwpqa-acrn-robot/acrn-hypervisor/build/hypervisor/configs/scenarios/hybrid_rt_fusa/vm_configurations.c', needed by 'default'.  Stop.
make[2]: Leaving directory '/home/rui/cwpqa-acrn-robot/acrn-hypervisor/misc/hv_prebuild'
Makefile:380: recipe for target 'pre_build' failed
make[1]: *** [pre_build] Error 2
make[1]: Leaving directory '/home/rui/cwpqa-acrn-robot/acrn-hypervisor/hypervisor'
Makefile:99: recipe for target 'hypervisor' failed
make: *** [hypervisor] Error 2

szhen11, 2021-02-05:
[External_System_ID] ACRN-6960

yuchuyang, 2021-03-11:
[External_System_ID] ACRN-6888

closed 2021-03-16 by fuzhongl