projectacrn/acrn-hypervisor #5714: [WindowsGuest]  Image disappears after windows user account login

ionutnechita, 2021-02-05:
Hi ACRN Team,

I rebuild acrn.bin with the last commit and notice problems when starting windows.
After the user logs in, the interface disappears and streaks appear on the screen.

OS SOS:
HW NUC NUC7i7BNH
SCENARIO: Industrial/Industry
OpenSUSE Tumbleweed
5.4.90-PKT-200203T060100Z-g42a5e48c3e11 #4 SMP PREEMPT Wed Feb 3 13:05:20 EET 2021 x86_64 x86_64 x86_64 GNU/Linux
acrn-dm --version
DM version is: 2.4-unstable-1f686aea-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-02-05 12:30:19
Windows 10 Enterprise Evaluation

Latest base/commit:
commit 1f686aeac8a2843496729f174b912e9dedaa9608 (HEAD -> master, origin/master, origin/HEAD)
Author: Geoffroy Van Cutsem <geoffroy.vancutsem@intel.com>
Date:   Thu Feb 4 12:20:15 2021 +0100

    doc: remove obsolete GVT-g kernel parameters

    Various GVT-g kernel parameters are obsolete and no longer implemented
    in the latest kernels used with ACRN. Remove these obsolete parameters.

    Also remove a few pictures that are no longer referenced by any document.

    Tracked-On: #5702
    Signed-off-by: Geoffroy Van Cutsem <geoffroy.vancutsem@intel.com>

commit 07f4b9f5eb67a44755ba908b07c553e0b7e1f117
Author: David B. Kinder <david.b.kinder@intel.com>
Date:   Tue Feb 2 17:28:21 2021 -0800

    doc: cleanup xsd-derived config doc text

    Start cleaning up formatting and content layout issues in the
    xsd-derived configuration option documentation.  Includes adding
    documentation for unnamed embedded simple types within an element (and
    updates to the XSLT transformation to display these), cleanup of element
    and type documentation, typos and description clarity.

    Improved xsdl translation to automatically include default values and if
    an option is optional (instead of manually documenting this in the
    description text).

    Tracked-On: #5692

    Signed-off-by: David B. Kinder <david.b.kinder@intel.com>
...

With this base/commit: 94a980c923cb235ccdb7bf62c13ea86ff90aca05 is fine, 15 hours working correctly with Cinebench stress CPU

Author: Li Fei1 <fei1.li@intel.com>
Date:   Mon Feb 1 11:29:14 2021 +0800

    hv: hypercall: prevent sos can touch hv/pre-launched VM resource

    Current implementation, SOS may allocate the memory region belonging to
    hypervisor/pre-launched VM to a post-launched VM. Because it only verifies
    the start address rather than the entire memory region.

    This patch verifies the validity of the entire memory region before
    allocating to a post-launched VM so that the specified memory can only
    be allocated to a post-launched VM if the entire memory region is mapped
    in SOS’s EPT.

    Tracked-On: #5555
    Signed-off-by: Li Fei1 <fei1.li@intel.com>
    Reviewed-by: Yonghua Huang  <yonghua.huang@intel.com>
...

Is there a significant change in PCIE? or PCIE bar?

gvancuts, 2021-02-05:
Hi @ionutnechita , were you testing with the same Service VM kernel on both accounts?

Can you also confirm whether you keep the user-space tools (`acrn-dm` for example) in sync when you use different `acrn.bin` versions?

gvancuts, 2021-02-05:
For the record, these are the commits in between these two you've tested:
```
1f686aea (HEAD -> master, origin/master, origin/HEAD) doc: remove obsolete GVT-g kernel parameters
07f4b9f5 doc: cleanup xsd-derived config doc text
c741468b doc: remove Kconfig reference documentation
0ea991fb (tag: acrn-2021w06.3-180000p, origin/release_2.4) config_tools: add SW SRAM config
50d85256 HV: deny HV owned PCI bar access from SOS
6e7ce4a7 HV: deny pre-launched VM ptdev bar access from SOS
bc40e455 vm_config: pre-launched VM must have higher severity than SOS
d4aaf99d hv: keylocker: Support keylocker backup MSRs for Guest VM
38cd5b48 hv: keylocker: host keylocker iwkey context switch
c11c07e0 hv: keylocker: Support Key Locker feature for guest VM
4483e93b hv: keylocker: Enable the tertiary VM-execution controls
e9247dbc hv: keylocker: Simulate CPUID of keylocker caps for guest VM
15c967ad hv: keylocker: Add CR4 bit CR4_KL as CR4_TRAP_AND_PASSTHRU_BITS
0c7d8dd8 config-tools: check vmsix enablement for ehl-crb-b
1a0ab78a config_tools: update config app when creating new scenarios according to folder reorg
bf88e242 config_tools: scenario setting UI with xsd schema config
```

ionutnechita, 2021-02-05:
Hi @gvancuts,

// were you testing with the same Service VM kernel on both accounts?
Yes, using same kernel.

acrn-dm and acrn.bin is latest.
DM version is: 2.4-unstable-1f686aea-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-02-05 12:30:19

15-16 commits. is it's pretty complicated to see where the problem would be.

Can someone help us from development in this case.

I will return to acrn.bin and acrn-dm on February 2 for to further test Zephyr.

ionutnechita, 2021-02-05:
I also notice that trace:
[    6.193659] smpboot: CPU 2 is now offline
[    7.114772] ------------[ cut here ]------------
[    7.114778] WARNING: CPU: 0 PID: 673 at arch/x86/kernel/apic/msi.c:103 msi_set_affinity+0x1ca/0x230
[    7.114779] Modules linked in: openvswitch nsh ebtable_nat ebtable_broute ebtable_filter btusb btrtl btbcm btintel bluetooth ecdh_generic ecc usbhid snd_hda_codec_hdmi snd_soc_skl snd_hda_codec_realtek snd_hda_codec_generic snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_hda_intel snd_intel_nhlt snd_hda_codec rtsx_pci_sdmmc e1000e snd_hda_core xhci_pci xhci_hcd rtsx_pci efivarfs
[    7.114790] CPU: 0 PID: 673 Comm: wickedd Tainted: G     U            5.4.90-PKT-200203T060100Z-g42a5e48c3e11 #4
[    7.114790] Hardware name: Intel(R) Client Systems NUC7i7BNH/NUC7i7BNB, BIOS BNKBL357.86A.0082.2020.0505.1614 05/05/2020
[    7.114792] RIP: 0010:msi_set_affinity+0x1ca/0x230
[    7.114794] Code: 48 8b 40 60 e8 87 67 1b 01 48 8b 45 d0 65 48 2b 04 25 28 00 00 00 75 68 48 83 c4 30 44 89 f8 5b 41 5c 41 5d 41 5e 41 5f 5d c3 <0f> 0b 48 8d 75 b8 4c 89 ef 48 c7 45 b8 00 00 00 00 48 c7 45 c0 00
[    7.114794] RSP: 0018:ffffad5b008c32f0 EFLAGS: 00010097
[    7.114795] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[    7.114796] RDX: 0000000000000003 RSI: 0000000000000027 RDI: 00000000ffffffff
[    7.114796] RBP: ffffad5b008c3348 R08: 0000000000000028 R09: 0000000000000000
[    7.114797] R10: 0000000000000246 R11: ffff9ab5c1b46aa8 R12: ffff9ab5c1b46a28
[    7.114797] R13: ffff9ab5c46810c0 R14: ffff9ab5c4681100 R15: 0000000000000000
[    7.114798] FS:  00007f5ed62287c0(0000) GS:ffff9ab5d0e00000(0000) knlGS:0000000000000000
[    7.114799] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    7.114800] CR2: 000055d973824120 CR3: 0000000841a09006 CR4: 00000000003606f0
[    7.114800] Call Trace:
[    7.114804]  ? __netdev_alloc_skb+0x41/0x170
[    7.114807]  irq_do_set_affinity+0x32/0xa0
[    7.114808]  irq_setup_affinity+0x8a/0xd0
[    7.114810]  irq_startup+0x5e/0x100
[    7.114811]  __setup_irq+0x6c4/0x720
[    7.114812]  request_threaded_irq+0xfe/0x160
[    7.114820]  e1000_request_irq+0x1a0/0x350 [e1000e]
[    7.114825]  e1000e_open+0x154/0x5d0 [e1000e]
[    7.114828]  __dev_open+0xd4/0x180
[    7.114829]  __dev_change_flags+0x1dc/0x250
[    7.114831]  dev_change_flags+0x26/0x60
[    7.114832]  do_setlink+0x251/0xe60
[    7.114834]  ? __local_bh_enable_ip+0x45/0x90
[    7.114836]  ? _raw_read_unlock_bh+0x1c/0x20
[    7.114838]  ? trace_hardirqs_on+0x38/0x100
[    7.114840]  ? __nla_validate_parse+0x50/0x8d0
[    7.114840]  ? __local_bh_enable_ip+0x45/0x90
[    7.114842]  ? _raw_read_unlock_bh+0x1c/0x20
[    7.114844]  ? inet6_fill_ifla6_attrs+0x4aa/0x4d0
[    7.114846]  __rtnl_newlink+0x607/0x970
[    7.114849]  ? cpumask_next+0x1a/0x20
[    7.114850]  ? __snmp6_fill_stats64.constprop.0+0xc3/0xf0
[    7.114851]  ? nla_put+0x6d/0xa0
[    7.114853]  ? fib_nexthop_info+0x10d/0x1c0
[    7.114854]  ? nla_put+0x6d/0xa0
[    7.114856]  ? fib_dump_info_fnhe+0x153/0x200
[    7.114858]  ? trace_hardirqs_on_caller+0xe0/0xf0
[    7.114860]  ? kmem_cache_alloc_trace+0x100/0x290
[    7.114862]  rtnl_newlink+0x49/0x70
[    7.114863]  rtnetlink_rcv_msg+0x134/0x3a0
[    7.114865]  ? rtnl_calcit.isra.0+0x110/0x110
[    7.114867]  netlink_rcv_skb+0x4e/0x110
[    7.114869]  rtnetlink_rcv+0x15/0x20
[    7.114870]  netlink_unicast+0x18b/0x280
[    7.114871]  netlink_sendmsg+0x21d/0x3d0
[    7.114873]  ? _copy_from_user+0x85/0xb0
[    7.114875]  sock_sendmsg+0x65/0x70
[    7.114876]  ____sys_sendmsg+0x251/0x290
[    7.114877]  ___sys_sendmsg+0x7f/0xd0
[    7.114878]  ? _raw_spin_unlock+0x18/0x30
[    7.114880]  ? do_wp_page+0x1b6/0x580
[    7.114881]  ? __handle_mm_fault+0xb1f/0x1100
[    7.114881]  ? trace_hardirqs_on+0x38/0x100
[    7.114883]  __sys_sendmsg+0x5c/0xa0
[    7.114885]  __x64_sys_sendmsg+0x1f/0x30
[    7.114886]  do_syscall_64+0x55/0x130
[    7.114888]  entry_SYSCALL_64_after_hwframe+0x49/0xbe
[    7.114889] RIP: 0033:0x7f5ed66dcf17
[    7.114890] Code: 0c 00 f7 d8 64 89 02 48 c7 c0 ff ff ff ff eb b7 0f 1f 00 f3 0f 1e fa 64 8b 04 25 18 00 00 00 85 c0 75 10 b8 2e 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 51 c3 48 83 ec 28 89 54 24 1c 48 89 74 24 10
[    7.114891] RSP: 002b:00007ffd0193e7b8 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[    7.114891] RAX: ffffffffffffffda RBX: 000055d973824100 RCX: 00007f5ed66dcf17
[    7.114892] RDX: 0000000000000000 RSI: 00007ffd0193e7f0 RDI: 0000000000000006
[    7.114892] RBP: 000055d9737983c0 R08: 000055d97379c5e0 R09: 00007f5ed679ea60
[    7.114893] R10: 0000000000000080 R11: 0000000000000246 R12: 000055d97387a950
[    7.114893] R13: 00007ffd0193e7f0 R14: 000055d9738388d0 R15: 000055d973838b10
[    7.114895] ---[ end trace 70c6d0f6c8a368b0 ]---

gvancuts, 2021-02-05:
> // were you testing with the same Service VM kernel on both accounts?
> Yes, using same kernel.

Thanks for the confirmation

> 
> acrn-dm and acrn.bin is latest.
> DM version is: 2.4-unstable-1f686aea-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-02-05 12:30:19
> 
> 15-16 commits. is it's pretty complicated to see where the problem would be.
> 
> Can someone help us from development in this case.

It's not too many actually. I'll see if I can "git bisect" this for you. Most of the engineering team will be off over the next for the Chinese New year.

> 
> I will return to acrn.bin and acrn-dm on February 2 for to further test Zephyr.

Sounds good!

ionutnechita, 2021-02-05:
It may be the problem in the kernel and not in the hypervisor.

[   22.693643] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:51:pipe A] flip_done timed out
I noticed quite a few of these messages when booting.
Windows is automatically turned on by acrnd.

There may be a problem connecting the entire kernel and the drm / i915 modules.
Or bugs in I915.

ionutnechita, 2021-02-05:
I found out what the problem was:
In cmdline inserted splash.
root=/dev/sda2 console=ttyS0 idle=halt rw rootwait console=tty0 consoleblank=0 no_timer_check **splash** quiet loglevel=1 kaslr clocksource=hpet i915.nuclear_pageflip=1 i915.enable_initial_modeset=1 hvlog=2M@0xe00000 memmap=0x200000$0xe00000 maxcpus=4 root=PARTUUID=d444e5a8-ef96-48da-a81b-d8c450b39b35 hugepagesz=1G hugepages=28

And splash / splash logo conflicts with starting the Windows Virtual Machine.
More message observed on dmesg with splash parameter inserted:
[   23.741230] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:51:pipe A] flip_done timed out

gvancuts, 2021-02-08:
Nice catch @ionutnechita ! Closing this issue, feel free to re-open if needed.

gvancuts, 2021-02-08:
Actually, let me ask: is there some default that needs to be fixed in our codebase for this? Or is it something you had inserted yourself?

I only found a single occurence of the `splash` keyword in our code, and it's in one of our tutorial. So I think  the code base should be OK but looking for a confirmation here. Thanks!

closed 2021-02-08 by gvancuts

ionutnechita, 2021-02-08:
I just add it just to see the splash boot on SOS. Just for testing only.
I think it's better not to put it this parameter, so as not to encounter this problem on windows.

I don't know what the solution may be here.