projectacrn/acrn-hypervisor #5726: vhm: failed to set memory region for vm[1]!

ionutnechita, 2021-02-08:
Hi ACRN Team,

I noticed these errors when starting the first VM.
The first VM for me is Windows 10.

ACRN 2.4
acrn-dm --version
DM version is: 2.4-unstable-1f686aea-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-02-08 10:36:33
uname -a
Linux 5.4.90-PKT-200203T060100Z-g42a5e48c3e11-dirty #5 SMP PREEMPT Fri Feb 5 16:24:41 EET 2021 x86_64 x86_64 x86_64 GNU/Linux

[    1.192148] vhm: initializing
[    1.192156] vhm: hv api version 1.0
[    1.192159] vhm: registered correctly with major number 243
[    1.192165] vhm: device class registered correctly
[    1.192287] vhm: Virtio & Hypervisor service module initialized
[    7.583739] vhm: try to offline cpu 1 with lapicid 2
[    8.606954] vhm: try to offline cpu 2 with lapicid 1
[    9.636981] vhm: try to offline cpu 3 with lapicid 3
[    9.743631] vhm_dev_open: opening device node
[    9.744061] vhm-ioreq: init request buffer @ 000000002634eab0!
[    9.744065] vhm-ioreq: created ioreq client 1 for ioeventfd-1
[    9.750408] ACRN vhm ioeventfd init done!
[    9.750415] ACRN vhm irqfd init done!
[    9.750417] vhm: VM 1 created
**_[   10.512086] vhm: failed to set memory regions!
[   10.512087] vhm: failed to set memory region for vm[1]!
[   10.512088] vhm: failed to del memory region 1!
[   10.513842] vhm: failed to set memory regions!
[   10.513844] vhm: failed to set memory region for vm[1]!
[   10.513844] vhm: failed to del memory region 1!_**
[   10.586399] vhm-ioreq: created ioreq client 2 for acrndm

Have you seen him before?

fuzhongl, 2021-02-22:
@ionutnechita There is WaaG stability issue  for mainline.
Please try v2.3 branch。
Thanks！

ionutnechita, 2021-02-23:
Hi @fuzhongl,

I tested with ACRN 2.3 and ACRN Kernel 5.4.90.
And I didn't see this problem with vhm when I started Windows.
Thanks for information.

Will it be repaired in 2.5?

fuzhongl, 2021-03-02:
@ionutnechita This issue should be fixed in v2.4.
Thanks!

ionutnechita, 2021-03-22:
Is there an update here?

In this moment this issue is reproduced:
Linux version 5.4.90-PKT-200203T060100Z-00006-g07fb8047e3dc (root@demo-acrn-opensuse-labl1) (gcc version 10.2.1 20210303 [revision 85977f624a34eac309f9d77a58164553dfc82975] (SUSE Linux)) #1 SMP PREEMPT Mon Mar 22 13:00:34 EET 2021
acrn-dm --version
DM version is: 2.4-unstable-ae659e58a-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-03-22 12:36:01

fuzhongl, 2021-03-23:
@ionutnechita It was fixed in v2.4 branch, Please have a try.
Thanks!

ionutnechita, 2021-03-24:
Hi @fuzhongl,

On the master branch does this fix not exist?

fuzhongl, 2021-03-25:
@ionutnechita 
Master branch is dev branch and it keeps moving...
There is no stability test for master branch. please try v2.4.
Thanks!

fuzhongl, 2022-02-25:
Fixed since v2.4.

closed 2022-02-25 by fuzhongl