projectacrn/acrn-hypervisor #5740: [WindowsGuest] unknown device "fbuf" in acrn-dm

ionutnechita, 2021-02-10:
Hi ACRN Team,

acrn-dm -A -m $mem_size -s 0:0,hostbridge -s 1:0,lpc -l com1,stdio \
  -s 2,passthru,0/2/0,gpu \
  -s 3,virtio-blk,/usr/share/acrn/conf/win.img \
  -s 4,virtio-net,tap0 \
  -s 5,xhci,1-1,1-2 \
  -s 8,fbuf,tcp=0.0.0.0:5900,w=800,h=600 \
  --ovmf /usr/share/acrn/bios/OVMF.fd \
  --windows \
  $vm_name
}

pci slot 8:0: unknown device "fbuf"

acrn-dm --version
DM version is: 2.4-unstable-1f686aea-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-02-08 10:36:33

For VNC, how to start this?

gvancuts, 2021-02-10:
Hmmm... excellent question.... I don't see any occurence of `fbuf` in our code... and it makes me wonder if we ever really used this. I don't see any implementation of that parameter in our `devicemodel`.

@anthonyzxu @ywan170 can you shed some light on this?

ywan170, 2021-02-18:
The fbuf is only used in very early phase of Windows enabling. And it hasn't merged to acrn-dm as we have supported the GOP for gvt-d and gvt-g. @NanlinXie We need to update the document..

dbkinder, 2022-01-26:
@ywan170 @NanlinXie What documentation needs to be updated here? Can you help provide material to include?

ywan170, 2022-01-27:
@dbkinder Please ignore this issue. The fbuf is one invalid parameter, it has dropped.

ionutnechita, 2022-01-28:
Ok. Please close this ticket if is invalid in this moment.

A good day.

amyreye, 2022-02-16:
@NanlinXie @dbkinder Can we close the ticket?

dbkinder, 2022-02-16:
There's no mention of fbuf in the mentioned doc (https://projectacrn.github.io/latest/tutorials/using_windows_as_user_vm.html)

closed 2022-02-16 by dbkinder