projectacrn/acrn-hypervisor #5741: doc: Supported HW doc is obsolete and misleading

dbkinder, 2021-02-10:
The supported HW document (https://projectacrn.github.io/latest/reference/hardware.html) describes platforms as "verified".  We know that older hardware platforms, such as Apollo Lake, cannot fully support newer versions of ACRN, so the hardware list is misleading.  Platforms were verified for a specific version of ACRN, but are longer verified as new versions of ACRN are released.

We at least need to add a notation indicating the ACRN version used to verify a platform (e.g., ``V1.0``).  We should also list which ACRN features are verified on particular (newer) platforms and won't run on older platforms.  (That way, someone with an older HW platform can know what they can and cannot do with ACRN.)  @NanlinXie Can the QA or support team provide such information?

Alternatively, we may need to just remove older platforms that don't fully support the latest release of ACRN.

dbkinder, 2021-02-10:
@junjiemao1 can we use the board.xml information to automatically figure out what ACRN features a board can or cannot support as part of the board/scenario validation?

junjiemao1, 2021-02-22:
@dbkinder We are about to refine the structure of board.xml in the v2.5 release to make the XML file more informative and machine-friendly. We also plan to introduce some examples how unavailable features due to hardware limitations can be recorded as XSD assertions and reported in the board configuration phase. More thorough checks will be introduced in the release after the next, after we verified the infrastructure.

gvancuts, 2021-03-03:
To add to this:
* Can we confirm whether the table indicating what is validated or not is still accurate? (note: we've discovered many issues when running on Apollo Lake which make me suspicious about the SDC validation on that platform...)
* The list of usage scenarios does not match what we have in the code today... we need to update the list **and** make sure we check the ones we **actually** validate ;-)

[image]

dbkinder, 2021-05-25:
@NanlinXie it is OK to tag this as a documentation task for v2.5?

PR #6217 (2021-06-21): doc: update supported hardware document
Fixes: #5741
commit 414f871b (2021-06-21): doc: update supported hardware document

Fixes: #5741
closed 2021-06-21 by dbkinder