projectacrn/acrn-hypervisor #5746: [WindowsGuest] After a few stops and starts of the virtual machine, nothing appears on the display

ionutnechita, 2021-02-12:
Hi ACRN Team,

I tried to make a stop and start scenario for windows, and it seems that after a while it no longer works.

I observe FULL RESET flag, but SOS is clean in this case, without conflict with other VM.

# acrnctl start win_vm1
cat: /usr/share/acrn/conf/add/win_vm1.args: No such file or directory
cpu1 online=0
cpu2 online=0
cpu3 online=0
ioctl(TUNSETIFF): Device or resource busy
ovs-vsctl: cannot create a port named tap0 because a port named tap0 already exists on bridge ovsbr_eno1
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x200000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/1 page size:0x200000
level 1 free/need pages:28/8 page size:0x40000000

try to setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
total_size 0x300000000

mmap ptr 0x0x7fd6b1eaf000 -> baseaddr 0x0x7fd6c0000000
mmap 0x80000000@0x7fd6c0000000
touch 2 pages with pagesz 0x40000000
mmap 0x180000000@0x7fd800000000
touch 6 pages with pagesz 0x40000000
mmap 0x200000@0x7fd7bfe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
vm_init_vdevs
pm_by_vuart_init idx: 0, path: /run/acrn/life_mngr_win_vm1
pci init hostbridge
pci init lpc
pci init passthru
polling 35...
Listening 35...
initiate a connection on a socket error
create socket to connect life-cycle manager failed
pci init virtio-blk
pci init virtio-net
pci init xhci
pci init igd-lpc
tpm: init_vtpm2:Invalid socket path!
acrn_sw_load
SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 0x200000 copied to addr 0x7fd7bfe00000
SW_LOAD: build e820 9 entries to addr: 0x7fd6c00ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x0000000000100000, size 0x000000003b700000,  type 0x1
SW_LOAD: entry[2]: addr 0x000000003b800000, size 0x0000000004004000,  type 0x2
SW_LOAD: entry[3]: addr 0x000000003f804000, size 0x000000003fffc000,  type 0x2
SW_LOAD: entry[4]: addr 0x000000007f800000, size 0x0000000000800000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000140000000, size 0x0000000180000000,  type 0x1
SW_LOAD: entry[7]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: entry[8]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: ovmf_entry 0xfffffff0
add_cpu
full reset
vm_suspend: setting VM state to FULL_RESET
VM state changed from[ RUNNING ] to [ FULL_RESET ]
failed: notify request finish
VM loop exit
pci deinit hostbridge
pci deinit lpc
pci deinit passthru
vm_reset_ptdev_intx:0x10-10, ioapic virpin=0.
pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
pci deinit xhci
pci deinit igd-lpc
mngr_client_new: Failed to accept from fd 35, err: Invalid argument
Stop listening 35...
Stop polling 35...
main: setting VM state to RUNNING
VM state changed from[ FULL_RESET ] to [ RUNNING ]
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x200000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:1/1 page size:0x200000
level 1 free/need pages:28/8 page size:0x40000000

try to setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
total_size 0x300000000

mmap ptr 0x0x7fd6b1eaf000 -> baseaddr 0x0x7fd6c0000000
mmap 0x80000000@0x7fd6c0000000
touch 2 pages with pagesz 0x40000000
mmap 0x180000000@0x7fd800000000
touch 6 pages with pagesz 0x40000000
mmap 0x200000@0x7fd7bfe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
        level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
        level 1 - lowmem 0x80000000, biosmem 0x0, highmem 0x180000000
vm_init_vdevs
pm_by_vuart_init idx: 0, path: /run/acrn/life_mngr_win_vm1
pci init hostbridge
pci init lpc
pci init passthru
pci slot 31:0 already occupied!
faild to create igd-lpcpolling 35...
Listening 35...
initiate a connection on a socket error
create socket to connect life-cycle manager failed
pci init virtio-blk
pci init virtio-net
pci init xhci
pci init igd-lpc
tpm: init_vtpm2:Invalid socket path!
acrn_sw_load
SW_LOAD: partition blob /usr/share/acrn/bios/OVMF.fd size 0x200000 copied to addr 0x7fd7bfe00000
SW_LOAD: build e820 9 entries to addr: 0x7fd6c00ef008
SW_LOAD: entry[0]: addr 0x0000000000000000, size 0x00000000000a0000,  type 0x1
SW_LOAD: entry[1]: addr 0x0000000000100000, size 0x000000003b700000,  type 0x1
SW_LOAD: entry[2]: addr 0x000000003b800000, size 0x0000000004004000,  type 0x2
SW_LOAD: entry[3]: addr 0x000000003f804000, size 0x000000003fffc000,  type 0x2
SW_LOAD: entry[4]: addr 0x000000007f800000, size 0x0000000000800000,  type 0x2
SW_LOAD: entry[5]: addr 0x00000000e0000000, size 0x0000000020000000,  type 0x2
SW_LOAD: entry[6]: addr 0x0000000140000000, size 0x0000000180000000,  type 0x1
SW_LOAD: entry[7]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: entry[8]: addr 0x0000000000000000, size 0x0000000000000000,  type 0x0
SW_LOAD: ovmf_entry 0xfffffff0
add_cpu
^CReceived SIGINT to terminate application...
sig_handler_term: setting VM state to POWEROFF
VM state changed from[ RUNNING ] to [ POWEROFF ]
attach ioreq client return 1 (1 = destroying, could be triggered by Power State change, others = error)
VM loop exit
pci deinit hostbridge
pci deinit lpc
pci deinit passthru
vm_reset_ptdev_intx:0x10-10, ioapic virpin=0.
pci deinit virtio-blk
pci deinit virtio-net
vtnet tx thread closing...
pci deinit xhci
pci deinit igd-lpc
mngr_client_new: Failed to accept from fd 35, err: Invalid argument
Stop listening 35...
Stop polling 35...

Test Script Scenario, executed from SOS:
cat ./stop-start-waag.sh
#!/bin/bash

for (( i=1; i<=100; i++ ))
do
        echo "Stop/Start WaaG/Windows: $i"
        acrnctl stop win_vm1
        sleep 120
        acrnctl start win_vm1 &
        sleep 120
done

Windows/WaaG is configured with S5 support.
life_mngr_win.exe is installed in shell:startup and working for 10 stop / start VM.

after 10 stop / start it happens not to work for starting VM.

SOS:
acrn-dm -v
DM version is: 2.4-unstable-85f4d798-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-02-11 10:55:39
OpenSUSE Tumbleweed
5.4.90-PKT-200203T060100Z-g42a5e48c3e11-dirty #5 SMP PREEMPT Fri Feb 5 16:24:41 EET 2021 x86_64 x86_64 x86_64 GNU/Linux

fuzhongl, 2021-04-06:
@ionutnechita This issue should be fixed with the latest code of v2.4.
Please heave a try.

fuzhongl, 2022-03-22:
Fixed in v2.4

closed 2022-03-22 by fuzhongl