projectacrn/acrn-hypervisor #5749: Make `make install` use the same parameters used during build

gvancuts, 2021-02-17:
Our build system takes a few parameters, such as `BOARD`, `SCENARIO` and `RELEASE`. So you can, as an example, build a release version for an UP2 board by running: `make BOARD=apl-up2 SCENARIO=industry RELEASE=1`

The next natural step, if you've been building this on your target, is to run `sudo make install`. **But**
1. The first problem is that you need to repeat the exact same parameters that were used during build for it to work as expected, i.e. in our particular case we'd need to run: `sudo make BOARD=apl-up2 SCENARIO=industry RELEASE=1 install` else it will throw an error.
2. The second problem is that if you are trying to be smart and directly call `sudo make BOARD=apl-up2 SCENARIO=industry RELEASE=1 install` because it will build the stuff as a pre-requisite, the build process will actually fail.

We should fix these to improve the overall developer experience.

gvancuts, 2021-02-17:
@NanlinXie @junjiemao1 , who can be assigned to this?

junjiemao1, 2021-03-03:
The first problem has been solved by #5753. I'll take a look at the second.

gvancuts, 2022-06-28:
Blocked by #7836 for today