projectacrn/acrn-hypervisor #5761: [RFE} Kata Containers CI doesn't support ACRN

fidencio, 2021-02-23:
Kata Containers has ACRN as one of its supported VMMs, even providing a configuration file that could be used by folks using ACRN. 

However, this not tested as part of Kata Containers CI (please, see: https://github.com/kata-containers/tests/issues/3269), and would be really really great if we (as Kata Containers) could have some help to setting up the CI on Kata Containers side, avoiding then possible regressions.

gvancuts, 2021-02-23:
I think it would be great indeed! I am not familiar with the Kata CI system, so not sure how to help but if you can describe how you test other VMMs today, hopefully I can guide you with ACRN?

@vijaydhanraj , FYI

fidencio, 2021-02-23:
@gvancuts, You can take a look at what we do as part of the CI [here](https://github.com/kata-containers/tests/tree/main/.ci).

The important parts that we'd need, as a starting point, is a script to install ACRN, something similar to what we have for CLH or QEMU (https://github.com/kata-containers/tests/blob/main/.ci/install_cloud_hypervisor.sh or https://github.com/kata-containers/tests/blob/main/.ci/install_qemu.sh), and then, from that point, we can have a discussion on what kind of tests would be interesting to run when using ACRN.

Kata Containers project has an Architecture Committee meeting every Tuesday at 3pm GMT.  I can add this topic to the next meeting and we can have a chat on what are the next steps.  Would this suit you?

/cc jcvenegas

vijaydhanraj, 2021-02-23:
Yes, this is much needed to ensure the support is not broken. Appreciate the initiative. @gvancuts please let me know if you need any help.

fidencio, 2021-02-23:
@vijaydhanraj, @gvancuts, I've added the topic to the next Kata Containers Architecture Committee meeting.

Please, take a look at https://etherpad.opendev.org/p/Kata_Containers_2021_Architecture_Committee_Mtgs for the agenda, instructions on how to join, etc.

In case that doesn't work for you, please, let me know and I can remove that accordingly.

fidencio, 2021-02-23:
Last but not least, we're always on Slack / IRC: https://katacontainers.io/community/ :-)

gvancuts, 2021-02-23:
> @vijaydhanraj, @gvancuts, I've added the topic to the next Kata Containers Architecture Committee meeting.
>
> Please, take a look at [...]

That works for me!

gvancuts, 2022-07-05:
What exactly was completed for this @fuzhongl ?

fuzhongl, 2022-07-05:
@gvancuts There is known issue that ACRN doesn't  work with kata master code.
Will fix it in acrn v3.1

gvancuts, 2022-07-05:
> @gvancuts There is known issue that ACRN doesn't work with kata master code. Will fix it in acrn v3.1

OK, why did you close it then?

fuzhongl, 2022-07-05:
This issue is tracked by: https://github.com/projectacrn/acrn-hypervisor/issues/6631
Do we need more tickets to track the same issue?

gvancuts, 2022-07-05:
> This issue is tracked by: #6631

Not the same, this one is about the Kata Containers CI system that needs to incorporate tests for ACRN. This is gated by the fact that ACRN support in Kata is broken of course.

> Do we need more tickets to track the same issue?

We do not, in this case we do because the issues are different. If you did believe they were the same, why did you create #6631 at the time while this one already existed? ;-)