projectacrn/acrn-hypervisor #5766: Hypervisor version information retrieval

gvancuts, 2021-02-24:
At the moment, the only way to retrieve the hypervisor version information is using the `version` command in the ACRN debug shell. There are a couple of restrictions with this:
* It is not possible to easily retrieve such information from the Service VM (more on that below [1])
* Such information is completely inaccessible in a system running a production build (`RELEASE=1`)

There should be a new API/hypercall that one can use to retrieve the hypervisor (`acrn.bin`) version information.

(see also [this thread](https://lists.projectacrn.org/g/acrn-dev/message/30895?p=,,,20,0,0,0::Created,,version,20,2,0,80707145) on the mailing list)

[1] About version information in the Service VM:
* When running Linux, there is an ACRN hypervisor API information displayed by the ACRN driver in dmesg. But it is not actually used
* `acrn-dm -v` will return the version information about the ACRN Device Model (`acrn-dm) **only**. It does not say, nor does it check, the version of the hypervisor.

gvancuts, 2022-01-27:
I don't know... would love to know too :-)

junjiemao1, 2022-04-22:
There are some internal discussions and efforts on this. Let me try summarizing.

As this idea was to help users identify if they had matched versions of components, the first thing we noticed was that it was not sufficient to know and compare the versions of the HV and the DM only. A third component, the ACRN driver (i.e. HSM) in the kernel, was also part of the game. Even the HV and DM have matched versions, new hypercalls are still unsupported if an older version of kernel is used in the service VM.

Thus, we tried to add some information about the version of that driver, but got rejected by the community because it made no sense to hard code ACRN versions in the source tree of the upstream kernel. The recommended way to tackle such incompatibility issues is API backward compatibility plus special behavior (possibly returning dedicated error codes) upon calling unsupported & obsoleted APIs. The former is embedded in the requirement analysis and architecture designs phases during feature planning, and some engineering efforts are made to implement the latter. The error logs of the DM will be updated accordingly (after we review those logs systemtically in the second half of this year) to translate error codes to user-friendly messages and instructions.

Back to fetching HV version (for informative purposes) from the service VM, we can print it in the ACRN log once the shared buffer for the logging mechanism is installed, if such information is helpful to users in other cases. Unlike the hypercall approach, using ACRN log allows us to provide arbitrary and extensible information without kernel changes. After we revised the "release" build type to be more useful by including some facilities of logging, using ACRN log will no longer be a problem there.

gvancuts, 2022-04-25:
Thanks for this detailed information @junjiemao1. I think this will cover the use case.

closed 2022-07-05 by fuzhongl