projectacrn/acrn-hypervisor #5781: ACRN Device Model arguments and parameters out-of-sync

gvancuts, 2021-02-25:
The ACRN Device Model (`acrn-dm`) parameter descriptions in our documentation has gone (once more :-() a bit out of sync with what's in the actual code. These three pages need to be back in sync:
* (Source code) https://github.com/projectacrn/acrn-hypervisor/blob/master/devicemodel/core/main.c (check consistency between `usage()` function and actual implementation
* https://projectacrn.github.io/latest/user-guides/acrn-dm-parameters.html
* https://projectacrn.github.io/latest/developer-guides/hld/hld-devicemodel.html#configuration (the examples throughout the doc also needs updating, inc. https://projectacrn.github.io/latest/developer-guides/hld/hld-devicemodel.html#pci-host-bridge-and-hierarchy)

wenlingz, 2021-02-26:
@gvancuts Please add JIRA ID, thanks

gvancuts, 2021-03-01:
[External_System_ID] ACRN-6997

gvancuts, 2021-03-01:
> 
> 
> @gvancuts Please add JIRA ID, thanks

Done - thanks

amyreye, 2022-02-17:
@gvancuts What's the status?

NanlinXie, 2022-05-04:
@gvancuts , could we close this issue now?

closed 2022-07-05 by fuzhongl