projectacrn/acrn-hypervisor #5789: [WindowsGuest] How to create a new virtual machine with Windows 10 LTS/LTSC

ionutnechita, 2021-02-26:
Hi ACRN Team,

I try to create one vm with windows, but windows installer not appear on my display.
How to proceed?

gvancuts, 2021-03-01:
Hi @ionutnechita , what is your HW platform in this case?

ionutnechita, 2021-03-01:
My HW platform is: 
NUC7I7BNH - Intel(R) Client Systems NUC7i7BNH/NUC7i7BNB, BIOS BNKBL357.86A.0082.2020.0505.1614 05/05/2020
Intel(R) Core(TM) i7-7567U CPU @ 3.50GHz
00:02.0 VGA compatible controller: Intel Corporation Iris Plus Graphics 650 (rev 06)

fuzhongl, 2021-03-07:
@ionutnechita 
The GOP drive is needed for GVT-d.
Please follow this  [link](https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html)

ionutnechita, 2021-03-07:
Hi @fuzhongl,

PREREQUISITES:
1) Get your specific "IntelGopDriver.efi" and "Vbt.bin"
    from your BIOS vendor

Where can i get these files from?  ( NUC7I7BNH )

fuzhongl, 2021-03-08:
@ionutnechita 
You should get them from manufactory of NUC7I7BNH.
Another solution about create WaaG image:
Install native windows in SATA/NVMe SSD or USB stick; then passthrough it to WaaG in launch script.

ionutnechita, 2021-03-08:
Ok, but don't these files exist in public?

I'll look for them in public and maybe I'll find them.

The second method will be the last option for me.
Thanks.

ionutnechita, 2021-03-08:
Hi @fuzhongl,

I found IntelGopDriver.efi and Vbt.bin for KBL.
I tested the installation of Windows 10 LTSC and Windows 2019 Server (year 2018).
Both went correctly exactly as in the documentation.

https://projectacrn.github.io/latest/tutorials/gpu-passthru.html#gpu-passthrough
https://projectacrn.github.io/latest/tutorials/using_windows_as_uos.html

ACRN HV / DM - DM version is: 2.4-unstable-c843d2bbc-dirty (daily tag:acrn-2021w06.3-180000p), build by root@2021-03-05 20:08:36
ACRN Kernel version is: Linux demo-acrn-opensuse-lab1 5.4.90-PKT-200203T060100Z-00002-g2d8c7622646a #1 SMP PREEMPT Fri Feb 12 20:21:08 EET 2021 x86_64 x86_64 x86_64 GNU/Linux
Scenario: Industrial / Industry

The ticket can be closed, the documentation is correctly.

Thanks.

gvancuts, 2021-03-08:
Thanks @ionutnechita !

@NanlinXie , I think we should catch this opportunity to update the doc to use the latest tags (`v2.4`), what do you think?

closed 2021-03-08 by gvancuts

fuzhongl, 2021-03-09:
@ionutnechita 
Could you please share IntelGopDriver.efi and Vbt.bin of KBL NUC7I7BNH?
Thanks!

iaGuoZhi, 2021-04-28:
Hi
@fuzhongl  and ACRN team
Do you have IntelGopDriver.efi and Vbt.bin of WHL-IPC-I5? Could you please share them wich me. I'd like to try running windows VM on ACRN in board WHL-IPC-I5.
Thanks!

fuzhongl, 2021-04-28:
@iaGuoZhi Please connect with Maxtang who own IntelGopDriver.efi and Vbt.bin of WHL-IPC-I5.
ACRN can't share them for license issue.