projectacrn/acrn-hypervisor #5793: Enhance `misc/Makefile` for better readability and more intuitive build folder

gvancuts, 2021-03-01:
The `misc/Makefile` can easily be enhanced to improve on a couple of points:
* Improve readability by grouping and consistently using the same order when building and installing all the tools
   * They should be grouped according on the subfolders (i.e. `debug_tools` and `services`)
   * They should be sorted, with `services` tools listed first
* All `services` and `debug_tools` get built and put in the same build folder, called `build/misc/debug_tools` by default. This is misleading and a bit confusing as all the `services` components get put in there too. Two separate subfolders should be created instead, called `build/misc/services` and `build/misc/debug_tools` by default.

wenlingz, 2021-03-04:
@gvancuts please add JIRA ID, thanks

gvancuts, 2021-03-04:
[External_System_ID] ACRN-7010

gvancuts, 2021-03-16:
Patch was merged, closing.

closed 2021-03-16 by gvancuts