projectacrn/acrn-hypervisor #5801: Windows User OS not starting on Apollo Lake platforms

gvancuts, 2021-03-02:
It is not possible to start a Windows User VM on Apollo Lake platforms.
* GVT-d is not enabled (as indicated here: https://github.com/projectacrn/acrn-hypervisor/blob/master/hypervisor/acpi_parser/dmar_parse.c#L136-L143
* GVT in general is broken - see https://github.com/intel/gvt-linux/issues/84

Attempting to start a Windows User VM with GV-g will lead to the following error:
```
cpu1 online=0
passed gvt-g optargs low_gm 64, high_gm 448, fence 8
SW_LOAD: get ovmf path /usr/share/acrn/bios/OVMF.fd, size 0x200000
vm_create: win_vm1
VHM api version 1.0
vm_setup_memory: size=0x40000000
open hugetlbfs file /run/hugepage/acrn/huge_lv1/win_vm1/D279543825D611E8864ECB7A18B34643
open hugetlbfs file /run/hugepage/acrn/huge_lv2/win_vm1/D279543825D611E8864ECB7A18B34643
level 0 free/need pages:513/1 page size:0x200000
level 1 free/need pages:0/1 page size:0x40000000
to reserve more free pages:
to reserve pages (+orig 0): echo 1 > /sys/kernel/mm/hugepages/hugepages-1048576kB/nr_hugepages
now enough free pages are reserved!

try to setup hugepage with:
	level 0 - lowmem 0x0, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x40000000, biosmem 0x0, highmem 0x0
total_size 0x180000000

mmap ptr 0x0x7f48676a3000 -> baseaddr 0x0x7f4880000000
mmap 0x40000000@0x7f4880000000
touch 512 pages with pagesz 0x200000
mmap 0x200000@0x7f497fe00000
touch 1 pages with pagesz 0x200000

really setup hugepage with:
	level 0 - lowmem 0x40000000, biosmem 0x200000, highmem 0x0
	level 1 - lowmem 0x0, biosmem 0x0, highmem 0x0
vm_init_vdevs
No correct pm notify channel given
polling 34...
Listening 34...
pci init hostbridge
pci init lpc
pci init pci-gvt
GVT: open /sys/kernel/gvt/control/create_gvt_instance failed
GVT: init failed
pci pci-gvt init failed
mngr_client_new: Failed to accept from fd 34, err: Invalid argument
Stop listening 34...
Stop polling 34...
No correct pm notify channel given
Unable to init vdev (2)
```

fuzhongl, 2021-03-07:
@gvancuts 
Please use GVT-g for APL platform since it  doesn't support GVT-d.

gvancuts, 2021-03-08:
> @gvancuts
> Please use GVT-g for APL platform since it doesn't support GVT-d.

GVT in general is broken on APL, that means both GVT-g **and** GVT-d.

gvancuts, 2021-03-08:
This https://github.com/intel/gvt-linux/issues/84#issuecomment-792502055 has information about the patches that would need to be put in the 5.4 LTS kernel (which I guess is the pre-requisite before they can land in our `acrn-kernel` repo, isn't it?).

I do not know if these would be sufficient to get **both** GVT-g **and** GTV-d working on APL. I guess these would only fix the former.

fuzhongl, 2021-03-09:
> > @gvancuts
> > Please use GVT-g for APL platform since it doesn't support GVT-d.
> 
> GVT in general is broken on APL, that means both GVT-g **and** GVT-d.

@gvancuts There is known issue for GVT-g since v2.1; v2.0 is working for GVT-g.

gvancuts, 2021-03-09:
> @gvancuts There is known issue for GVT-g since v2.1; v2.0 is working for GVT-g.

OK, so the summary as far as Apollo Lake is concerned is the following:
1. GVT-g: working up until (and inc.) ACRN v2.0.
2. GVT-d: was never enabled on Apollo Lake

Is that a correct statement?

Regarding point 1, does it boil down to back-porting these patches (https://github.com/intel/gvt-linux/issues/84#issuecomment-792502055) on teh 5.4 LTS kernel (and pulling them into our `acrn-kernel` tree)?

Regarding point 2 (GVT-d), do you know how much effort would be needed to support it?

Thanks!

fuzhongl, 2021-03-10:
> > @gvancuts There is known issue for GVT-g since v2.1; v2.0 is working for GVT-g.
> 
> OK, so the summary as far as Apollo Lake is concerned is the following:
> 
> 1. GVT-g: working up until (and inc.) ACRN v2.0.
> 2. GVT-d: was never enabled on Apollo Lake
> 
> Is that a correct statement?
> 
> Regarding point 1, does it boil down to back-porting these patches ([intel/gvt-linux#84 (comment)](https://github.com/intel/gvt-linux/issues/84#issuecomment-792502055)) on teh 5.4 LTS kernel (and pulling them into our `acrn-kernel` tree)?
> 
> Regarding point 2 (GVT-d), do you know how much effort would be needed to support it?
> 
> Thanks!

About point 1: This topic was discussed in ACRN project meeting several month ago; the decision is low priority for GVT-g; waiting for KVM upstream  to fix it since IDV solution is used for ACRN(since v2.1) and KVM for GVT-g;
About point 2: My understanding: It is APL hardware doesn't support GVT-d. Please double check with @NanlinXie 
Thanks!

fuzhongl, 2022-03-22:
Work as design GVT-g isn't supported since v2.1

closed 2022-03-22 by fuzhongl