projectacrn/acrn-hypervisor #5805: using initrd with pre-launch kernel giving outof memory issue

pchand20, 2021-03-03:
**Describe the bug**
when using ramdisk_tag in hybrid config for initrd image . we are seeing grub alocator issue our of memory.

**Platform**
TGL

**Codebase**
V-2.3

**Scenario**
Hybrid: with pre-launch linux kernel and  initrd image as module

**To Reproduce**
Steps to reproduce the behavior:
1. add ramdisk_tag in hybrid config 
2. build image with prelaunch vm and initrd
3. change config to add "module2 /<initrd image name> <tag>
4. boot into pre-launch

**Expected behavior**
WARNING: no console will be available to OS
error: ../../git/grub-core/lib/relocator.c:1439:out of memory.

Press any key to continue...

**Additional context**
Attach the following logs if available:

WARNING: no console will be available to OS
error: ../../git/grub-core/lib/relocator.c:1439:out of memory.

Press any key to continue...
error: ../../git/grub-core/video/video.c:761:no suitable video mode found.
Multiboot2 Bootloader: GRUB 2.04
ACRN Hypervisor
calibrate_tsc, tsc_khz=2803200
[478113718us][cpu=0][(null)][sev=2][seq=1]:HV version 2.3-unstable-2021-03-03 00:36:31-d57a21324-dirty DBG (daily tag:acrn-2020w48.5-180000p) hybrid_rt@tgl-rvp build by chandrap with acrn-config, start time 478105152us
[478136307us][cpu=0][(null)][sev=2][seq=2]:API version 1.0
[478142543us][cpu=0][(null)][sev=2][seq=3]:Detect processor: 11th Gen Intel(R) Core(TM) i7-1185GRE @ 2.80GHz
[478532342us][cpu=2][(null)][sev=3][seq=18]:Unsupported multiboot module: start at 0x0, end at 0x0
[478534902us][cpu=0][(null)][sev=2][seq=25]:Start VM id: 1 name: ACRN SOS VM
[478542617us][cpu=2][(null)][sev=3][seq=26]:failed to load VM 0 acpi module
[478560408us][cpu=2][(null)][sev=2][seq=41]:Start VM id: 0 name: ACRN PRE-LAUNCHED VM0
ACRN:\>[478814277us][cpu=0][vm1:vcpu0][sev=3][seq=44]:vlapic: Start Secondary VCPU1 for VM[1]...
[480550756us][cpu=0][vm1:vcpu0][sev=2][seq=46]:vpci_write_cfg 0:1c.0 not found! off: 0xdb, val: 0x0

[480561344us][cpu=0][vm1:vcpu0][sev=2][seq=47]:vpci_write_cfg 0:1c.1 not found! off: 0xdb, val: 0x0

[480571985us][cpu=0][vm1:vcpu0][sev=2][seq=48]:vpci_write_cfg 0:1c.2 not found! off: 0xdb, val: 0x0

[480582607us][cpu=0][vm1:vcpu0][sev=2][seq=49]:vpci_write_cfg 0:1c.3 not found! off: 0xdb, val: 0x0

[480593238us][cpu=0][vm1:vcpu0][sev=2][seq=50]:vpci_write_cfg 0:1c.4 not found! off: 0xdb, val: 0x0

[480603881us][cpu=0][vm1:vcpu0][sev=2][seq=51]:vpci_write_cfg 0:1c.5 not found! off: 0xdb, val: 0x0

[480614511us][cpu=0][vm1:vcpu0][sev=2][seq=52]:vpci_write_cfg 0:1c.6 not found! off: 0xdb, val: 0x0

[480625134us][cpu=0][vm1:vcpu0][sev=2][seq=53]:vpci_write_cfg 0:1c.7 not found! off: 0xdb, val: 0x0

[480635767us][cpu=0][vm1:vcpu0][sev=2][seq=54]:vpci_write_cfg 0:1d.0 not found! off: 0xdb, val: 0x0

[480646410us][cpu=0][vm1:vcpu0][sev=2][seq=55]:vpci_write_cfg 0:1d.1 not found! off: 0xdb, val: 0x0

[480657050us][cpu=0][vm1:vcpu0][sev=2][seq=56]:vpci_write_cfg 0:1d.2 not found! off: 0xdb, val: 0x0

[480667674us][cpu=0][vm1:vcpu0][sev=2][seq=57]:vpci_write_cfg 0:1d.3 not found! off: 0xdb, val: 0x0

[480678299us][cpu=0][vm1:vcpu0][sev=2][seq=58]:vpci_write_cfg 0:1d.4 not found! off: 0xdb, val: 0x0

[... 96 lines cut ...]
[481202696us][cpu=0][vm1:vcpu0][sev=2][seq=107]:vpci_write_cfg 0:1d.1 not found! off: 0xdb, val: 0x0

[481213440us][cpu=0][vm1:vcpu0][sev=2][seq=108]:vpci_write_cfg 0:1d.2 not found! off: 0xdb, val: 0x0

[481224165us][cpu=0][vm1:vcpu0][sev=2][seq=109]:vpci_write_cfg 0:1d.3 not found! off: 0xdb, val: 0x0

[481234901us][cpu=0][vm1:vcpu0][sev=2][seq=110]:vpci_write_cfg 0:1d.4 not found! off: 0xdb, val: 0x0

[481245634us][cpu=0][vm1:vcpu0][sev=2][seq=111]:vpci_write_cfg 0:1d.5 not found! off: 0xdb, val: 0x0

[481256379us][cpu=0][vm1:vcpu0][sev=2][seq=112]:vpci_write_cfg 0:1d.6 not found! off: 0xdb, val: 0x0

[481267106us][cpu=0][vm1:vcpu0][sev=2][seq=113]:vpci_write_cfg 0:1d.7 not found! off: 0xdb, val: 0x0

[481277839us][cpu=0][vm1:vcpu0][sev=2][seq=114]:vpci_write_cfg 0:1b.0 not found! off: 0xdb, val: 0x0

[481288574us][cpu=0][vm1:vcpu0][sev=2][seq=115]:vpci_write_cfg 0:1b.1 not found! off: 0xdb, val: 0x0

[481299316us][cpu=0][vm1:vcpu0][sev=2][seq=116]:vpci_write_cfg 0:1b.2 not found! off: 0xdb, val: 0x0

[481310044us][cpu=0][vm1:vcpu0][sev=2][seq=117]:vpci_write_cfg 0:1b.3 not found! off: 0xdb, val: 0x0

[481320777us][cpu=0][vm1:vcpu0][sev=2][seq=118]:vpci_write_cfg 0:1b.4 not found! off: 0xdb, val: 0x0

[481331520us][cpu=0][vm1:vcpu0][sev=2][seq=119]:vpci_write_cfg 0:1b.5 not found! off: 0xdb, val: 0x0

pchand20, 2021-03-03:
[image]

pchand20, 2021-03-04:
got a some debug detail when relocator is called. 
[image]

gvancuts, 2021-03-04:
Not sure if this is completely related but some patches are needed in order to use large initramfs/initrd. Can you try to integrate and test those?
* https://github.com/tttech-industrial-buchsbaum/acrn-hypervisor/blob/debian/master/debian/patches/0009-hv-vm_load-avoid-overwriting-source-images.patch,
* https://github.com/tttech-industrial-buchsbaum/acrn-hypervisor/blob/debian/master/debian/patches/0011-hv-boot-enhance-Linux-boot-header-structure.patch
* https://github.com/tttech-industrial-buchsbaum/acrn-hypervisor/blob/debian/master/debian/patches/0012-vboot_info-load-ramdisk-at-highest-possible-memory-l.patch

pchand20, 2021-03-04:
counter some issue after these patches. hangs on console when run any VM. 

ACRN Hypervisor
calibrate_tsc, tsc_khz=2803200
[35206856us][cpu=0][(null)][sev=2][seq=1]:HV version 2.3-unstable-2021-03-03 00:36:31-d57a21324-dirty DBG (daily tag:acrn-2020w48.5-180000p) hybrid_rt@tgl-rvp build by chandrap with acrn-config, start time 35198288us
[35229241us][cpu=0][(null)][sev=2][seq=2]:API version 1.0
[35235375us][cpu=0][(null)][sev=2][seq=3]:Detect processor: 11th Gen Intel(R) Core(TM) i7-1185GRE @ 2.80GHz
[35625246us][cpu=2][(null)][sev=3][seq=28]:vm_sw_loader, Cannot copy required SW modules: -1
[35626348us][cpu=0][(null)][sev=2][seq=33]:Start VM id: 1 name: ACRN SOS VM
[35634899us][cpu=2][(null)][sev=2][seq=34]:Start VM id: 0 name: ACRN PRE-LAUNCHED VM0
[35653655us][cpu=2][vm0:vcpu0][sev=3][seq=39]:decode instruction failed @ 0x00000000000f2400:ACRN:\>
[35664119us][cpu=2][vm0:vcpu0][sev=2][seq=41]:Guest Linear Address: 0x00000000fffffffc
[35673215us][cpu=2][vm0:vcpu0][sev=2][seq=42]:Guest Physical Address address: 0x00000000fffffffc
[35683337us][cpu=2][vm0:vcpu0][sev=1][seq=43]:VM0: triple fault @ guest RIP 0x00000000000f2400, exit qualification: 0x0000000000000000
[35697347us][cpu=2][vm0:vcpu0][sev=3][seq=44]:decode instruction failed @ 0x00000000000f2406:
[35707163us][cpu=2][vm0:vcpu0][sev=2][seq=46]:Guest Linear Address: 0x00000000fffffffc
[35716253us][cpu=2][vm0:vcpu0][sev=2][seq=47]:Guest Physical Address address: 0x00000000fffffffc
[35726391us][cpu=2][vm0:vcpu0][sev=1][seq=48]:VM0: triple fault @ guest RIP 0x00000000000f2406, exit qualification: 0x0000000000000000
[35740398us][cpu=2][vm0:vcpu0][sev=3][seq=49]:decode instruction failed @ 0x00000000000f240b:
[35750204us][cpu=2][vm0:vcpu0][sev=2][seq=51]:Guest Linear Address: 0x00000000fffffffc
[35759299us][cpu=2][vm0:vcpu0][sev=2][seq=52]:Guest Physical Address address: 0x00000000fffffffc
[35769421us][cpu=2][vm0:vcpu0][sev=1][seq=53]:VM0: triple fault @ guest RIP 0x00000000000f240b, exit qualification: 0x0000000000000000
[35901869us][cpu=2][vm0:vcpu0][sev=1][seq=54]:VM0: triple fault @ guest RIP 0x0000000007c884b6, exit qualification: 0x0000000000000000
[35915813us][cpu=2][vm0:vcpu0][sev=1][seq=55]:VM0: triple fault @ guest RIP 0x0000000007c884be, exit qualification: 0x0000000000000000
[35998736us][cpu=2][vm0:vcpu0][sev=3][seq=56]:decode instruction failed @ 0x000000000c7fffff:
[36008488us][cpu=2][vm0:vcpu0][sev=2][seq=58]:Guest Linear Address: 0x00000000ffffff8e
[36017577us][cpu=2][vm0:vcpu0][sev=2][seq=59]:Guest Physical Address address: 0x00000000ffffff8e
[36027693us][cpu=2][vm0:vcpu0][sev=1][seq=60]:VM0: triple fault @ guest RIP 0x000000000c7fffff, exit qualification: 0x0000000000000000
[36041702us][cpu=2][vm0:vcpu0][sev=3][seq=61]:decode instruction failed @ 0x000000000c800007:
[36051515us][cpu=2][vm0:vcpu0][sev=2][seq=63]:Guest Linear Address: 0x00000000fffffffc
[36060610us][cpu=2][vm0:vcpu0][sev=2][seq=64]:Guest Physical Address address: 0x00000000fffffffc
[36070725us][cpu=2][vm0:vcpu0][sev=1][seq=65]:VM0: triple fault @ guest RIP 0x000000000c800007, exit qualification: 0x0000000000000000
[36084733us][cpu=2][vm0:vcpu0][sev=3][seq=66]:decode instruction failed @ 0x000000000c800013:
[36094545us][cpu=2][vm0:vcpu0][sev=2][seq=68]:Guest Linear Address: 0x00000000fffffffc
[36103635us][cpu=2][vm0:vcpu0][sev=2][seq=69]:Guest Physical Address address: 0x00000000fffffffc
[36113757us][cpu=2][vm0:vcpu0][sev=1][seq=70]:VM0: triple fault @ guest RIP 0x000000000c800013, exit qualification: 0x0000000000000000

ACRN:\>vm_list

VM_UUID                          VM_ID VM_NAME                          VM_STATE
================================ ===== ================================ ========
b2a92becca6b11eab1063716a8ba0bb9   0   ACRN PRE-LAUNCHED VM0            Running
dbbbd4347a574216a12c2201f1ab0240   1   ACRN SOS VM                      Running
ACRN:\>vm_console 0

----- Entering VM 0 Shell -----

 ---Entering ACRN SHELL---
ACRN:\>vm_console 1

----- Entering VM 1 Shell -----

pchand20, 2021-03-04:
here is my boards e820 config from native linux boot.

 dmesg | grep e820
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000338bcfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000338bd000-0x0000000036a90fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000036a91000-0x0000000036b32fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000036b33000-0x0000000036bfefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000036bff000-0x0000000036bfffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000036c00000-0x0000000038ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000039400000-0x00000000395fffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000003a200000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff400000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000049fbfffff] usable
[    0.000005] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000006] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.007724] e820: update [mem 0x2fca2000-0x2fca2fff] usable ==> reserved
[    3.353994] e820: reserve RAM buffer [mem 0x0009f000-0x0009ffff]
[    3.353997] e820: reserve RAM buffer [mem 0x2fca2000-0x2fffffff]
[    3.353998] e820: reserve RAM buffer [mem 0x338bd000-0x33ffffff]
[    3.353999] e820: reserve RAM buffer [mem 0x36c00000-0x37ffffff]
[    3.354000] e820: reserve RAM buffer [mem 0x49fc00000-0x49fffffff]

jsun26intel, 2021-03-23:
the message of "error: ../../git/grub-core/lib/relocator.c:1439:out of memory" comes from grub, means grub could not find memory slot to put the module.
"--nounzip" parameter could be used in grub to prevent .gz file be extracted when loading the module. If it still does not work, then seems no better solution to resolve this issue, except changing grub code.
I suggest to close this issue.
For the issue that large initrd could be loaded by grub but doesn't work with ACRN, please issue another ticket to track.

Or please modify title if you still want to track the issue by this ticket.

fuzhongl, 2021-03-26:
@jsun26intel Thanks for comments
https://github.com/projectacrn/acrn-hypervisor/issues/5879 
will track **`hybrid_rt scenario does not work with large initrd in pre-launched VM issue`**.

So close this issue now.

closed 2021-03-26 by fuzhongl