projectacrn/acrn-hypervisor #5811: Booting ACRN on Apollo Lake UP2 causes platform reboot

gvancuts, 2021-03-03:
**Describe the bug**
Booting ACRN (v2.3 or master) on UP2 leads to a reboot of the platform.

**Platform**
Apollo Lake UP2

**Codebase**
v2.3 and master

**Scenario**
Tried with both `sdc` and `industry`, using `RELEASE=0`

**To Reproduce**
* Build ACRN v2.3 or master using `make BOARD=apl-up2 SCENARIO=sdc`
* Install all build artefacts, configure Grub
* Reboot the UP2 board

**Expected behavior**
Service VM (Ubuntu) coming up

**Additional context**
It was observed that turning off the serial output "fixes" the issue and the Service VM comes up as expected (and ACRN is running correctly).

So either building with `RELEASE=1` or using `uart=disabled` makes the Service VM come up correctly.

fuzhongl, 2021-03-12:
@gvancuts please share `dmesg |grep ttyS`  for native and Service VM boot.

gvancuts, 2021-03-18:
Native (Ubuntu 20.04) environment:
```
$ dmesg | grep ttyS
[    3.225468] dw-apb-uart.8: ttyS4 at MMIO 0x91526000 (irq = 4, base_baud = 115200) is a 16550A
[    3.245452] dw-apb-uart.9: ttyS5 at MMIO 0x91524000 (irq = 5, base_baud = 115200) is a 16550A
```

In the Service VM (using `uart=disabled`):
```
$ dmesg | grep ttyS
[    0.000000] Command line: root=/dev/mmcblk0p3 console=ttyS0 idle=halt rw rootwait console=tty0 consoleblank=0 no_timer_check quiet loglevel=3 i915.nuclear_pageflip=1 i915.enable_gvt=1 hvlog=2M@0xe00000 memmap=0x600000$0xa00000 ramoops.mem_address=0xa00000 ramoops.mem_size=0x400000 ramoops.console_size=0x200000 reboot_panic=p,w module_blacklist=dwc3_pci i915.enable_guc=0x02 cma=64M@0- maxcpus=4 uart=disabled root=PARTUUID=3ceead4a-f25a-4526-87aa-6950a9670537
[    0.059669] Kernel command line: root=/dev/mmcblk0p3 console=ttyS0 idle=halt rw rootwait console=tty0 consoleblank=0 no_timer_check quiet loglevel=3 i915.nuclear_pageflip=1 i915.enable_gvt=1 hvlog=2M@0xe00000 memmap=0x600000$0xa00000 ramoops.mem_address=0xa00000 ramoops.mem_size=0x400000 ramoops.console_size=0x200000 reboot_panic=p,w module_blacklist=dwc3_pci i915.enable_guc=0x02 cma=64M@0- maxcpus=4 uart=disabled root=PARTUUID=3ceead4a-f25a-4526-87aa-6950a9670537
[    0.170939] printk: console [ttyS0] enabled
[    0.989115] serial8250: ttyS0 at I/O 0x3f8 (irq = 4, base_baud = 115200) is a 16550A
[    4.444943] dw-apb-uart.8: ttyS1 at MMIO 0x91526000 (irq = 4, base_baud = 115200) is a 16550A
[    4.446632] dw-apb-uart.9: ttyS2 at MMIO 0x91524000 (irq = 5, base_baud = 115200) is a 16550A
```